libimobiledevice/idevicerestore #229: Specify SHSH blobs.

MrBenFTW, 2018-09-09:
Hello, and yes I know, something like this already exists with `FutureRestore` However. Future restore does not support FaceID // Savage, meaning that using FutureRestore on a iPhone X will inevitably break FaceID.

I noticed in issue #196 that You added support so faceID could work. I am wondering if you could either fork future restore and apply your fix, or add the option to specify Blobs using iDeviceRestore. I a happy to give a €20 if you can get it working :)

truonggiang0710, 2018-09-24:
Hi @MrBenFTW , do you need this now ?

MrBenFTW, 2018-09-24:
Afaik, future restore is still not working with faceid so.. yes please :)

truonggiang0710, 2018-09-25:
@MrBenFTW I need know more your request. Contact me Skype: truonggiang0710