libimobiledevice/idevicerestore #232: idevicerestore fails with iPhone Xs and Xs Max

taanders, 2018-09-26:
idevicerestore fails with iPhone Xs and Xs Max. Devices boot to black screen with Apple logo and circular wait indicator after "About to restore device..." log message.

There seems to be some new parts on ipsw files:
ANE, AVE, GFX, ISP, LowPowerWallet0, SIO, WCHFirmwareUpdater and Yonkers

```
C:\msys64_build\mingw32\bin>idevicerestore.exe -e -l -d
PROGRESS:0
Reading data from version.xml
NOTE: using cached version data
PROGRESS:10
Found device in Normal mode
PROGRESS:20
Identified device as d321ap, iPhone11,2
Latest firmware is iPhone11,2_12.0_16A366_Restore.ipsw
Verifying 'iPhone11,2_12.0_16A366_Restore.ipsw'...
Checksum matches.
PROGRESS:59
Extracting BuildManifest from IPSW
PROGRESS:80
Product Version: 12.0
Product Build: 16A366 Major: 16
Device supports Image4: true
Variant: Customer Erase Install (IPSW)
This restore will erase your device data.
PROGRESS:0
Using cached filesystem from 'iPhone11,2_12.0_16A366_Restore/048-10822-224.dmg'
Extracting filesystem from IPSW completed
PROGRESS:20
Getting device's ECID for TSS request
Found ECID 594930619121710
Getting ApNonce in normal mode... 18 04 8f 8e a2 85 97 d9 fe f0 ae 66 df d8 ab c7 9c 29 fd 4d f8 fa 42 3e ef a6 c6 57 56 9c f8 c0
Trying to fetch new SHSH blob
Getting SepNonce in normal mode... 5f a1 0e c9 08 0d 17 a2 18 0d 26 61 d4 3e cc c8 c0 04 51 3d
NOTE: Unable to find BbCalibrationManifestKeyHash node
NOTE: Unable to find BbFactoryActivationManifestKeyHash node
WARNING: Unable to find BbSkeyId node
DEBUG: Applying restore request rules for entry ANE
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry AOP
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry AVE
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry AppleLogo
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry AudioCodecFirmware
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry BatteryCharging0
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry BatteryCharging1
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry BatteryFull
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry BatteryLow0
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry BatteryLow1
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry BatteryPlugin
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
[... 1599 lines cut ...]
NOTE: No path for component RestoreDeviceTree in TSS, will fetch from build_identity
Extracting DeviceTree.d321ap.im4p...
Personalizing IMG4 component RestoreDeviceTree...
Tag found
Sending RestoreDeviceTree (192605 bytes)...
DEBUG: tss_response_get_path_by_entry: No entry 'RestoreKernelCache' in TSS response
NOTE: No path for component RestoreKernelCache in TSS, will fetch from build_identity
Extracting kernelcache.release.iphone11...
Personalizing IMG4 component RestoreKernelCache...
Tag found
Sending RestoreKernelCache (17850949 bytes)...
PROGRESS:90
About to restore device...
Waiting for device...
Attempt 1 to connect to restore mode device...
Attempt 2 to connect to restore mode device...
Attempt 3 to connect to restore mode device...
...
Attempt 180 to connect to restore mode device...
ERROR: Unable to connect to device in restore mode
ERROR: Unable to open device in restore mode
ERROR: Unable to restore device
```

[iphone_xs_iTunes_logs.zip](https://github.com/libimobiledevice/idevicerestore/files/2419789/iphone_xs_iTunes_logs.zip)

taanders, 2018-09-26:
I tried also with stuff included on this pull request but it doesn't help only with those devices:
https://github.com/libimobiledevice/idevicerestore/pull/231

Kanong017, 2018-09-27:
@taanders I also have the same problem
Do you have mail or contact information like what'sApp?

nikias, 2018-09-27:
I am aware of this problem and I think I know how to fix it. Working on it.

taanders, 2018-10-02:
@nikias HI, have you found any fixes for this issue? What is causing that flash process is not starting on device side?

rekyagis, 2018-10-03:
The attached patch (idevicerestore_iphone-xs.patch) make the restore process to finish. (And I expect the restore is correct.)
The log file of restoring iPhone XS is also attached (iphone-xs_restore.log), FYI.

I found three major difference in iPhone XS.
- We need to send additional files in recovery mode. They are marked as "IsLoadedByiBoot" in build manifest. (Please see recovery_send_loaded_by_iboot().)
- It has new component. The patch will get the component-name dynamically from the build manifest. (Please see get_component_name().)
- It requests another firmware named "Yonkers". (Please see restore_get_yonkers_firmware_data() and tss_request_add_yonkers_tags().)

I hope I can be of any help to you. Thanks.

[idevicerestore_iphone-xs.patch.txt](https://github.com/libimobiledevice/idevicerestore/files/2443038/idevicerestore_iphone-xs.patch.txt)
[iphone-xs_restore.log](https://github.com/libimobiledevice/idevicerestore/files/2443044/iphone-xs_restore.log)

nikias, 2018-10-03:
@rekyagis thanks. This is exactly what needs to be done. I will need to apply this slighty modified but the general idea is correct here.

taanders, 2018-10-04:
Thanks @rekyagis  and @nikias. Should there be componentbuf = NULL;?
```
@@ -1055,6 +1059,9 @@ int restore_send_nor(restored_client_t restore, struct idevicerestore_client_t*
 		component_data = NULL;
 		component_size = 0;
 
+		free(componentbuf);
+		component = NULL;  <--
+
 		/* make sure iBoot is the first entry in the array */
 		if (!strncmp("iBoot", filename, 5)) {
 			plist_array_insert_item(norimage_array, plist_new_data((char*)nor_data, (uint64_t)nor_size), 0);
```

rekyagis, 2018-10-05:
It is 'component'. ('componentbuf' also should be set to NULL here, I think.)
It implies that we can not use 'component' after 'componentbuf' is freed, since 'component' (returned by get_component_name()) and 'componentbuf' (allocated in get_component_name()) is same address.

pubsocent, 2018-10-19:
@rekyagis I tried using your patch on an iPhone 7 but still ran into the same issue.
I have a full log but to summarize here is the tail end of the log:

`About to restore device...
Waiting for device...
Attempt 1 to connect to restore mode device...
restore_is_current_device: device f27855e93dcdeddf4b7d08b4b0ce572eb49194c5 is not in restore mode
Attempt 2 to connect to restore mode device...
Attempt 3 to connect to restore mode device...
Attempt 4 to connect to restore mode device...
Attempt 5 to connect to restore mode device...
Attempt 6 to connect to restore mode device...
Attempt 7 to connect to restore mode device...
Attempt 8 to connect to restore mode device...
Attempt 9 to connect to restore mode device...
Attempt 10 to connect to restore mode device...
Attempt 11 to connect to restore mode device...
Attempt 12 to connect to restore mode device...
Attempt 13 to connect to restore mode device...
Attempt 14 to connect to restore mode device...
Attempt 15 to connect to restore mode device...
Attempt 16 to connect to restore mode device...
Attempt 17 to connect to restore mode device...
Attempt 18 to connect to restore mode device...
Attempt 19 to connect to restore mode device...
Attempt 20 to connect to restore mode device...
Attempt 21 to connect to restore mode device...
Attempt 22 to connect to restore mode device...
Attempt 23 to connect to restore mode device...
Attempt 24 to connect to restore mode device...
Attempt 25 to connect to restore mode device...
Attempt 26 to connect to restore mode device...
Attempt 27 to connect to restore mode device...
Attempt 28 to connect to restore mode device...
Attempt 29 to connect to restore mode device...
Attempt 30 to connect to restore mode device...
Attempt 31 to connect to restore mode device...
Attempt 32 to connect to restore mode device...
Attempt 33 to connect to restore mode device...
Attempt 34 to connect to restore mode device...
Attempt 35 to connect to restore mode device...
Attempt 36 to connect to restore mode device...
Attempt 37 to connect to restore mode device...
Attempt 38 to connect to restore mode device...
Attempt 39 to connect to restore mode device...
Attempt 40 to connect to restore mode device...
Attempt 41 to connect to restore mode device...
Attempt 42 to connect to restore mode device...
Attempt 43 to connect to restore mode device...
Attempt 44 to connect to restore mode device...
Attempt 45 to connect to restore mode device...
Attempt 46 to connect to restore mode device...
Attempt 47 to connect to restore mode device...
Attempt 48 to connect to restore mode device...
Attempt 49 to connect to restore mode device...
Attempt 50 to connect to restore mode device...
Attempt 51 to connect to restore mode device...
Attempt 52 to connect to restore mode device...
Attempt 53 to connect to restore mode device...
Attempt 54 to connect to restore mode device...
Attempt 55 to connect to restore mode device...
Attempt 56 to connect to restore mode device...
Attempt 57 to connect to restore mode device...
Attempt 58 to connect to restore mode device...
Attempt 59 to connect to restore mode device...
Attempt 60 to connect to restore mode device...
Attempt 61 to connect to restore mode device...
Attempt 62 to connect to restore mode device...
Attempt 63 to connect to restore mode device...
Attempt 64 to connect to restore mode device...
[... 95 lines cut ...]
Attempt 160 to connect to restore mode device...
Attempt 161 to connect to restore mode device...
Attempt 162 to connect to restore mode device...
Attempt 163 to connect to restore mode device...
Attempt 164 to connect to restore mode device...
Attempt 165 to connect to restore mode device...
Attempt 166 to connect to restore mode device...
Attempt 167 to connect to restore mode device...
Attempt 168 to connect to restore mode device...
Attempt 169 to connect to restore mode device...
Attempt 170 to connect to restore mode device...
Attempt 171 to connect to restore mode device...
Attempt 172 to connect to restore mode device...
Attempt 173 to connect to restore mode device...
Attempt 174 to connect to restore mode device...
Attempt 175 to connect to restore mode device...
Attempt 176 to connect to restore mode device...
Attempt 177 to connect to restore mode device...
Attempt 178 to connect to restore mode device...
Attempt 179 to connect to restore mode device...
Attempt 180 to connect to restore mode device...
ERROR: Unable to connect to device in restore mode
ERROR: Unable to open device in restore`

I am happy to test, I have several devices. Let me know how I can help!

rekyagis, 2018-10-23:
@pubsocent It seems that your issue is not the same as this original one.
Sometimes I have also faced the same issue as yours, and re-plugging iPhone at this point can help.
I am not sure the reason but idevice_event_subscribe function (in libimobiledevice) may sometimes fail to detect re-connection.

pekn, 2018-10-31:
I noticed similar with iOS 12.1. You cannot restore using idevicerestore without user interaction - you need to disconnect & connect a couple of times in order to restore successfully. This didn't occur with 12.0.x.

Blefish, 2018-10-31:
It seems to be related to specific hardware configuration on host side, I am able to trigger iTunes in Windows to do the same. It varies but I think it has to do with hubs that can power devices without host connectivity.

pubsocent, 2018-10-31:
Well just to reiterate... I am happy to provide testing. I have all devices including all configurations. Happy to contribute if any progress is made here!

Blefish, 2018-11-01:
I tested with a Cypress SDK where I can manually turn on/off USB features. I found out that CDP (charging downstream port) is causing this problem. Unfortunately this cannot be fixed by libimobiledevice as it is directly hardware related and iOS detection of USB host. I believe the phone only starts to charge if it finds out CDP is enabled on the USB port.

A current fix is to use a USB cable that for example, does not bring any meaningful current to the device. It has been described here as well: https://discussions.apple.com/thread/8608085

A reddit thread, I also posted my findings there: https://www.reddit.com/r/iphone/comments/9t0ddq/ios_121_update_bricked_my_iphone_8_plus/

This bug only started happening after iOS 12.1 betas and 12.1 official release. It happens for me with all devices that got the update.

truonggiang0710, 2018-11-02:
Hi @Blefish, 

Do you think Apple can fix this in next iOS or is it new secure of Apple on iOS 12.1 ?

Blefish, 2018-11-02:
I am not sure, in iOS 12.1.1 beta this is not fixed. Damage has been already done though.
I tested in my laptop's dedicated USB charging port (battery icon next to it), it also does not work there.

pekn, 2018-11-06:
I was able to reproduce this with iTunes.. so it's not really idevicerestore specific problem.

pubsocent, 2018-12-07:
Just tested iOS 12.1.1 on iPhone 7 running version 11.4.1 and another iPhone 7 running 12.0. 
Both encountered same issue as stated above. 
I was hoping 12.1.1 release would have addressed the issue. Anyone else have any luck?

truonggiang0710, 2018-12-08:
@Blefish @pubsocent Yes, I test OK on 12.1.1

pubsocent, 2018-12-08:
@truonggiang0710 Did you build from master or did you also apply patches or a specific branch?

nikias, 2019-01-26:
This issue should be resolved with latest code.

closed 2019-01-26 by nikias