libimobiledevice/idevicerestore #235: iPhone 6S ERROR: device didn't accept BasebandData

gstone787, 2018-11-01:
I am consistently failing to restore an iPhone 6S ending in the following error:
Unhandled progress operation 36 (36)
Updating baseband (19)
ERROR: device didn't accept BasebandData
ERROR: Unable to successfully restore device
I have tried with and without sudo and get the same result. Full log is attached. Any help is greatly appreciated.
[idevicerestorefail.txt](https://github.com/libimobiledevice/idevicerestore/files/2539291/idevicerestorefail.txt)

gstone787, 2018-12-01:
I am still having this issue. Anybody have any suggestions?

Kanong017, 2018-12-13:
Because your device baseband is broken.
If you want to skip, try this.

`int restore_handle_bb_update_status_msg(restored_client_t client, plist_t msg)
{
	int result = -1;
	plist_t node = plist_dict_get_item(msg, "Accepted");
	uint8_t accepted = 0;
	plist_get_bool_val(node, &accepted);

	if (!accepted)
	{
		error("ERROR: device didn't accept BasebandData\n");
		//return result;
		return 0; //add this
	}`

gstone787, 2018-12-14:
> Because your device baseband is broken.
> If you want to skip, try this.
> 
> int restore_handle_bb_update_status_msg(restored_client_t client, plist_t msg)
> {
> int result = -1;
> plist_t node = plist_dict_get_item(msg, "Accepted");
> uint8_t accepted = 0;
> plist_get_bool_val(node, &accepted);
> 
> ```
> if (!accepted)
> {
> 	error("ERROR: device didn't accept BasebandData\n");
> 	//return result; 
> 	return 0; //add this
> }
> ```
> ...
> @gstone787

Thank you so much for the reply. Forgive my newbieness but where do those lines need to go?

taanders, 2019-01-08:
I have got few other BaseBandData errors:
```
Done sending BasebandData
ERROR: device didn't accept BasebandData
ERROR: Unable to successfully restore device
common.c:printing 565 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>CHECKPOINT_COMPLETE</key>
	<true/>
	<key>CHECKPOINT_ERROR</key>
	<string>[0]D(update_baseband: failed to perform next stage)[1]D(Fail to bootup baseband: kBBUReturnIOError)[2]D(failed connecting to sahara
)</string>
	<key>CHECKPOINT_ID</key>
	<integer>4867</integer>
	<key>CHECKPOINT_RESULT</key>
	<integer>0</integer>
	<key>MsgType</key>
	<string>CheckpointMsg</string>
</dict>
</plist>
ERROR: Unable to restore device
```

And other error:
```
Updating baseband (19)
ERROR: device didn't accept BasebandData
ERROR: Unable to successfully restore device
common.c:printing 544 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>CHECKPOINT_COMPLETE</key>
	<true/>
	<key>CHECKPOINT_ERROR</key>
	<string>[0]D(update_baseband: failed querying baseband info)[1]D(Failed creating a communication transport to baseband.)</string>
	<key>CHECKPOINT_ID</key>
	<integer>4867</integer>
	<key>CHECKPOINT_RESULT</key>
	<integer>0</integer>
	<key>MsgType</key>
	<string>CheckpointMsg</string>
</dict>
</plist>
ERROR: Unable to restore device
```

taanders, 2019-01-08:
Is it possible that iOS device is on some hanging state after previously failed flashing?