libimobiledevice/idevicerestore #237: Error in downgrading iPhone 3GS to iOS 3.0 using idevicerestore

Jessehh9, 2018-11-17:
Hi. I'm following a guide (https://diosra2.hatenadiary.jp/entry/20180712/1531402157) to downgrade my old bootrom iPhone 3GS to iOS 3.0 using idevicerestore. In the last step, when I use the command described in the guide, I find this error: 

> MacBook-Pro-di-Marco:macos Marco$ ./idevicerestore -e -w iPhone2,1_3.0_7A341_Custom.ipsw
> NOTE: using cached version data
> Found device in DFU mode
> Identified device as iPhone2,1
> Extracting BuildManifest from IPSW
> Product Version: 3.0
> Product Build: 7A341 Major: 7
> Device supports Image4: false
> Variant: Customer Erase Install (IPSW)
> This restore will erase your device data.
> Found ECID 600390296460
> checking for local shsh
> Using cached SHSH
> Using cached filesystem from 'iPhone2,1_3.0_7A341_Custom/018-5302-002.dmg'
> Extracting iBSS.n88ap.RELEASE.dfu...
> Not personalizing component iBSS...
> Sending iBSS (104772 bytes)...
> [==================================================] 100.0%
> ERROR: Unable to connect to recovery device
> ERROR: Unable to place device into recovery mode from DFU mode

How can I fix this error? Thank you!

h4tena, 2019-05-19:
I have the same problem. Since this was a while ago, have you found a solution?

Jessehh9, 2019-05-19:
Nope... Since no one has replied, I gave up and downgraded my 3GS to iOS 3.1.3 building a custom signed firmware with redsn0w.

allbombson, 2019-07-31:
where you able to restore that firmware using this?

Jessehh9, 2019-08-01:
There's a particular version of redsn0w that allows you to do it. There are many videos on YouTube that explain how to do it, just search "how to download iPhone 3GS to iOS 3.1.3".

ShimaRin2502, 2021-12-18:
Put it in PwnedDFU mode and try the restore again.

uf3n3qoasdnapnkem, 2022-04-06:
> 你好。我正在按照指南（https://diosra2.hatenadiary.jp/entry/20180712/1531402157）使用idevicerestore将我的旧bootrom iPhone 3GS降级到iOS 3.0。在最后一步中，当我使用指南中描述的命令时，我发现此错误：
> 
> > MacBook-Pro-di-Marco：macos Marco$ ./idevicerestore -e -w iPhone2，1_3.0_7A341_Custom.ipsw 注意：使用缓存的版本数据 在 DFU 模式下找到设备 识别为 iPhone2，1 从 IPSW 提取 BuildManifest 产品版本：3.0 产品内部版本：7A341 主要：7 设备支持映像4：假 变体：客户擦除安装 （IPSW） 此恢复将擦除您的设备数据。发现 ECID 600390296460检查本地 shsh 使用缓存的 SHSH 使用缓存文件系统从 'iPhone2，1_3.0_7A341_Custom/018-5302-002.dmg' 提取 iBSS.n88ap.RELEASE.dfu...未对组件 iBSS 进行个性化设置...正在发送 iBSS（104772 字节）... [==================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================
> 
> 调试模式：
> 
> > MacBook-Pro-di-Marco：macos Marco$ ./idevicerestore -e -w -d iPhone2，1_3.0_7A341_Custom.ipsw 从版本读取数据.xml 注意：使用缓存版本数据打开设备 05ac：1227... irecv_copy_nonce_with_tag： 警告： 在字符串中找不到标记 NONC Apple Inc. irecv_copy_nonce_with_tag： 警告： 在字符串中找不到标记 SNON Apple Inc. 设置为配置 1 设置到接口 0：0 打开设备 05ac：1227... irecv_copy_nonce_with_tag： 警告： 在字符串中找不到标记 NONC Apple Inc. irecv_copy_nonce_with_tag： 警告： 在字符串中找不到标记 SNON Apple Inc. 设置为配置 1 设置到接口 0：0 在 DFU 模式下找到设备 打开设备 05ac：1227... irecv_copy_nonce_with_tag： 警告： 在字符串中找不到标签 NONC Apple Inc. irecv_copy_nonce_with_tag： 警告： 在字符串中找不到标签 SNON Apple Inc. 设置为配置 1 设置到接口 0：0 识别的设备为 iPhone2，1 从 IPSW 提取 BuildManifest 产品版本： 3.0 产品内部版本： 7A341 主要： 7 打开设备 05ac：1227... irecv_copy_nonce_with_tag：警告：在字符串中找不到标记NONC Apple Inc. irecv_copy_nonce_with_tag：警告：在字符串中找不到标记SNON Apple Inc. 设置为配置 1 设置接口 0：0 设备支持映像4：假变体：客户擦除安装（IPSW） 此恢复将擦除您的设备数据。获取设备的 TSS 请求的 ECID 找到 ECID 600390296460检查本地 shsh 使用缓存的 SHSH 使用缓存的文件系统从 'iPhone2，1_3.0_7A341_Custom/018-5302-002.dmg' 打开设备 05ac：1227... irecv_copy_nonce_with_tag： 警告： 在字符串中找不到标记 NONC Apple Inc. irecv_copy_nonce_with_tag： 警告： 在字符串中找不到标记 SNON Apple Inc. 找到具有 ECID 0000008bca0ce38c 的设备 设置为配置 1 设置到接口 0：0 调试： tss_response_get_path_by_entry： TSS 响应中没有条目"iBSS" 注意：TSS 中没有组件 iBSS 的路径， 将从build_identity提取 iBSS.n88ap.RELEASE.dfu...调试：tss_response_get_data_by_key：TSS 响应中没有条目"ApImg4Ticket" 调试：tss_response_get_blob_by_entry：TSS 响应中没有条目"iBSS"注意：未找到组件 iBSS 的 SHSH blob 未对组件 iBSS 进行个性化设置...正在将数据写入 iBSS 发送 iBSS（104772字节）... [================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================== irecv_copy_nonce_with_tag：警告：无法在字符串中找到标记NONC Apple Inc. irecv_copy_nonce_with_tag：警告：在字符串中找不到标记SNON Apple Inc.找到具有ECID 0000008bca0ce38c的设备设置为配置1设置到接口0：0错误：无法连接到恢复设备错误：无法将设备从DFU模式置于恢复模式
> 
> How can I fix this error? Thank you!
> 
> PS: I'm using an original Apple 30pin dock-USB cable.

Put your 3GS in Pwneddfu mode by using redsn0w 0.9.15b3

uf3n3qoasdnapnkem, 2022-04-06:
Put your 3GS in Pwneddfu mode by using redsn0w 0.9.15b3 and try again