libimobiledevice/idevicerestore #239: ios 12.1.1 stuck on Attempt to connect to restore mode device

ankitrwt, 2018-12-14:
Hello,

I am unable to restore iPhone with ios 12.1.1, Iphone stuck on apple logo and disconnects with computer

idevicerestore -e -d -l -C .

> Reading data from ./version.xml
> NOTE: using cached version data
> opening device 05ac:1281...
> Setting to configuration 1
> Setting to interface 0:0
> Found device in Recovery mode
> opening device 05ac:1281...
> Setting to configuration 1
> Setting to interface 0:0
> Identified device as n71map, iPhone8,1
> Latest firmware is iPhone_4.7_12.1.1_16C50_Restore.ipsw
> Verifying './iPhone_4.7_12.1.1_16C50_Restore.ipsw'...
> Checksum matches.
> Extracting BuildManifest from IPSW
> Product Version: 12.1.1
> Product Build: 16C50 Major: 16
> opening device 05ac:1281...
> Setting to configuration 1
> Setting to interface 0:0
> INFO: device serial number is DNPRR323GRYD
> Device supports Image4: true
> Variant: Customer Erase Install (IPSW)
> This restore will erase your device data.
> Using cached filesystem from './iPhone_4.7_12.1.1_16C50_Restore/048-40109-058.dmg'
> Getting device's ECID for TSS request
> Found ECID 6888458343383590
> Getting ApNonce in recovery mode... b1 c5 98 ae 89 40 3a ca 88 35 5e 4b eb 98 4c 0e cf cd 3f 5b 
> Trying to fetch new SHSH blob
> Getting SepNonce in recovery mode... f4 cc ba 8e 65 e9 6e 56 cd 59 94 58 97 64 2c 05 46 2c 35 9e 
> NOTE: Unable to find BbCalibrationManifestKeyHash node
> NOTE: Unable to find BbFactoryActivationManifestKeyHash node
> WARNING: Unable to find BbSkeyId node
> DEBUG: Applying restore request rules for entry AOP
> DEBUG: Adding action EPRO to TSS entry
> DEBUG: Adding action ESEC to TSS entry
> DEBUG: Applying restore request rules for entry AppleLogo
> DEBUG: Adding action EPRO to TSS entry
> DEBUG: Adding action ESEC to TSS entry
> DEBUG: Applying restore request rules for entry BatteryCharging0
> DEBUG: Adding action EPRO to TSS entry
> DEBUG: Adding action ESEC to TSS entry
> DEBUG: Applying restore request rules for entry BatteryCharging1
> DEBUG: Adding action EPRO to TSS entry
> DEBUG: Adding action ESEC to TSS entry
> DEBUG: Applying restore request rules for entry BatteryFull
> DEBUG: Adding action EPRO to TSS entry
> DEBUG: Adding action ESEC to TSS entry
> DEBUG: Applying restore request rules for entry BatteryLow0
> DEBUG: Adding action EPRO to TSS entry
> DEBUG: Adding action ESEC to TSS entry
> DEBUG: Applying restore request rules for entry BatteryLow1
> DEBUG: Adding action EPRO to TSS entry
> DEBUG: Adding action ESEC to TSS entry
> DEBUG: Applying restore request rules for entry BatteryPlugin
> DEBUG: Adding action EPRO to TSS entry
> DEBUG: Adding action ESEC to TSS entry
> DEBUG: Applying restore request rules for entry DeviceTree
> DEBUG: Adding action EPRO to TSS entry
> DEBUG: Adding action ESEC to TSS entry
> DEBUG: Applying restore request rules for entry KernelCache
> DEBUG: Adding action EPRO to TSS entry
> DEBUG: Adding action ESEC to TSS entry
> DEBUG: Applying restore request rules for entry LLB
> DEBUG: Adding action EPRO to TSS entry
[... 794 lines cut ...]
> Attempt 162 to connect to restore mode device...
> Attempt 163 to connect to restore mode device...
> Attempt 164 to connect to restore mode device...
> Attempt 165 to connect to restore mode device...
> Attempt 166 to connect to restore mode device...
> Attempt 167 to connect to restore mode device...
> Attempt 168 to connect to restore mode device...
> Attempt 169 to connect to restore mode device...
> Attempt 170 to connect to restore mode device...
> Attempt 171 to connect to restore mode device...
> Attempt 172 to connect to restore mode device...
> Attempt 173 to connect to restore mode device...
> Attempt 174 to connect to restore mode device...
> Attempt 175 to connect to restore mode device...
> Attempt 176 to connect to restore mode device...
> Attempt 177 to connect to restore mode device...
> Attempt 178 to connect to restore mode device...
> Attempt 179 to connect to restore mode device...
> Attempt 180 to connect to restore mode device...
> ERROR: Unable to connect to device in restore mode
> ERROR: Unable to open device in restore mode
> ERROR: Unable to restore device

Any help will be appreciated
Thanks

pubsocent, 2018-12-19:
I experienced the same issue.
I tried using Blefish's branch which is discussed here
[issue #234 ](https://github.com/libimobiledevice/idevicerestore/pull/234)
I was able to successfully restore a device using latest firmware 12.1.2 on linux using @Blefish branch
[Blefish idevicerestore](https://github.com/Blefish/idevicerestore)

nikias, 2019-01-18:
@ankitrwt because I didn't add the necessary fixes yet. It should work now with latest code.

nikias, 2019-01-26:
This issue should be resolved.

closed 2019-01-26 by nikias