libimobiledevice/idevicerestore #241: Waiting for device... ERROR: Unable to restore device Iphone 5s

NoDaa, 2018-12-20:
I have this problem
C:\WINDOWS\system32>cd C:\Users\snowd\Desktop\imbolidevice2

C:\Users\snowd\Desktop\imbolidevice2>idevicerestore.exe -u ae97df60243cafb31b2432e7bdc4f7c62cea36e9 -d -e C:\Users\snowd\Desktop\iPhone6,2_9.3_13E237_Restore.ipsw
Reading data from version.xml
NOTE: using cached version data
Setting to interface 0:0
WARNING: set interface failed, error -8
Found device in Recovery mode
Setting to interface 0:0
WARNING: set interface failed, error -8
Identified device as n53ap, iPhone6,2
Extracting BuildManifest from IPSW
Product Version: 9.3
Product Build: 13E237 Major: 13
Setting to interface 0:0
WARNING: set interface failed, error -8
INFO: device serial number is DNPMM9WLFFG9
Device supports Image4: true
Variant: Customer Erase Install (IPSW)
This restore will erase your device data.
Getting device's ECID for TSS request
Found ECID 8485926078000
Getting ApNonce in recovery mode... 7f de e3 a8 7a ae f1 29 58 3c b4 c7 26 e4 76 2a 25 ce 4d 52
Trying to fetch new SHSH blob
Getting SepNonce in recovery mode... f3 88 a1 31 bf 52 6c a0 1f b8 cd f2 3e fb 42 62 39 0c 63 74
NOTE: Unable to find BbCalibrationManifestKeyHash node
NOTE: Unable to find BbFactoryActivationManifestKeyHash node
NOTE: Unable to find BbFDRSecurityKeyHash node
WARNING: Unable to find BbSkeyId node
DEBUG: Applying restore request rules for entry AppleLogo
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry BatteryCharging0
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry BatteryCharging1
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry BatteryFull
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry BatteryLow0
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry BatteryLow1
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry BatteryPlugin
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry DeviceTree
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry KernelCache
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry LLB
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry RecoveryMode
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry RestoreDeviceTree
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry RestoreKernelCache
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry RestoreLogo
[... 919 lines cut ...]
Extracting 048-40310-062.dmg...
Personalizing IMG4 component RestoreRamDisk...
Sending RestoreRamDisk (106782446 bytes)...
DEBUG: tss_response_get_path_by_entry: No entry 'RestoreDeviceTree' in TSS response
NOTE: No path for component RestoreDeviceTree in TSS, will fetch from build_identity
Extracting DeviceTree.n53ap.im4p...
Personalizing IMG4 component RestoreDeviceTree...
Sending RestoreDeviceTree (108940 bytes)...
DEBUG: tss_response_get_path_by_entry: No entry 'RestoreKernelCache' in TSS response
NOTE: No path for component RestoreKernelCache in TSS, will fetch from build_identity
Extracting kernelcache.release.iphone6...
Personalizing IMG4 component RestoreKernelCache...
Sending RestoreKernelCache (13256542 bytes)...
About to restore device...
Waiting for device...
Attempt 1 to connect to restore mode device...
...
Attempt 180 to connect to restore mode device...
ERROR: Unable to connect to device in restore mode
ERROR: Unable to open device in restore mode
ERROR: Unable to restore device

C:\Users\snowd\Desktop\imbolidevice2>

And I try again again, help pls

NoDaa, 2018-12-20:
I try with Firmware IOS 12.1.2.1 too and have the same problem

taanders, 2019-01-17:
Hi NoDaa,

Have you tried with latest codes pulled from master? There are maybe fixes added also for this kind of error.

Have you used this patch to libirecovery and does it seem to work ok?:
https://github.com/libimobiledevice/libirecovery/commit/96b0c2eb571498749c9e8b926c02559f6956e273

nikias, 2019-01-18:
@NoDaa please try again with latest fixes.