libimobiledevice/idevicerestore #243: Face ID  is not available on iPhone XS/XSMax after update to 12.1.3

jduranc, 2019-01-23:
Hi,

Xs models are failing to activate Face ID after updating 12.1.3, estore process completes success and no error messages appears on logs, but feature its not available. Trying restore using iTunes enable feature again.  Someone is facing same problem?

regards

nikias, 2019-01-24:
Since 12.1.2 is still signed, can you check if faceID works when restoring that version? Are you using master branch? In #234 there is a commit with your changes that I didn't merge completely since it seemed to have worked without it, but I doubt this has anything to do with FaceID firmware. I will check the buildmanifest for changes.

pekn, 2019-01-24:
This is reproducible with iPhone XR even using 12.1.2. idevicerestore succeeds (tested with blefish version of idevicerestore), but face id is not available.

nikias, 2019-01-24:
@pekn ok thanks. I don't have this device generation at hand currently so I can't test myself. I had a similar issue with the X when it came out; the firmware was sent to the device and no error was reported, just faceID not working then. I fixed that, but for Xs/r (probably new iPad Pro too) it seems different again...

pekn, 2019-01-24:
Tried with iPhone XS with latest idevicerestore and original problem didn't reproduce.

Blefish, 2019-01-24:
How to reproduce this?

I have an XS max that is on 12.1.3 beta 4 (flashed with my branch). I now updated it to 12.1.3 using master branch and Face ID is working.

nikias, 2019-01-24:
@pekn very well observed. So somehow we need to figure out when to select what component. Weirdly enough both firmware archives seem to ship with multiple images instead of just one. Isn't the device requesting a specific one in the `MessageArgInfo` dict of the `FirmwareDataRequest`?

pekn, 2019-01-24:
Here's full arguments:
`<plist version="1.0">
<dict>
	<key>MessageArgInfo</key>
	<dict>
		<key>Savage,AllowOfflineBoot</key>
		<true/>
		<key>Savage,Nonce</key>
		<data>
		f4+zlz6+nBt60Uv+Ty9ESQ==
		</data>
		<key>Savage,ProductionMode</key>
		<true/>
		<key>Savage,ReadECKey</key>
		<true/>
		<key>Savage,ReadFWKey</key>
		<true/>
		<key>Savage,Revision</key>
		<data>
		rAs=
		</data>
		<key>Savage,UID</key>
		<data>
		R//b0PmsmpHWNVKvqMh58w==
		</data>
	</dict>
	<key>MessageArgType</key>
	<string>FirmwareResponseData</string>
	<key>MessageArgUpdaterLoopCount</key>
	<integer>0</integer>
	<key>MessageArgUpdaterName</key>
	<string>Savage</string>
</dict>
</plist>`

nikias, 2019-01-24:
Yep, got it. It's derived from Savage,Revision. Working on a fix.

nikias, 2019-01-24:
Alright, can you test this: https://gist.github.com/nikias/c01c396b660260d137c15262ebc3d0b1 ?
Also, there's a revision check for Yonkers. I wonder how this still works though. It's derived from `Yonkers,FabRevision` @pekn could you maybe get a sample of the `MessageArgInfo` for the Yonkers request too? Thanks ;)

nikias, 2019-01-24:
@pekn nevermind, Yonkers stuff is already handled correctly in the code thanks to @Blefish. Interestingly the code to select it from the build manifest is way better than the hardcoded comparisons in Apple's code IMO ;)

pekn, 2019-01-24:
Tested above patch with iPhone XR and XS. Works OK.

nikias, 2019-01-24:
Great! I will improve this a bit to eliminate code duplication and push a fix to master. Thanks for testing.

nikias, 2019-01-25:
I cleaned up a bit and also moved the Yonkers component selection code to tss, like I did for Savage. If you can, could you test again with latest code (commit 917f2ae4e3e6245a81e4b853ade240b7644f69f3)? Thanks.

pekn, 2019-01-25:
Tested with latest idevicerestore and seems to work OK.

nikias, 2019-01-25:
@pekn thanks. @jduranc I will close this ticket since the problem seems to be resolved.

closed 2019-01-25 by nikias