libimobiledevice/idevicerestore #244: iphone 7 plus crash when restore to ios 12.1.3

jackczhang, 2019-01-25:
iphone 7 plus crash when restore to ios 12.1.3

nikias, 2019-01-25:
The phone crashes? Or idevicrestore? More details please...

a-gunter, 2019-01-25:
No issues here restoring an iPhone 7+ with the latest commit

jackczhang, 2019-01-28:
I used windows msys64 to compile the latest commit.
When run command "idevicerestore.exe -e iPhone_5.5_P3_12.1.3_16D39_Restore.ipsw", the idevicerestore.exe crashed.

nikias, 2019-01-28:
@jackczhang can you provide any more info? any chance you could run it with gdb:
```gdb --args idevicerestore.exe -e iPhone_5.5_P3_12.1.3_16D39_Restore.ipsw```

jackczhang, 2019-01-28:
run it with gdb --

free(type) function begin.
warning: HEAP[idevicerestore.exe]:
warning: Invalid address specified to RtlFreeHeap( 00640000, 00528828 )

Thread 1 received signal SIGTRAP, Trace/breakpoint trap.
0x774597c6 in ntdll!RtlpNtSetValueKey () from C:\Windows\SYSTEM32\ntdll.dll
(gdb)

nikias, 2019-01-28:
Hmm that's weird, why would it crash there... if it reaches that point, `type` should have a valid address, otherwise it would have crashed at the above `strcmp()` already. What if you change the code to this:
```
	printf("type is %s (%p)\n", type, type);
	if (strcmp(type, "com.apple.mobile.lockdown") != 0) {
			free(type);
			type = NULL;
			continue;
		}
		free(type);
```

jackczhang, 2019-01-28:
So strange.

type is com.apple.mobile.lockdown (00CF8828)
free(type) function begin.
warning: HEAP[idevicerestore.exe]:
warning: Invalid address specified to RtlFreeHeap( 00E90000, 00CF8828 )

Thread 1 received signal SIGTRAP, Trace/breakpoint trap.
0x774597c6 in ntdll!RtlpNtSetValueKey () from C:\Windows\SYSTEM32\ntdll.dll

jackczhang, 2019-01-28:
@nikias I used the source code of Commits on Jan 16, 2019 (661efc8f3a724fdd38174f3a886fd3aeb26fde37) to re-compile the application "idevicerestore.exe", the crahed disappeared. I guessed there were somethings wrong during the period.

jackczhang, 2019-01-28:
Caught it,  commit e0b44cfb99c7e400c0fc51474f240dba3facd412 on  Jan 22, 2019
mingw*)
			GLOBAL_CFLAGS+="-DWIN32 -D__LITTLE_ENDIAN__=1"
			AC_LDFLAGS+="-static-libgcc"
			AC_LDADD="-lsetupapi"
			AC_LDFLAGS+="-static-libgcc -lmsvcr100"
		;;
		*)

-lmsvcr100 makes free() function crash.

nikias, 2019-01-28:
Thanks for tracking this down. I had to add this to get it working with _fseeki64 etc. on msys (mingw32). I will see if I can autodetect if msvcr100 is required or not.

nikias, 2019-01-28:
I think I know why it happens. Inside of libimobiledevice the string is allocated, using malloc from msvcrt.dll, and in idevicerestore the string is freed using free from msvcr100.dll - these two don't seem to match up and that's probably why the error occurs.

nikias, 2019-02-02:
@jackczhang can you try with my latest commit? 8aa4246ddef30c0e114362b6ef2a8e6ee0ed3817
That should fix the issue (no more msvcr100.dll!)

jackczhang, 2019-04-10:
Yes, thanks @nikias , already fixed.

closed 2019-04-10 by jackczhang