libimobiledevice/idevicerestore #248: Targeting normal mode device with -u [UDID] fails if a recovery/DFU mode device is also plugged in

a-gunter, 2019-02-12:
I have 2 devices plugged in: one in normal mode, and one in recovery mode. I want to restore the device that's in normal mode, so I execute the command "idevicerestore -e -l -u [UDID]". The result is "Found device in recovery mode" .. etc, and the restore ends up targeting the device in recovery mode. This result also occurs whenever you replace the recovery mode device with a DFU device. The result will be "Found device in DFU mode..." and will continue restoring the device in DFU mode instead of the normal mode device.

I figure this is because recovery_check_mode returns 0 as long as there's a recovery mode device present, and recovery_check_mode is called before normal_check_mode. The same thing probably applies to dfu_check_mode.

As a fix, I went ahead and added a tiny bit of logic in both recovery_check_mode (recovery.c) and dfu_check_mode (dfu.c) just to check if a UDID was already attached to the client. If a UDID is supplied, I just return -1 early on, because I don't want to search for devices in recovery/DFU mode when I supply a UDID. I then retried my scenario, and everything seemed to work as intended.

Somebody may come up with a more elegant solution or permanent fix, but I just wanted to throw the issue out there in case anybody else experienced the same thing.

nikias, 2019-02-13:
Good observation. I will think if this is the right approach and commit a fix. It seems about right what you did, since if you pass a UDID you can't start from recovery or DFU mode.

nikias, 2019-02-13:
The only reliable way - regardless of the mode - is to always use the ECID.

nikias, 2019-09-11:
Fixed with commit fbaafdaf87cc01197ad86812a46cf17b7b392b8e. If no ecid is given but a udid, don't discover recovery mode / dfu mode devices.

closed 2019-09-11 by nikias