libimobiledevice/idevicerestore #250: Downgrade problem

EricLenSher, 2019-03-02:
Hey everyone

I'm quite new here and trying to restore my Phone (iphone 6) from IOS 12.1.1b3 to 11.4.1 using SHSH2 and when i'm doing it.I came across this issue

> **Personalizing IMG4 component iBEC...
> Sending iBEC (714933 bytes)...
> ERROR: Unable to send iBEC component: Unable to find device
> ERROR: Unable to send iBEC to device.
> [Error] ERROR: Unable to send iBEC
> 
> To exit recovery mode, use --exit-recovery
> 
> Done: restoring failed.
> Failed with errorcode=-8**
And i don't know how to fix it.Can someone help me?

julienkrijnen8, 2019-03-03:
Hello, do you have the IPSW?

Try : idevicerestore -e [youripsw].IPSW

EricLenSher, 2019-03-03:
can you say it more specifically? I don't know where to put that command. this is the command line I used
`futurerestore.exe -t "C:\Users\Eric\Downloads\Documents\New folder\3962820712680890_iPhone7,2_n61ap_11.4.1-15G77_860d30ca314b7f596abdb14d6b4f86957757b432.shsh2" --latest-sep --latest-baseband "C:\Users\Eric\Downloads\Documents\New folder\iPhone_4.7_11.4.1_15G77_Restore.ipsw"`

nikias, 2019-03-03:
@EricLenSher can you please post futurerestore related issues here: https://github.com/tihmstar/futurerestore/issues

closed 2019-03-03 by nikias