libimobiledevice/idevicerestore #255: Fail to restore iOS 12.2 iPhone 8 Plus

ChrisCheney, 2019-03-29:
Using current git snapshot as of today for all components I get the following when trying to restore:

# idevicerestore -e iPhone_5.5_P3_12.2_16E227_Restore.ipsw 
Found device in DFU mode
Identified device as d21ap, iPhone10,2
Extracting BuildManifest from IPSW
Product Version: 12.2
Product Build: 16E227 Major: 16
Device supports Image4: true
Variant: Customer Erase Install (IPSW)
This restore will erase your device data.
Checking IPSW for required components...
All required components found in IPSW
Using cached filesystem from 'iPhone_5.5_P3_12.2_16E227_Restore/048-23422-264.dmg'
Found ECID (redacted)
Getting ApNonce in dfu mode...  (redacted) 
Trying to fetch new SHSH blob
Getting SepNonce in dfu mode... (redacted)
WARNING: Unable to find BbSkeyId node
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received SHSH blobs
Extracting iBSS.d21.RELEASE.im4p...
Personalizing IMG4 component iBSS...
Sending iBSS (970153 bytes)...
[==================================================] 100.0%
Nonce: (redacted)
Extracting iBEC.d21.RELEASE.im4p...
Personalizing IMG4 component iBEC...
Sending iBEC (970153 bytes)...
[==================================================] 100.0%
INFO: device serial number is (redacted)
Recovery Mode Environment:
iBoot build-version=iBoot-4513.250.288
iBoot build-style=RELEASE
Sending RestoreLogo...
Extracting applelogo@3x~iphone.im4p...
Personalizing IMG4 component RestoreLogo...
Sending RestoreLogo (19932 bytes)...
Extracting 048-23877-271.dmg.trustcache...
Personalizing IMG4 component RestoreTrustCache...
Sending RestoreTrustCache (10561 bytes)...
ramdisk-size=0x20000000
Extracting 048-23877-271.dmg...
Personalizing IMG4 component RestoreRamDisk...
Sending RestoreRamDisk (90977339 bytes)...
Extracting DeviceTree.d21ap.im4p...
Personalizing IMG4 component RestoreDeviceTree...
Sending RestoreDeviceTree (178463 bytes)...
Extracting kernelcache.release.iphone10...
Personalizing IMG4 component RestoreKernelCache...
Sending RestoreKernelCache (17744058 bytes)...
About to restore device... 
Waiting for device...
Device ffffffffffffffffffffffffffffffff00000001 is now connected in restore mode...
Connecting now...
Connected to com.apple.mobile.restored, version 15
Device ffffffffffffffffffffffffffffffff00000001 has successfully entered restore mode
Hardware Information:
BoardID: 4
ChipID: 32789
UniqueChipID: (redacted)
ProductionMode: true
Starting FDR listener thread
About to send NORData...
Found firmware path Firmware/all_flash
Getting firmware manifest from build identity
Extracting LLB.d21.RELEASE.im4p...
Personalizing IMG4 component LLB...
Extracting applelogo@3x~iphone.im4p...
Personalizing IMG4 component AppleLogo...
Extracting batterycharging0@3x~iphone.im4p...
Personalizing IMG4 component BatteryCharging0...
Extracting batterycharging1@3x~iphone.im4p...
Personalizing IMG4 component BatteryCharging1...
Extracting batteryfull@3x~iphone.im4p...
Personalizing IMG4 component BatteryFull...
Extracting batterylow0@3x~iphone.im4p...
Personalizing IMG4 component BatteryLow0...
Extracting batterylow1@3x~iphone.im4p...
Personalizing IMG4 component BatteryLow1...
Extracting glyphplugin@1920~iphone-lightning.im4p...
Personalizing IMG4 component BatteryPlugin...
Extracting DeviceTree.d21ap.im4p...
Personalizing IMG4 component DeviceTree...
Extracting liquiddetect@1920~iphone-lightning.im4p...
Personalizing IMG4 component Liquid...
Extracting recoverymode@1920~iphone-lightning.im4p...
Personalizing IMG4 component RecoveryMode...
Extracting iBoot.d21.RELEASE.im4p...
Personalizing IMG4 component iBoot...
Extracting sep-firmware.d21.RELEASE.im4p...
Personalizing IMG4 component RestoreSEP...
Extracting sep-firmware.d21.RELEASE.im4p...
Personalizing IMG4 component SEP...
Sending NORData now...
Done sending NORData
About to send RootTicket...
Sending RootTicket now...
Done sending RootTicket
Waiting for NAND (28)
Updating S3E Firmware (58)
Checking filesystems (15)

Sits for a few seconds the following shows up:

ERROR: Unable to receive message from FDR 0x558fa99fa540 (-4). 0/2 bytes

There it appears to hang forever, with the Apple on the screen with an empty gauge.

ChrisCheney, 2019-03-29:
If I try to do a regular recovery restore it fails slightly later:

...
Updating S3E Firmware (58)
Checking filesystems (15)
Checking filesystems (15)
About to send FDR Trust data...
Sending FDR Trust data now...
Done sending FDR Trust Data
Unmounting filesystems (29)
Unmounting filesystems (29)
Unmounting filesystems (29)
Unmounting filesystems (29)
Creating partition map (11)
Creating filesystem (12)
About to send filesystem...
Connected to ASR
Validating the filesystem
Filesystem validated
Sending filesystem now...
[=                                                 ]   1.0%ERROR: Unable to receive message from FDR 0x564689b786c0 (-4). 0/2 bytes
ERROR: FDR 0x7f6354001700 Unable to receive proxy payload (-4)
ERROR: Unable to send data to ASR. Sent 0 of 1450 bytes.
ERROR: Unable to send filesystem payload
ERROR: Unable to send payload to ASR
ERROR: Unable to send filesystem
ERROR: Unable to successfully restore device
ERROR: Unable to restore device

ChrisCheney, 2019-03-29:
Also note that restoring the phone using the same ipsw file, lightning cable, and phone works fine under Windows 10 with iTunes 12.9.4.

Blefish, 2019-03-29:
I've seen this too, but not constantly. It's hard to track down why it happens.

yogendra123456, 2020-06-02:
hi @ChrisCheney and @Blefish,

Can you provide me libmobiledevice code which is supported to window and update the firmware on iPhone device any suggestion how to restore device using libmobiledevice idevicerestore.exe ...on window system.....
[Debug.txt](https://github.com/libimobiledevice/idevicerestore/files/4714937/Debug.txt)
[Only debug.txt](https://github.com/libimobiledevice/idevicerestore/files/4714938/Only.debug.txt)

thanks in advanced...