libimobiledevice/idevicerestore #256: Logo delays on appear after restore.

jduranc, 2019-04-08:
Hi,

I'm facing some weird situation, after restore device, when device power on, the apple logo delays about 6 seconds after screen turns on, compared with another device restored using iTunes the other device show apple logo as soon as screen turns on. This is more noticeable on white backcover devices that screens turns white and apple logo its black. Tested on iPhone X, iPhone 8 and iPhone 8P with same issue.

Any help it's appreciated.

closed 2019-04-08 by jduranc

nikias, 2019-04-10:
I wouldn't know why this would be caused by a restore. This is purely device behavior; if something is wrong during the restore the device would abort anyway.

jduranc, 2019-04-10:
Hi Nikias,

On img4_stitch_component function two components related to logo are received, "RestoreLogo" and "AppleLogo", my bad was treat both as "rlgo", apparently using "AppleLogo" it's causing the initial logo delays on appears. 

regards,

nikias, 2019-04-21:
@jduranc is there any change required to the code in img4_stitch_component?

jduranc, 2019-04-22:
Hi @nikias 
No changes required, the code works fine, I just remove some changes on my side.

regards,