libimobiledevice/idevicerestore #260: Question about custom ipsw

armintelker, 2019-05-24:
Hey :) 
I try to restore a device from an already downloaded .ipsw file.

As example:
I download the .ipsw file that I found via the ipsw.me API 
```
https://api.ipsw.me/v2.1/device/iPhone5,4

http://appldnld.apple.com/ios10.3.3/091-23384-20170719-CA966D80-6977-11E7-9F96-3E9100BA0AE3/iPhone_4.0_32bit_10.3.3_14G60_Restore.ipsw
```
I try to execute following command
```
 idevicerestore -i 0x002E91A0E5DC9 -c ~/Desktop/iPhone_4.0_32bit_10.3.3_14G60_Restore.ipsw  
```

And I get this output:
```
Found device in Normal mode
Identified device as n49ap, iPhone5,4
Extracting Restore.plist from IPSW
Product Version: 10.3.3
Product Build: 14G60 Major: 14
Device supports Image4: false
Custom firmware requested. Disabled TSS request.
ERROR: could not extract Firmware/all_flash/all_flash.n49ap.production/manifest from IPSW
```

Is there some additional step to do so idevicerestore understands how to extract the ipsw file? Or do I missed some documentation that already explains the "HowTo" 

Thank you very much :)

allbombson, 2019-06-05:
lmao -c is for jailbreaks on old ass devices like the iphone 3g, run the same command without -c and it should work or just use -l and no ipsw and it will download latest for you

nikias, 2019-06-05:
@allbombson not everyone is aware of that fact.
@armintelker you can only restore what is currently being signed by Apple.

closed 2019-06-05 by armintelker