libimobiledevice/idevicerestore #262: Could not restore  iphone XS

decembersake, 2019-05-31:
Hello
I am having trouble restoring my iphone which is now in recovery mode loop.
Any advice to solve this problem?
I know that there's not enough space on the device but I do not have any backup and desperate to recover all photos, videos, record and etc.
Please help to solve this error. 
(If there's any way to mount recovery mode iphone as memory device in Mac OS, please let me know the command.)

ramrod_display_set_granular_progress_forced: 31.000000
ASR RESTORE PROGRESS: 92%
ramrod_display_set_granular_progress_forced: 31.000000
ASR RESTORE PROGRESS: 94%
ramrod_display_set_granular_progress_forced: 32.000000
ASR STATUS: fail
ASR: Could not restore - No space left on device
ASR STATUS: fail
restore_apfs_image : failed to restore an APFS image, error = 14
[12:46:10.0690-GMT]{3>7} CHECKPOINT FAILURE:(FAILURE:14) RESTORED:[0x065B] asr_and_invert_image [0]D(failed to restore APFS image)
restore-step-results = {0x1107065B:{0:14}}
restore-step-codes = {0x1107065B:{0:14}}
restore-step-domains = {0x1107065B:{0:"AMRestoreErrorDomain"}}
restore-step-error = {0x1107065B:"[0]D(failed to restore APFS image)"}
restore-step-uptime = 140
restore-step-user-progress = 32
[12:46:10.0695-GMT]{3>7} CHECKPOINT NOTICE: (NVRAM set) restore-step-user-progress=32 [sync=true] (first failure)
[12:46:10.0695-GMT]{3>7} CHECKPOINT BEGIN: RESTORED:[0x0647] cleanup_check_result
restore-step-ids = {0x1103065B:47;0x11030647:48}
restore-step-names = {0x1103065B:asr_and_invert_image;0x11030647:cleanup_check_result}
restore-step-uptime = 140
restore-step-user-progress = 32
[12:46:10.0698-GMT]{3>7} CHECKPOINT END: RESTORED:[0x0647] cleanup_check_result
restore-step-ids = {0x1103065B:47}
restore-step-names = {0x1103065B:asr_and_invert_image}
restore-step-uptime = 140
restore-step-user-progress = 32
[12:46:10.0702-GMT]{3>7} CHECKPOINT BEGIN: RESTORED:[0x0648] cleanup_send_final_status
restore-step-ids = {0x1103065B:47;0x11030648:49}
restore-step-names = {0x1103065B:asr_and_invert_image;0x11030648:cleanup_send_final_status}
restore-step-uptime = 140
restore-step-user-progress = 32

ERROR: Unable to restore device

decembersake, 2019-05-31:
`Found device in Recovery mode
Identified device as d321ap, iPhone11,2
The following firmwares are currently being signed for iPhone11,2:
  [1] 12.3.1 (build 16F203)
  [2] 12.3 (build 16F156)
Select the firmware you want to restore: 2
Selected firmware 12.3 (build 16F156)
Downloading firmware (http://updates-http.cdn-apple.com/2019SpringFCS/fullrestores/041-61681/B8D99728-71E2-11E9-B014-F483412A1E9F/iPhone11,2_12.3_16F156_Restore.ipsw)
downloading: 1%
downloading: 2%
downloading: 3%
downloading: 4%
downloading: 5%
downloading: 6%
downloading: 7%
downloading: 8%
downloading: 9%
downloading: 10%
downloading: 11%
downloading: 12%
downloading: 13%
downloading: 14%
downloading: 15%
downloading: 16%
downloading: 17%
downloading: 18%
downloading: 19%
downloading: 20%
downloading: 21%
downloading: 22%
downloading: 23%
downloading: 24%
downloading: 25%
downloading: 26%
downloading: 27%
downloading: 28%
downloading: 29%
downloading: 30%
downloading: 31%
downloading: 32%
downloading: 33%
downloading: 34%
downloading: 35%
downloading: 36%
downloading: 37%
downloading: 38%
downloading: 39%
downloading: 40%
downloading: 41%
downloading: 42%
downloading: 43%
downloading: 44%
downloading: 45%
downloading: 46%
downloading: 47%
downloading: 48%
downloading: 49%
downloading: 50%
downloading: 51%
downloading: 52%
downloading: 53%
downloading: 54%
downloading: 55%
downloading: 56%
downloading: 57%
downloading: 58%
downloading: 59%
downloading: 60%
downloading: 61%
downloading: 62%
[... 1483 lines cut ...]
restore-step-results = {0x1107065B:{0:14}}
restore-step-codes = {0x1107065B:{0:14}}
restore-step-domains = {0x1107065B:{0:"AMRestoreErrorDomain"}}
restore-step-error = {0x1107065B:"[0]D(failed to restore APFS image)"}
restore-step-uptime = 140
restore-step-user-progress = 32
[12:46:10.0695-GMT]{3>7} CHECKPOINT NOTICE: (NVRAM set) restore-step-user-progress=32 [sync=true] (first failure)
[12:46:10.0695-GMT]{3>7} CHECKPOINT BEGIN: RESTORED:[0x0647] cleanup_check_result
restore-step-ids = {0x1103065B:47;0x11030647:48}
restore-step-names = {0x1103065B:asr_and_invert_image;0x11030647:cleanup_check_result}
restore-step-uptime = 140
restore-step-user-progress = 32
[12:46:10.0698-GMT]{3>7} CHECKPOINT END: RESTORED:[0x0647] cleanup_check_result
restore-step-ids = {0x1103065B:47}
restore-step-names = {0x1103065B:asr_and_invert_image}
restore-step-uptime = 140
restore-step-user-progress = 32
[12:46:10.0702-GMT]{3>7} CHECKPOINT BEGIN: RESTORED:[0x0648] cleanup_send_final_status
restore-step-ids = {0x1103065B:47;0x11030648:49}
restore-step-names = {0x1103065B:asr_and_invert_image;0x11030648:cleanup_send_final_status}
restore-step-uptime = 140
restore-step-user-progress = 32

ERROR: Unable to restore device
`

nikias, 2019-05-31:
This is a strange problem... it should not resize the root filesystem to something else. What is the storage size of this iPhone?
Also, there is no way to "mount" an iPhone in recovery mode.

decembersake, 2019-05-31:
It's 256GB .. Is there's no way to recover my phone from recovery mode please save me
I work in design part and really do need all data I'm going to die please help me.. any advice to try out

decembersake, 2019-05-31:
> This is a strange problem... it ( = Apple's restore code that runs on the device) should not resize the root filesystem to something else. What is the storage size of this iPhone?
> Also, there is no way to "mount" an iPhone in recovery mode.

It's a 256GB .. Is there's no way to recover my phone from recovery mode please save me
I work in the design part and really do need all the data I'm going to die please help me.. any advice to try out

nikias, 2019-05-31:
Are you sure you are using the latest code? It should copy the system partition padding from the BuildManifest. That specifies how large the root filesystem should be. Maybe you can try this patch too:
```
diff --git a/src/restore.c b/src/restore.c
index 4e6b91d..1cd9e46 100644
--- a/src/restore.c
+++ b/src/restore.c
@@ -2417,11 +2417,16 @@ int restore_device(struct idevicerestore_client_t* client, plist_t build_identit
                spp = plist_copy(spp);
        } else {
                spp = plist_new_dict();
+               plist_dict_set_item(spp, "1024", plist_new_uint(1280));
                plist_dict_set_item(spp, "128", plist_new_uint(1280));
                plist_dict_set_item(spp, "16", plist_new_uint(160));
+               plist_dict_set_item(spp, "256", plist_new_uint(1280));
                plist_dict_set_item(spp, "32", plist_new_uint(320));
+               plist_dict_set_item(spp, "512", plist_new_uint(1280));
                plist_dict_set_item(spp, "64", plist_new_uint(640));
+               plist_dict_set_item(spp, "768", plist_new_uint(1280));
                plist_dict_set_item(spp, "8", plist_new_uint(80));
+
        }
        plist_dict_set_item(opts, "SystemPartitionPadding", spp);
        char* guid = generate_guid();
```

nikias, 2019-06-01:
Also have you tried with 12.3.1 ?

decembersake, 2019-06-02:
Yes I am using the latest code. I have tried with 12.3.1 and 12.3 both are not working.. I have tried your new patch after committing but still not working.. still same issue and stop exactly at the same point.

`ASR RESTORE PROGRESS: 94%
ramrod_display_set_granular_progress_forced: 32.000000
ASR STATUS: fail
ASR: Could not restore - No space left on device
ASR STATUS: fail
restore_apfs_image : failed to restore an APFS image, error = 14
[10:04:47.0307-GMT]{3>7} CHECKPOINT FAILURE:(FAILURE:14) RESTORED:[0x065B] asr_and_invert_image [0]D(failed to restore APFS image)
restore-step-results = {0x1107065B:{0:14}}
restore-step-codes = {0x1107065B:{0:14}}
restore-step-domains = {0x1107065B:{0:"AMRestoreErrorDomain"}}
restore-step-error = {0x1107065B:"[0]D(failed to restore APFS image)"}
restore-step-uptime = 141
restore-step-user-progress = 32
[10:04:47.0311-GMT]{3>7} CHECKPOINT NOTICE: (NVRAM set) restore-step-user-progress=32 [sync=true] (first failure)
[10:04:47.0311-GMT]{3>7} CHECKPOINT BEGIN: RESTORED:[0x0647] cleanup_check_result
restore-step-ids = {0x1103065B:47;0x11030647:48}
restore-step-names = {0x1103065B:asr_and_invert_image;0x11030647:cleanup_check_result}
restore-step-uptime = 141
restore-step-user-progress = 32
[10:04:47.0315-GMT]{3>7} CHECKPOINT END: RESTORED:[0x0647] cleanup_check_result
restore-step-ids = {0x1103065B:47}
restore-step-names = {0x1103065B:asr_and_invert_image}
restore-step-uptime = 141
restore-step-user-progress = 32
[10:04:47.0318-GMT]{3>7} CHECKPOINT BEGIN: RESTORED:[0x0648] cleanup_send_final_status
restore-step-ids = {0x1103065B:47;0x11030648:49}
restore-step-names = {0x1103065B:asr_and_invert_image;0x11030648:cleanup_send_final_status}
restore-step-uptime = 141
restore-step-user-progress = 32

ERROR: Unable to restore device
`

decembersake, 2019-06-02:
I really do appreciate your help Thank you SO SO much but Is there's literally NO way to recover my photos? This is so bad and can't believe.. I really need all data it's just insane that phone dies because of no space in 2019... could you please suggest any other things to try, please...

decembersake, 2019-06-02:
i dont know if it helps but here's full log

Password:
Found device in Recovery mode
Identified device as d321ap, iPhone11,2
The following firmwares are currently being signed for iPhone11,2:
  [1] 12.3.1 (build 16F203)
  [2] 12.3 (build 16F156)
Select the firmware you want to restore: 1
Selected firmware 12.3.1 (build 16F203)

Verifying 'iPhone11,2_12.3.1_16F203_Restore.ipsw'...
Checksum matches.
Extracting BuildManifest from IPSW
Product Version: 12.3.1
Product Build: 16F203 Major: 16
INFO: device serial number is DNPX9193KPG4
Device supports Image4: true
Variant: Customer Upgrade Install (IPSW)
This restore will update your device without losing data.
Checking IPSW for required components...
All required components found in IPSW
Extracting filesystem from IPSW: 048-43159-182.dmg
[==================================================] 100.0%
Found ECID 4909083227848750
Getting ApNonce in recovery mode... c1 f3 30 06 41 85 31 a0 22 7d 1a 26 91 09 65 02 3b 61 c0 c6 72 cc 49 c2 19 af 5c 58 b8 21 a3 70 
Trying to fetch new SHSH blob
Getting SepNonce in recovery mode... 8a 5a 37 d2 29 77 4f c4 7a 4d c2 70 e7 e0 95 6f 3e 67 7a fd 
WARNING: Unable to find BbSkeyId node
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received SHSH blobs
Extracting iBEC.d321.RELEASE.im4p...
Personalizing IMG4 component iBEC...
Sending iBEC (1063306 bytes)...
Recovery Mode Environment:
iBoot build-version=iBoot-4513.260.81
iBoot build-style=RELEASE
Sending RestoreLogo...
Extracting applelogo@3x~iphone.im4p...
Personalizing IMG4 component RestoreLogo...
Sending RestoreLogo (20910 bytes)...
Extracting h11_ane_fw_quin.im4p...
Personalizing IMG4 component ANE...
Sending ANE (9394706 bytes)...
Extracting aopfw-iphone11aop.im4p...
Personalizing IMG4 component AOP...
Sending AOP (1396234 bytes)...
Extracting AppleAVE2FW.im4p...
Personalizing IMG4 component AVE...
Sending AVE (1111370 bytes)...
Extracting armfw_g11p.im4p...
Personalizing IMG4 component GFX...
Sending GFX (1555834 bytes)...
Extracting adc-petra-d3x.im4p...
Personalizing IMG4 component ISP...
Sending ISP (14639242 bytes)...
Extracting 048-43107-184.dmg.trustcache...
Personalizing IMG4 component RestoreTrustCache...
Sending RestoreTrustCache (11671 bytes)...
Extracting SmartIOFirmwareT8020.im4p...
Personalizing IMG4 component SIO...
Sending SIO (1356094 bytes)...
Extracting WirelessPower.iphone11.im4p...
Personalizing IMG4 component WCHFirmwareUpdater...
Sending WCHFirmwareUpdater (28773 bytes)...
ramdisk-size=0x20000000
Extracting 048-43107-184.dmg...
Personalizing IMG4 component RestoreRamDisk...
Sending RestoreRamDisk (93872653 bytes)...
[... 1385 lines cut ...]
[10:04:47.0307-GMT]{3>7} CHECKPOINT FAILURE:(FAILURE:14) RESTORED:[0x065B] asr_and_invert_image [0]D(failed to restore APFS image)
restore-step-results = {0x1107065B:{0:14}}
restore-step-codes = {0x1107065B:{0:14}}
restore-step-domains = {0x1107065B:{0:"AMRestoreErrorDomain"}}
restore-step-error = {0x1107065B:"[0]D(failed to restore APFS image)"}
restore-step-uptime = 141
restore-step-user-progress = 32
[10:04:47.0311-GMT]{3>7} CHECKPOINT NOTICE: (NVRAM set) restore-step-user-progress=32 [sync=true] (first failure)
[10:04:47.0311-GMT]{3>7} CHECKPOINT BEGIN: RESTORED:[0x0647] cleanup_check_result
restore-step-ids = {0x1103065B:47;0x11030647:48}
restore-step-names = {0x1103065B:asr_and_invert_image;0x11030647:cleanup_check_result}
restore-step-uptime = 141
restore-step-user-progress = 32
[10:04:47.0315-GMT]{3>7} CHECKPOINT END: RESTORED:[0x0647] cleanup_check_result
restore-step-ids = {0x1103065B:47}
restore-step-names = {0x1103065B:asr_and_invert_image}
restore-step-uptime = 141
restore-step-user-progress = 32
[10:04:47.0318-GMT]{3>7} CHECKPOINT BEGIN: RESTORED:[0x0648] cleanup_send_final_status
restore-step-ids = {0x1103065B:47;0x11030648:49}
restore-step-names = {0x1103065B:asr_and_invert_image;0x11030648:cleanup_send_final_status}
restore-step-uptime = 141
restore-step-user-progress = 32

ERROR: Unable to restore device

nikias, 2019-06-02:
Hi, sorry but there is no obvious issue from the computer side. The device has multiple partitions (System, user data, baseband data, ...) and for some reason the first one which is where the root filesystems is to be restored is not large enough. Not sure from what version you initially had before trying to update, but apparently the root filesystem size changed, for example:

iOS 12.1.2 image 048-40501-063.dmg is 3431679581 bytes

vs.

iOS 12.3.1 image 048-43159-182.dmg is 3498654431 bytes

So the root filesystem size increased by about 2%, for you it fails at about 94%. This could be related, even though from what I understood is that the rootfs will have a sufficient size overhead (hence the "SystemPartitionPadding").

Unfortunately I don't see anything you can do here. You can't supply a different image (e.g. from a previous version) because it will cryptographically verify it.
Technically Apple would be able to help you: they could boot the device with some special ramdisk to recover data; however knowing them they will pretend there's nothing they can do. Especially not for a private person. You could ask but I doubt they will really help you. Sorry :-(

decembersake, 2019-06-02:
Oh my god.. I see. That's crazy it's just 2% and can't update. Yeah, I can see how Apple will reply but I appreciate your help and explanation so so much. Again, thank you so much. Have a lovely day!

closed 2019-06-02 by decembersake

nikias, 2019-06-02:
You can still try. If you go to a genius bar and they want to try something, they need to do a firmware update restore (not an erase restore which would wipe all your data).

nikias, 2019-06-04:
Hey I might have found something. There's an entry MinimumSystemPartition, maybe it needs to be added.

nikias, 2019-06-04:
Can you try this:
```
diff --git a/src/restore.c b/src/restore.c
index 4e6b91d..fc274ee 100644
--- a/src/restore.c
+++ b/src/restore.c
@@ -2412,16 +2412,26 @@ int restore_device(struct idevicerestore_client_t* client, plist_t build_identit
        plist_dict_set_item(opts, "SystemImage", plist_new_bool(1));
        // FIXME: new on iOS 5 ?
        plist_dict_set_item(opts, "SystemImageType", plist_new_string("User"));
+       plist_t msp = plist_access_path(build_identity, 2, "Info", "MinimumSystemPartition");
+       if (msp) {
+               plist_dict_set_item(opts, "MinimumSystemPartition", plist_copy(msp));
+               plist_dict_set_item(opts, "SystemPartitionSize", plist_copy(msp));
+       }
        plist_t spp = plist_access_path(build_identity, 2, "Info", "SystemPartitionPadding");
        if (spp) {
                spp = plist_copy(spp);
        } else {
                spp = plist_new_dict();
+               plist_dict_set_item(spp, "1024", plist_new_uint(1280));
                plist_dict_set_item(spp, "128", plist_new_uint(1280));
                plist_dict_set_item(spp, "16", plist_new_uint(160));
+               plist_dict_set_item(spp, "256", plist_new_uint(1280));
                plist_dict_set_item(spp, "32", plist_new_uint(320));
+               plist_dict_set_item(spp, "512", plist_new_uint(1280));
                plist_dict_set_item(spp, "64", plist_new_uint(640));
+               plist_dict_set_item(spp, "768", plist_new_uint(1280));
                plist_dict_set_item(spp, "8", plist_new_uint(80));
+
        }
        plist_dict_set_item(opts, "SystemPartitionPadding", spp);
        char* guid = generate_guid();
```

decembersake, 2019-06-05:
Hey sorry for the late reply, I almost gave up.
I obviously visited the genius bar apple store and they tried the firmware update which occurred error 14 at the very beginning of this disaster.
I have tried your new patch and it also didn't work, but I can't thank you enough. I'm really grateful for your help.

decembersake, 2019-06-05:
This is what happened, which seems to be exactly the same as before.

Changes to be committed:
  (use "git reset HEAD <file>..." to unstage)

	modified:   src/restore.c

usrname:idevicerestore Usrname$ git commit -a
[master d681e99] commit restore.c
 Committer: Usr <Usrname@whatsoever.local>

 2 files changed, 11 insertions(+)

create mode 160000 libimobiledevice

Verifying 'iPhone11,2_12.3.1_16F203_Restore.ipsw'...
Checksum matches.
Extracting BuildManifest from IPSW
Product Version: 12.3.1
Product Build: 16F203 Major: 16
INFO: device serial number is DNPX9193KPG4
Device supports Image4: true
Variant: Customer Upgrade Install (IPSW)
This restore will update your device without losing data.
Checking IPSW for required components...
All required components found in IPSW
Extracting filesystem from IPSW: 048-43159-182.dmg
[==================================================] 100.0%
Found ECID 4909083227848750
Getting ApNonce in recovery mode... c7 63 e9 00 67 57 84 b9 af de a7 6a 31 b7 9e ad ef 89 4c 72 e7 75 91 ab 35 92 a2 86 5b 1d 53 30 
Trying to fetch new SHSH blob
Getting SepNonce in recovery mode... a4 76 1c c5 bb 15 a0 e1 1c 73 9a 79 2f 24 e6 98 6a 97 bb 76 
WARNING: Unable to find BbSkeyId node
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received SHSH blobs
Extracting iBEC.d321.RELEASE.im4p...
Personalizing IMG4 component iBEC...
Sending iBEC (1063306 bytes)...
Recovery Mode Environment:
iBoot build-version=iBoot-4513.260.81
iBoot build-style=RELEASE
Sending RestoreLogo...
Extracting applelogo@3x~iphone.im4p...
Personalizing IMG4 component RestoreLogo...
Sending RestoreLogo (20910 bytes)...
Extracting h11_ane_fw_quin.im4p...
Personalizing IMG4 component ANE...
Sending ANE (9394706 bytes)...
Extracting aopfw-iphone11aop.im4p...
Personalizing IMG4 component AOP...
Sending AOP (1396234 bytes)...
Extracting AppleAVE2FW.im4p...
Personalizing IMG4 component AVE...
Sending AVE (1111370 bytes)...
Extracting armfw_g11p.im4p...
Personalizing IMG4 component GFX...
Sending GFX (1555834 bytes)...
Extracting adc-petra-d3x.im4p...
Personalizing IMG4 component ISP...
Sending ISP (14639242 bytes)...
Extracting 048-43107-184.dmg.trustcache...
Personalizing IMG4 component RestoreTrustCache...
Sending RestoreTrustCache (11671 bytes)...
Extracting SmartIOFirmwareT8020.im4p...
Personalizing IMG4 component SIO...
Sending SIO (1356094 bytes)...
Extracting WirelessPower.iphone11.im4p...
Personalizing IMG4 component WCHFirmwareUpdater...
Sending WCHFirmwareUpdater (28773 bytes)...
[... 1389 lines cut ...]
[20:03:41.0036-GMT]{3>7} CHECKPOINT FAILURE:(FAILURE:14) RESTORED:[0x065B] asr_and_invert_image [0]D(failed to restore APFS image)
restore-step-results = {0x1107065B:{0:14}}
restore-step-codes = {0x1107065B:{0:14}}
restore-step-domains = {0x1107065B:{0:"AMRestoreErrorDomain"}}
restore-step-error = {0x1107065B:"[0]D(failed to restore APFS image)"}
restore-step-uptime = 141
restore-step-user-progress = 32
[20:03:41.0041-GMT]{3>7} CHECKPOINT NOTICE: (NVRAM set) restore-step-user-progress=32 [sync=true] (first failure)
[20:03:41.0042-GMT]{3>7} CHECKPOINT BEGIN: RESTORED:[0x0647] cleanup_check_result
restore-step-ids = {0x1103065B:47;0x11030647:48}
restore-step-names = {0x1103065B:asr_and_invert_image;0x11030647:cleanup_check_result}
restore-step-uptime = 141
restore-step-user-progress = 32
[20:03:41.0044-GMT]{3>7} CHECKPOINT END: RESTORED:[0x0647] cleanup_check_result
restore-step-ids = {0x1103065B:47}
restore-step-names = {0x1103065B:asr_and_invert_image}
restore-step-uptime = 141
restore-step-user-progress = 32
[20:03:41.0048-GMT]{3>7} CHECKPOINT BEGIN: RESTORED:[0x0648] cleanup_send_final_status
restore-step-ids = {0x1103065B:47;0x11030648:49}
restore-step-names = {0x1103065B:asr_and_invert_image;0x11030648:cleanup_send_final_status}
restore-step-uptime = 141
restore-step-user-progress = 32

ERROR: Unable to restore device

cliewdevelop, 2019-08-22:
Hi @decembersake @nikias, I'm facing the exact same issue here - my ASR RESTORE PROGRESS is at 96%. 

At this point it seems like there's no good solution, until when Apple releases a compatible firmware that "does not resize the root filesystem to something else." Or a firmware that is very lean (though unlikely given the current trends). Is this understanding correct?

Trying to decide if I should leave the phone as is as a brick for until that day comes, or take the it and restore. Thank you!

cliewdevelop, 2019-08-22:
Current signed firmware is 12.4 only.

sungzh, 2019-09-18:
@cliewdevelop @decembersake  Have you recovered your mobile phone? I don't know the mechanism of the system, but I have seen some people constantly try. After dozens of times, there is enough space. I don't know if there is any theoretical basis. @nikias Do you know the theory inside?

lzl124631x, 2020-08-21:
Having the same issue. My OS version was 13.5.1 and the current version I'm trying to update it to 13.6.1. Stuck at 96% with error 14.

```
[23:01:41.0129] ASR RESTORE PROGRESS: 96%
[23:01:41.0129] ramrod_display_set_granular_progress_forced: 32.000000
[23:01:41.0129] ASR STATUS: fail
[23:01:41.0129] ASR: Could not restore - No space left on device
[23:01:41.0129] ASR STATUS: fail
[23:01:41.0129] restore_apfs_image : failed to restore an APFS image, error = 14
[23:01:41.0129] [06:01:37.0450-GMT]{3>6} CHECKPOINT FAILURE:(FAILURE:14) (null):[0x065B] asr_and_invert_image [0]D(failed to restore APFS image)
```

I think this is a insanely stupid error in the ios system. How can it have such a critical issue that updating the system causes the entire system be down?  Can't it pre-check if there is enough space for the new system? Apple great job!

cliewdevelop, 2020-08-21:
@sungzh - on my end, I just restored and lost my pictures and all data. I know its not the best news, but just updating for closure.

xslidian, 2021-01-05:
The FDR & ASR process changed in the iOS 14.4 beta (`18D5030e`). 
A few users reported successful restoration without wiping data. You may want to try it.   cc @lzl124631x

FYI, here's an example of unsuccessful update (no enough space):
````
## 14.3_18C66 ##
ASR RESTORE PROGRESS: 66%
ramrod_display_set_granular_progress_forced: 24.000000
ASR RESTORE PROGRESS: 68%
ramrod_display_set_granular_progress_forced: 24.000000
ASR RESTORE PROGRESS: 70%
ramrod_display_set_granular_progress_forced: 25.000000
ASR RESTORE PROGRESS: 72%
ramrod_display_set_granular_progress_forced: 26.000000
ASR STATUS: fail
ASR: Could not restore - No space left on device
ASR STATUS: fail
restore_apfs_image : failed to restore an APFS image, error = 14
[16:16:12.0102-GMT]{4>8} CHECKPOINT FAILURE:(FAILURE:14) (null):[0x065B] asr_and_invert_image [0]D(failed to restore APFS image)
````

````
## 14.4_18D5030e ##
Sending NORData now...
Done sending NORData
Unmounting filesystems (29)
Unmounting filesystems (29)
Unmounting filesystems (29)
Unmounting filesystems (29)
Unmounting filesystems (29)
Unmounting filesystems (29)
Checking filesystems (15)
About to send FDR Trust data...
Sending FDR Trust data now...
Done sending FDR Trust Data
Checking for uncollected logs (44)
Checking for uncollected logs (44)
ERROR: Unable to receive message from FDR 0x7ffa07f0a640 (-2). 0/2 bytes
ERROR: Unable to receive message from FDR 0x7ffa07d05420 (-2). 0/2 bytes
ERROR: Unable to receive message from FDR 0x7ffa07d05430 (-2). 0/2 bytes
ERROR: Unable to receive message from FDR 0x7ffa07c9aa00 (-2). 0/2 bytes
Unmounting filesystems (29)
Unmounting filesystems (29)
Unmounting filesystems (29)
Unmounting filesystems (29)
Unmounting filesystems (29)
Unmounting filesystems (29)
Creating filesystem (12)
About to send filesystem...
Connected to ASR
Validating the filesystem
Filesystem validated
Sending filesystem now...
[==================================                ]  66.0%ERROR: Unable to send data to ASR. Sent 0 of 131072 bytes.
ERROR: Unable to send filesystem payload
ERROR: Unable to send payload to ASR
ERROR: Unable to send filesystem
ERROR: Unable to successfully restore device
About to send filesystem...
Connected to ASR
Validating the filesystem
Filesystem validated
Sending filesystem now...
[==================================                ]  66.0%ERROR: Unable to send data to ASR. Sent 0 of 131072 bytes.
ERROR: Unable to send filesystem payload
ERROR: Unable to send payload to ASR
ERROR: Unable to send filesystem
ERROR: Unable to restore device
````

lzl124631x, 2021-01-05:
@xslidian Thanks! There are 50+ friend in our group who faced the same issue successfully recovered their data with the 14.4 beta. It indeed works.

Unfortunately, my phone was reset by my mistake and a insanely stupid bug of iTunes:

I was using iTunes for the restoring. Every time after the failure I need to unplug and replug the wire to do another trial. But iTunes has a bug that if you unplug and replug the wire during restoration, it changes it behavior to RESET! I mistakenly unplug/repluged my phone during the restore process and my phone was reset... I was a fan of Apple but now I hate it...

iTunes sucks. I should have used other 3rd party softwares that are less error prone. I put my phone in the drawer and maybe one day we can find ways to decrypt the date from it.

xslidian, 2021-01-06:
> 50+ friends in our group who faced the same issue successfully recovered their data with the 14.4 beta. It indeed works.

Glad to hear that. Amazing number.

> But iTunes has a bug that if you unplug and replug the wire during restoration, it changes it behavior to RESET! 

Thanks for telling us of such a bug. Hope you find a solution soon.

`idevicerestore` chooses the variant perfectly as far as I know. 
If your friends aren't comfortable with the command line, the free-to-use part of `iMazing` should help.