libimobiledevice/idevicerestore #268: iPhone 6s FDR error

brianclinkenbeard, 2019-06-24:
At first the device would get stuck at the start of a restore with an FDR timeout error. I tried modifying the code to increase the timeout from 20s to 100s and this would allow sending the filesystem to complete, however it would give the timeout error later and then it would fail to recover the FDR data.

Now (after changing the source code back and recompiling) the same thing occurs. Here is the full debug log:
```
opening device 05ac:1281...
Setting to configuration 1
Setting to interface 0:0
opening device 05ac:1281...
Setting to configuration 1
Setting to interface 0:0
opening device 05ac:1281...
Setting to configuration 1
Setting to interface 0:0
Getting device's ECID for TSS request
NOTE: Unable to find BbCalibrationManifestKeyHash node
NOTE: Unable to find BbFactoryActivationManifestKeyHash node
WARNING: Unable to find BbSkeyId node
DEBUG: Applying restore request rules for entry AOP
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry AppleLogo
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry BatteryCharging0
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry BatteryCharging1
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry BatteryFull
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry BatteryLow0
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry BatteryLow1
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry BatteryPlugin
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry DeviceTree
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry KernelCache
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry LLB
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry OS
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry RecoveryMode
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry RestoreDeviceTree
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry RestoreKernelCache
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry RestoreLogo
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry RestoreRamDisk
DEBUG: Adding action EPRO to TSS entry
DEBUG: Adding action ESEC to TSS entry
DEBUG: Applying restore request rules for entry RestoreSEP
DEBUG: Adding action ESEC to TSS entry
[... 4538 lines cut ...]
restore-step-names = {0x1103067B:perform_restore_installing;0x11030628:install_kernel_cache;0x11030E00:install_kernel_cache}
restore-step-uptime = 216
restore-step-user-progress = 52
entering install_kernel_cache
ramrod_display_set_granular_progress_forced: 52.000000
writing kernelcache to /mnt1/System/Library/Caches/com.apple.kernelcaches/kernelcache
[10:18:14.0813-GMT]{4>5} CHECKPOINT END: SYSTEM:[0x0E00] install_kernel_cache
restore-step-ids = {0x1103067B:55;0x11030628:57}
restore-step-names = {0x1103067B:perform_restore_installing;0x11030628:install_kernel_cache}
restore-step-uptime = 217
restore-step-user-progress = 52
[10:18:14.0817-GMT]{4>5} CHECKPOINT BEGIN: SYSTEM:[0x0E01] install_device_tree
restore-step-ids = {0x1103067B:55;0x11030628:57;0x11030E01:59}
restore-step-names = {0x1103067B:perform_restore_installing;0x11030628:install_kernel_cache;0x11030E01:install_device_tree}
restore-step-uptime = 217
restore-step-user-progress = 52
entering install_device_tree
ramrod_display_set_granular_progress_forced: 52.000000
install_device_tree: writing device tree to /mnt1/usr/standalone/firmware/devicetree.img4
[10:18:14.0893-GMT]{4>5} CHECKPOINT END: SYSTEM:[0x0E01] install_device_tree
restore-step-ids = {0x1103067B:55;0x11030628:57}
restore-step-names = {0x1103067B:perform_restore_installing;0x11030628:install_kernel_cache}
restore-step-uptime = 217
restore-step-user-progress = 52
[10:18:14.0898-GMT]{4>5} CHECKPOINT BEGIN: SYSTEM:[0x0E

brianclinkenbeard, 2019-06-24:
I tried updating + restoring with iTunes with another device and was successful. Now why I try to update with idevicerestore, nothing happens after `Fixing up /var`. The FDR error is still there so I'm not sure if that could be what's caused it to hang.
```
Found device in Normal mode
Identified device as n71ap, iPhone8,1
Extracting BuildManifest from IPSW
Product Version: 13.0
Product Build: 17A5508m Major: 17
Device supports Image4: true
Variant: Developer Upgrade Install (IPSW)
This restore will update your device without erasing user data.
Checking IPSW for required components...
All required components found in IPSW
Extracting filesystem from IPSW: 048-62316-052.dmg
[==================================================] 100.0%
Found ECID 2700186987548
Getting ApNonce in normal mode... 9f d2 aa a9 fc 12 8f 21 3f a2 cf b8 7d 60 1b 22 35 e1 0a 7e 
Trying to fetch new SHSH blob
Getting SepNonce in normal mode... 7c 6f 76 5d ae bb be 63 16 70 8b 9b d8 b3 5d 07 9c 1e c2 7a 
WARNING: Unable to find BbSkeyId node
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received SHSH blobs
Entering recovery mode...
INFO: device serial number is F17RF8B6GRY5
Extracting iBEC.n71.RELEASE.im4p...
Personalizing IMG4 component iBEC...
Sending iBEC (409247 bytes)...
Recovery Mode Environment:
iBoot build-version=iBoot-5540.0.49.0.1
iBoot build-style=RELEASE
Sending RestoreLogo...
Extracting applelogo@2x~iphone.im4p...
Personalizing IMG4 component RestoreLogo...
Sending RestoreLogo (12407 bytes)...
Extracting 048-62229-075.dmg.trustcache...
Personalizing IMG4 component RestoreTrustCache...
Sending RestoreTrustCache (10376 bytes)...
ramdisk-size=0x20000000
Extracting 048-62229-075.dmg...
Personalizing IMG4 component RestoreRamDisk...
Sending RestoreRamDisk (102825754 bytes)...
Extracting DeviceTree.n71ap.im4p...
Personalizing IMG4 component RestoreDeviceTree...
Sending RestoreDeviceTree (142221 bytes)...
Extracting kernelcache.release.n71...
Personalizing IMG4 component RestoreKernelCache...
Sending RestoreKernelCache (16922130 bytes)...
About to restore device... 
Waiting for device...
Device ffffffffffffffffffffffffffffffff00000002 is now connected in restore mode...
Connecting now...
Connected to com.apple.mobile.restored, version 15
Device ffffffffffffffffffffffffffffffff00000002 has successfully entered restore mode
Hardware Information:
BoardID: 4
ChipID: 32768
UniqueChipID: 2700186987548
ProductionMode: true
Starting FDR listener thread
About to send RootTicket...
Sending RootTicket now...
Done sending RootTicket
Waiting for NAND (28)
Updating S3E Firmware (58)
About to send NORData...
Found firmware path Firmware/all_flash
Getting firmware manifest from build identity
Extracting LLB.n71.RELEASE.im4p...
Personalizing IMG4 component LLB...
Extracting applelogo@2x~iphone.im4p...
[... 197 lines cut ...]
Extracting glyphplugin@1334~iphone-lightning.im4p...
Personalizing IMG4 component BatteryPlugin...
Extracting DeviceTree.n71ap.im4p...
Personalizing IMG4 component DeviceTree...
Extracting recoverymode@1334~iphone-lightning.im4p...
Personalizing IMG4 component RecoveryMode...
Extracting iBoot.n71.RELEASE.im4p...
Personalizing IMG4 component iBoot...
Extracting sep-firmware.n71.RELEASE.im4p...
Personalizing IMG4 component RestoreSEP...
Extracting sep-firmware.n71.RELEASE.im4p...
Personalizing IMG4 component SEP...
Sending NORData now...
Done sending NORData
Unmounting filesystems (29)
Unmounting filesystems (29)
Unmounting filesystems (29)
Unmounting filesystems (29)
Checking filesystems (15)
Checking filesystems (15)
About to send FDR Trust data...
Sending FDR Trust data now...
Done sending FDR Trust Data
ERROR: Unable to receive message from FDR 0x5619211b5df0 (-7). 0/2 bytes
```

chilik, 2019-06-24:
Had the same issue on iPhone9,1 iOS13.0 using an erase upgrade.
After it hung, I disconnect the cable, and use
irecovery -n
to let it finish the restore successfully.

nikias, 2019-06-24:
Ah. I think I know what's wrong. Let me make a fix for that. It is because of a recent change in libimobiledevice. Keep calm and wait for push.

nikias, 2019-06-24:
If you want to try, it should work now with latest commit 7cb1bdf78ce8963da2923d6098645cb648e30259.

brianclinkenbeard, 2019-06-24:
Worked perfect, thanks!

closed 2019-06-24 by brianclinkenbeard

yogendra123456, 2020-06-10:
Hi All,

i try to restore my device but it create error again and again on windows 10... i am using libmobiledevice code.

Extracting recoverymode@1334~iphone-lightning.im4p...
Personalizing IMG4 component RecoveryMode...
Extracting iBoot.n71.RELEASE.im4p...
Personalizing IMG4 component iBoot...
Extracting sep-firmware.n71.RELEASE.im4p...
Personalizing IMG4 component RestoreSEP...
Extracting sep-firmware.n71.RELEASE.im4p...
Personalizing IMG4 component SEP...
Sending NORData now...
Done sending NORData
Unmounting filesystems (29)
Unmounting filesystems (29)
Unmounting filesystems (29)
Unmounting filesystems (29)
Checking filesystems (15)
ERROR: Unable to receive message from FDR 00000124E29C2130 (-7). 0/2 bytes

Details in attached file..
[ErrorRelease.txt](https://github.com/libimobiledevice/idevicerestore/files/4758009/ErrorRelease.txt)

 Hi @nikias @brianclinkenbeard, i have already changed (7cb1bdf )..

kindly provide the update what is the solution of this error, i am using latest libmobiledevice code with visual studio 2019...

thanks in advanced....