libimobiledevice/idevicerestore #269: error on make

timonline, 2019-08-03:
The ./autogen.sh runs successfully and ends with

> Configuration for idevicerestore 1.0:
> -------------------------------------------
> 
>   Install prefix: .........: /usr/local
> 
>   Now type 'make' to build idevicerestore 1.0,
>   and then 'make install' for installation.

But when I try to run the make command I get:

> make  all-recursive
> make[1]: Entering directory `/home/tim/futrure_restore_tim/idevicerestore'
> Making all in src
> make[2]: Entering directory `/home/tim/futrure_restore_tim/idevicerestore/src'
>   CC       idevicerestore-img4.o
> img4.c: In function ‘asn1_find_element’:
> img4.c:126:2: error: ‘for’ loop initial declarations are only allowed in C99 mode
>   for (int i = 0; i <= index; i++) {
>   ^
> img4.c:126:2: note: use option -std=c99 or -std=gnu99 to compile your code
> make[2]: *** [idevicerestore-img4.o] Error 1
> make[2]: Leaving directory `/home/tim/futrure_restore_tim/idevicerestore/src'
> make[1]: *** [all-recursive] Error 1
> make[1]: Leaving directory `/home/tim/futrure_restore_tim/idevicerestore'
> make: *** [all] Error 2

iM4CH3T3, 2019-10-05:
u found the solution?

nikias, 2019-10-05:
See latest commit.