libimobiledevice/idevicerestore #270: I think there is some bug on idevicerestore main: Allow passing ECID in hex or decimal (commit ccd3ebe24feb3fa5fa60099144678137b4c8ad6f)

taanders, 2019-08-14:
I think this commit https://github.com/libimobiledevice/idevicerestore/commit/ccd3ebe24feb3fa5fa60099144678137b4c8ad6f#diff-e4400d4f65a91c1bfb021295e5a684aa breaks using ECID as HEX value. I tried this line on WIN32-build and I get this error.
```
ERROR: Could not parse ECID from '0000017C3A021D49'
```

main: Allow passing ECID in hex or decimal
Should this be something like?:
```
					char* tail = NULL;
					client->ecid = strtoull(optarg, &tail, 16);
					if (tail && (tail[0] != '\0')) {
						client->ecid = 0;
						tail = NULL;
					}
					if (client->ecid == 0) {
						client->ecid = strtoull(optarg, &tail, 0);
						if (tail && (tail[0] != '\0')) {
							client->ecid = 0;
						}
						if (client->ecid == 0) {
							error("ERROR: Could not parse ECID from '%s'\n", optarg);
							return -1;
						}
					}
```

nikias, 2019-08-14:
I changed the behavior, yes. You need to prefix with `0x` if the ECID is meant to be hexadecimal, otherwise it is expected to be decimal.

nikias, 2019-08-14:
The problem with your suggestions is if you specify a decimal ECID it will be treated as hex, so if you pass -i 12345678 it would be considered hex, so in decimal it would be 305419896. The `0x` is mandatory now and the --help output clearly says so.

taanders, 2019-08-16:
Ok, I have missed that change on instructions. It sound reasonable to add that 0x to ecid hex values. I close this issue.

closed 2019-08-16 by taanders