libimobiledevice/idevicerestore #272: Error on autogen ...

aiamuzz, 2019-08-22:
Hi,

running `autogen.sh` is giving the following error ...

```
configure: error: Package requirements (libcurl >= 7.0) were not met:

No package 'libcurl' found

Consider adjusting the PKG_CONFIG_PATH environment variable if you
installed software in a non-standard prefix.

Alternatively, you may set the environment variables libcurl_CFLAGS
and libcurl_LIBS to avoid the need to call pkg-config.
See the pkg-config man page for more details.
```

as you can see the `libcurl` version clearly shows its greater than the required dependency pointed by the error above ...

```
$ curl --version
curl 7.52.1 (x86_64-pc-linux-gnu) **libcurl/7.52.1** OpenSSL/1.0.2s zlib/1.2.8 libidn2/0.16 libpsl/0.17.0 (+libidn2/0.16) libssh2/1.7.0 nghttp2/1.18.1 librtmp/2.3
Protocols: dict file ftp ftps gopher http https imap imaps ldap ldaps pop3 pop3s rtmp rtsp scp sftp smb smbs smtp smtps telnet tftp 
Features: AsynchDNS IDN IPv6 Largefile GSS-API Kerberos SPNEGO NTLM NTLM_WB SSL libz TLS-SRP HTTP2 UnixSockets HTTPS-proxy PSL
```

Blefish, 2019-08-23:
You need to install the dev package of libcurl.

aiamuzz, 2019-10-13:
> You need to install the dev package of libcurl.

ok ... which one of these packages ... ?

```
$ sudo apt-get install libcurl
libcurl3              libcurl3-gnutls       libcurl4-doc          libcurl4-nss-dev      libcurl-ocaml         
libcurl3-dbg          libcurl3-nss          libcurl4-gnutls-dev   libcurl4-openssl-dev  libcurl-ocaml-dev   
```

aiamuzz, 2019-10-13:
@Blefish ... FYI ... seems like `autogen.sh` is running properly ... is it ?

`./autogen.sh `

```
libtoolize: putting auxiliary files in '.'.
libtoolize: linking file './ltmain.sh'
libtoolize: putting macros in AC_CONFIG_MACRO_DIRS, 'm4'.
libtoolize: linking file 'm4/libtool.m4'
libtoolize: linking file 'm4/ltoptions.m4'
libtoolize: linking file 'm4/ltsugar.m4'
libtoolize: linking file 'm4/ltversion.m4'
libtoolize: linking file 'm4/lt~obsolete.m4'
configure.ac:33: installing './compile'
configure.ac:8: installing './missing'
src/Makefile.am: installing './depcomp'
checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-pc-linux-gnu
checking target system type... x86_64-pc-linux-gnu
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether make supports nested variables... (cached) yes
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking for style of include used by make... GNU
checking dependency style of gcc... gcc3
checking how to print strings... printf
checking for a sed that does not truncate output... /usr/bin/sed
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for fgrep... /usr/bin/grep -F
checking for ld used by gcc... /usr/bin/ld
checking if the linker (/usr/bin/ld) is GNU ld... yes
checking for BSD- or MS-compatible name lister (nm)... /usr/bin/nm -B
checking the name lister (/usr/bin/nm -B) interface... BSD nm
checking whether ln -s works... yes
checking the maximum length of command line arguments... 1572864
checking how to convert x86_64-pc-linux-gnu file names to x86_64-pc-linux-gnu format... func_convert_file_noop
checking how to convert x86_64-pc-linux-gnu file names to toolchain format... func_convert_file_noop
checking for /usr/bin/ld option to reload object files... -r
checking for objdump... objdump
checking how to recognize dependent libraries... pass_all
checking for dlltool... no
checking how to associate runtime and link libraries... printf %s\n
checking for ar... ar
checking for archiver @FILE support... @
checking for strip... strip
checking for ranlib... ranlib
checking command to parse /usr/bin/nm -B output from gcc object... ok
checking for sysroot... no
checking for a working dd... /usr/bin/dd
checking how to truncate binary pipes... /usr/bin/dd bs=4096 count=1
checking for mt... mt
checking if mt is a manifest tool... no
checking how to run the C preprocessor... gcc -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
[... 43 lines cut ...]
checking for RESTORE_E_RECEIVE_TIMEOUT in enum restored_error_t... no
checking for special C compiler options needed for large files... no
checking for _FILE_OFFSET_BITS value needed for large files... no
checking for native large file support... yes
checking for _LARGEFILE_SOURCE value needed for large files... no
checking whether make supports nested variables... (cached) yes
checking that generated files are newer than configure... done
configure: creating ./config.status
config.status: creating Makefile
config.status: creating src/Makefile
config.status: creating docs/Makefile
config.status: creating config.h
config.status: config.h is unchanged
config.status: executing depfiles commands
config.status: executing libtool commands

Configuration for idevicerestore 1.0:
-------------------------------------------

  Install prefix: .........: /usr/local

  Now type 'make' to build idevicerestore 1.0,
  and then 'make install' for installation.

```

gelinger777, 2019-10-29:
@aiamuzz  did you found a solution??? I have also libimobiledevice/preboard.h problem

foreveryyl, 2019-12-31:
@gelinger777 me too

mathiasdg, 2020-02-13:
me too!

Any updates?

J3vaughn, 2020-02-16:
I think i found a solution

ARH-2020, 2020-02-16:
> I think i found a solution

Any luck? Struggling with this exact step myself - feel like it might need to run off an old version of libimobile but ideally not (since I already have libirecovery working and don't want to mess with it)

mathiasdg, 2020-02-16:
> I think i found a solution

can you share the solution please?

KTMnorth, 2020-02-19:
Did anyone get the solution to the make and preborad.h issue?  I'm a noob here.

sandhoners123, 2020-05-26:
build libimobiledevice from master

cusitosr88, 2020-06-06:
Did anyone get the solution?

emanuele-leopardi, 2024-10-12:
> Did anyone get the solution?

pacman -S libcurl-devel did the trick for me