libimobiledevice/idevicerestore #277: Fail to restore iOS 13.0 on XS

Blefish, 2019-09-20:
Hi

It appears that restoring iOS 13.0 on XS fails. I can see that the device will show loading spinner for a brief time and then the device restarts, with the "connect to laptop" logo (which was the iTunes logo before 13.0).

The very same firmware is also used on XS Max, where it works just fine. I have not noticed any other devices misbehaving.

Is it possible that a wrong binary is chosen from the firmware?

Also noteworthy - I used idevicerestore to flash a 13.1 build on the device, where it worked just fine. Returning to 13.0 and also another device which was on 12.4.1 fail to restore.

[log.txt](https://github.com/libimobiledevice/idevicerestore/files/3635613/log.txt)

truonggiang0710, 2019-09-20:
Hi @Blesfish,
I run but seem have problem on when extract iOS package
this is log
./idevicerestore -e /home/greystone/Desktop/iPhone10,3,iPhone10,6_13.0_17A577_Restore.ipsw 
Found device in Normal mode
Identified device as d22ap, iPhone10,3
Extracting BuildManifest from IPSW
ERROR: zip_open: /home/user/Desktop/iPhone10,3,iPhone10,6_13.0_17A577_Restore.ipsw: 5
ERROR: zip_open: /home/user/Desktop/iPhone10,3,iPhone10,6_13.0_17A577_Restore.ipsw: 5
ERROR: Invalid archive
ERROR: Unable to extract BuildManifest from /home/user/Desktop/iPhone10,3,iPhone10,6_13.0_17A577_Restore.ipsw. Firmware file might be corrupt.

Please advise.

Blefish, 2019-09-23:
Not sure if I'm in the right direction, but I noticed the RestoreRamDisk ending is different vs iTunes. It starts with the IM4M MANB etc sections, which were recently introduced in idevicerestore as well, but only in normal mode?

I attached diff of the buffer before the last one when sending RestoreRamDisk, this is where I noticed differences. The last buffer is also different. I don't understand why it would only affect XS though.

https://www.diffchecker.com/wOySylVH

Blefish, 2019-09-24:
It looks like working iTunes has a longer blob size, I noticed this when investigating the header sent to the device has a different size. I guess the IM4M sections are the Apple magic crypto which I do not quite understand yet and these should be different when using new SHSH files every time.

Blefish, 2019-09-24:
I found the reason, stand by for a patch

Blefish, 2019-09-24:
It seems like this patch breaks SHSH key request for iPhone 8, possibly more. I wonder how this should be differentiated.

closed 2019-09-26 by Blefish