libimobiledevice/idevicerestore #278: Fail to restore iOS 13.0 on iphone11 pro

barluc, 2019-09-23:
ERROR: tss_send_request: Unhandled status code 133
ERROR: TSS request failed: Failed connect to 17.151.36.30:443; No error (status=133)
ERROR: Unable to fetch Baseband TSS
ERROR: Unable to send baseband data
ERROR: Unable to successfully restore devic
No data to read
No data to read
FDR 03DDB678 timeout waiting for command
FDR 03DDB678 timeout waiting for command
FDR 03DDB678 terminating...
FDR 03DDB678 terminating...
ERROR: Unable to restore device### ### 
[fixlog.txt](https://github.com/libimobiledevice/idevicerestore/files/3641999/fixlog.txt)

nikias, 2019-09-23:
Yes, I am aware of that. I have a device here and am working on it. Might take a few days.

nikias, 2019-09-26:
Fixed with new commits. Check out latest master.

closed 2019-09-26 by nikias

iM4CH3T3, 2019-10-03:
Getting unhandled status code 133 on iPhone 11 with the latest build