libimobiledevice/idevicerestore #280: Unable to receive message from FDR : (

sakibahmed2002, 2019-09-24:
Guys help me with this error I'm getting while restoring my iPhone 6 
( Imma paste the whole stuff ) : 

`root@kali:~# idevicerestore '/media/root/Windows Xp Professional/iPhone_4.7_12.4.1_16G102_Restore.ipsw' -e 
Found device in Recovery mode
Identified device as n61ap, iPhone7,2
Extracting BuildManifest from IPSW
Product Version: 12.4.1
Product Build: 16G102 Major: 16
INFO: device serial number is C7JNMJRLG5MP
Device supports Image4: true
Variant: Customer Erase Install (IPSW)
This restore will erase your device data.
################################ [ WARNING ] #################################
# You are about to perform an *ERASE* restore. ALL DATA on the target device #
# will be IRREVERSIBLY DESTROYED. If you want to update your device without  #
# erasing the user data, hit CTRL+C now and restart without -e or --erase    #
# command line switch.                                                       #
# If you want to continue with the ERASE, please type YES and press ENTER.   #
##############################################################################
> YES
Checking IPSW for required components...
All required components found in IPSW
Using cached filesystem from '/media/root/Windows Xp Professional/iPhone_4.7_12.4.1_16G102_Restore/048-76490-095.dmg'
Found ECID 3770070889760806
Getting ApNonce in recovery mode... 5a 5e ec fb b4 24 25 5a 94 f5 71 df 60 3e 89 92 f7 c7 55 b3 
Trying to fetch new SHSH blob
Getting SepNonce in recovery mode... ff 31 f1 fa 6d 8f 04 7c fc c8 c0 d1 60 57 3f ff 91 b0 51 e2 
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received SHSH blobs
Extracting iBEC.n61.RELEASE.im4p...
Personalizing IMG4 component iBEC...
Sending iBEC (731534 bytes)...
Recovery Mode Environment:
iBoot build-version=iBoot-4513.270.14
iBoot build-style=RELEASE
Sending RestoreLogo...
Extracting applelogo@2x~iphone.im4p...
Personalizing IMG4 component RestoreLogo...
Sending RestoreLogo (12334 bytes)...
Extracting 048-78047-096.dmg.trustcache...
Personalizing IMG4 component RestoreTrustCache...
Sending RestoreTrustCache (9681 bytes)...
ramdisk-size=0x10000000
Extracting 048-78047-096.dmg...
Personalizing IMG4 component RestoreRamDisk...
Sending RestoreRamDisk (91608779 bytes)...
Extracting DeviceTree.n61ap.im4p...
Personalizing IMG4 component RestoreDeviceTree...
Sending RestoreDeviceTree (125714 bytes)...
Extracting kernelcache.release.iphone7...
Personalizing IMG4 component RestoreKernelCache...
Sending RestoreKernelCache (14069096 bytes)...
About to restore device... 
Waiting for device...
Device 9490734f955af83d6673c4681690fd0dede3b010 is now connected in restore mode...
Connecting now...
Connected to com.apple.mobile.restored, version 15
Device 9490734f955af83d6673c4681690fd0dede3b010 has successfully entered restore mode
Hardware Information:
BoardID: 6
ChipID: 28672
UniqueChipID: 3770070889760806
ProductionMode: true
Starting FDR listener thread
About to send RootTicket...
Sending RootTicket now...
Done sending RootTicket
Waiting for NAND (28)
About to send NORData...
Found firmware path Firmware/all_flash
Getting firmware manifest from build identity
Extracting LLB.n61.RELEASE.im4p...
Personalizing IMG4 component LLB...
Extracting applelogo@2x~iphone.im4p...
Personalizing IMG4 component AppleLogo...
Extracting batterycharging0@2x~iphone.im4p...
Personalizing IMG4 component BatteryCharging0...
Extracting batterycharging1@2x~iphone.im4p...
Personalizing IMG4 component BatteryCharging1...
Extracting batteryfull@2x~iphone.im4p...
Personalizing IMG4 component BatteryFull...
Extracting batterylow0@2x~iphone.im4p...
Personalizing IMG4 component BatteryLow0...
Extracting batterylow1@2x~iphone.im4p...
Personalizing IMG4 component BatteryLow1...
Extracting glyphplugin@1334~iphone-lightning.im4p...
Personalizing IMG4 component BatteryPlugin...
Extracting DeviceTree.n61ap.im4p...
Personalizing IMG4 component DeviceTree...
Extracting recoverymode@1334~iphone-lightning.im4p...
Personalizing IMG4 component RecoveryMode...
Extracting iBoot.n61.RELEASE.im4p...
Personalizing IMG4 component iBoot...
Extracting sep-firmware.n61.RELEASE.im4p...
Personalizing IMG4 component RestoreSEP...
Extracting sep-firmware.n61.RELEASE.im4p...
Personalizing IMG4 component SEP...
Sending NORData now...
Done sending NORData
ERROR: Unable to receive message from FDR 0x55c152cca6c0 (-2). 0/2 bytes`

PS : my device goes back to recovery mode after that and the terminal keeps showing the message.
Help Me Guys !

taanders, 2019-09-24:
I got this kind of errors when I used on msys2 environment other than mingw32 prefix. It was caused by some error on this:
```
# check if libimobiledevice has timeout errors
AC_CACHE_CHECK(for IDEVICE_E_TIMEOUT in enum idevice_error_t, ac_cv_idevice_error_has_timeout,
	AC_TRY_COMPILE([
		#include <libimobiledevice/libimobiledevice.h>
		], [
			return IDEVICE_E_TIMEOUT;
		], ac_cv_idevice_error_has_timeout=yes, ac_cv_idevice_error_has_timeout=no))

if test "$ac_cv_idevice_error_has_timeout" = "yes"; then
	AC_DEFINE(HAVE_IDEVICE_E_TIMEOUT, 1, [Define if enum idevice_error_t defines IDEVICE_E_TIMEOUT])
fi
```

Using CPPFLAGS define fixes that issue on my case. I think that kind of errors could be also related for some kind of USB Hub issue.

Blefish, 2019-09-25:
I'm using CPPFLAGS define for my use as well, so it always prefers built headers and autoconf check passes. I've seen the infamous `ERROR: Unable to receive message from FDR 0x55c152cca6c0 (-2). 0/2 bytes` many times, yet I can't find a solution. Try it another time and the same device flashes just fine..

yogendra123456, 2020-06-10:
Hi All,

i try to restore my device but it create error again and again on windows 10... i am using libmobiledevice code.

Extracting recoverymode@1334~iphone-lightning.im4p...
Personalizing IMG4 component RecoveryMode...
Extracting iBoot.n71.RELEASE.im4p...
Personalizing IMG4 component iBoot...
Extracting sep-firmware.n71.RELEASE.im4p...
Personalizing IMG4 component RestoreSEP...
Extracting sep-firmware.n71.RELEASE.im4p...
Personalizing IMG4 component SEP...
Sending NORData now...
Done sending NORData
Unmounting filesystems (29)
Unmounting filesystems (29)
Unmounting filesystems (29)
Unmounting filesystems (29)
Checking filesystems (15)
ERROR: Unable to receive message from FDR 00000124E29C2130 (-7). 0/2 bytes

Details in attached file..
[ErrorRelease.txt](https://github.com/libimobiledevice/idevicerestore/files/4757152/ErrorRelease.txt)

kindly provide the update what is the solution of this error, i am using latest libmobiledevice code with visual studio 2019...

thanks in advanced....

kenkit, 2020-12-27:
The same happens on iphone 5s

kenkit, 2020-12-30:
Now I have 3 devices that are unable to restore. Not even with itunes or 3utools because of this bug.
I suspect it has something to do with jailbreak

iOSPrincekin, 2023-09-02:
I have this problem too

iOSPrincekin, 2023-09-02:
`	device_error = idevice_connection_receive_timeout(fdr->connection, (char *)&cmd, sizeof(cmd), &bytes, 40000);
`
upload timeout to 40000 sovled my problem!

Andosonson, 2023-10-29:
> ` device_error = idevice_connection_receive_timeout(fdr->connection, (char *)&cmd, sizeof(cmd), &bytes, 40000); ` upload timeout to 40000 sovled my problem!

Hello!!! Where did you set this variable please???