libimobiledevice/idevicerestore #281: ERROR: Could not find BasebandFirmware Dictionary node in Baseband TSS response

a-gunter, 2019-09-27:
Replicated so far on iPhone 8P and iPhone XS, restoring to both iOS 12.4.1 and 13.1. Testing on Ubuntu with latest commits in idevicerestore, libimobiledevice, libplist, libirecovery & libusbmuxd.

With latest idevicerestore commits, restoring the device fails when sending the Baseband TSS request (in function restore_sign_bbfw). Baseband TSS response plist will include the BBTicket node but not the BasebandFirmware node, causing it to exit with "ERROR: Could not find BasebandFirmware Dictionary node in Baseband TSS response".

I rolled back to commit 2de3c1960ced7d8ca2225a42c949c51b9b993278 and was able to replicate the issue, but using commit 879715ce960ddb8e84d2d070edb5735782e3f452 (1 commit earlier) is successful for both devices.

https://www.diffchecker.com/NhKpHNGm shows the logs for an iPhone 8P restoring to 12.4.1. Outputs are of commit 2de3c1960ced7d8ca2225a42c949c51b9b993278 (successful) vs latest commits (fail). The only code change I made was to print the bbtss plist right before the point of failure, before realizing that the debug logs will already print the plist :laughing: 

Any idea why the BasebandFirmware node is hiding from me now?

nikias, 2019-09-27:
Can you try with commit 9181cd5eed8c5046133d2f2fe9cee59e52a30ccf ?

a-gunter, 2019-09-27:
That worked, thanks! Verified on 7, 8P, and XS so far.

Closing.

closed 2019-09-27 by a-gunter