libimobiledevice/idevicerestore #282: idevicerestore can't find iPad in DFU mode

ChrisCheney, 2019-09-30:
After updating all libimobiledevice binaries to current git as of today I can no longer use idevicerestore due to it not finding iPad in DFU mode.

Bus 001 Device 010: ID 05ac:1227 Apple, Inc. Mobile Device (DFU Mode)

idevicerestore -e -d iPad_64bit_TouchID_ASTC_13.1.1_17A854_Restore.ipsw
ERROR: Unable to discover device mode. Please make sure a device is attached.

I restored an iPad around Sep 24 using current git at that time and it worked, so it appears to be a recent change.

nikias, 2019-09-30:
can you connect to the DFU mode device with irecovery ?

Orbif, 2019-09-30:
I have the same problem on iPhone with latest sources. 

irecovery see the device in _Recovery Mode_, but i have the error _unable to recover device mode_ with idevicerestore.

taanders, 2019-09-30:
I got same kind of action with latest sources.

nikias, 2019-09-30:
What platform are you guys on? I tested this on macOS

Orbif, 2019-09-30:
Windows, compiled with mingw. I had to apply taanders patch https://github.com/libimobiledevice/idevicerestore/issues/283 for the compilation.

taanders, 2019-09-30:
I have tested also with Windows.

a-gunter, 2019-09-30:
I noticed the same behavior (on Linux). However, if you target the specific device with -i [ECID], it successfully finds the device. 

I think this only applies to devices in recovery/DFU mode, as you can still pass no ECID/UDID for a device in normal mode & it'll find it.

ChrisCheney, 2019-09-30:
I am using it under Linux.

I was able to query it with irecovery while in DFU and get the ECID and then restore using idevicerestore by specifying the ECID directly. So it appears the issue is that detection was recently broken.

taanders, 2019-09-30:
I think this commit breaks it some way:
https://github.com/libimobiledevice/idevicerestore/commit/3050e61588bfc79b3b360fa25db32c81e3264efb

nikias, 2019-09-30:
I know why, and it's the same on macOS. So I was actually planning to make it mandatory to specify ECID or UDID, after I realized that it won't select the device in recovery/DFU mode. But I didn't do it yet.
Let me change it this way:
If only one device in either Recovery/DFU (or normal) mode is connected, don't complain and pick the first one found. This works fine with normal mode.
Should it find multiple ones it should print a warning and exit.

taanders, 2019-09-30:
I was not able to do restore on normal mode device either. When device changes to recovery mode then idevicerestore exits with message:
Entering recovery mode...
ERROR: Failed to enter recovery mode
ERROR: Unable to place device into recovery mode from (null) mode

taanders, 2019-09-30:
I was not using ECID or UDID, it might causing that issue also with normal mode devices on Windows but it still not working as it used to work.

nikias, 2019-09-30:
ok will check it...

nikias, 2019-10-07:
Ok I pushed some commits, but I ended up not checking for multiple devices and just pick the first one it finds.

taanders, 2019-11-21:
Function _irecv_event_handler does not work on Windows at libirecovery side. It 2132:
```
		FOREACH(struct irecv_usb_device_info *devinfo, &devices) {
			if (!devinfo->alive) {
				_irecv_handle_device_remove(devinfo);
			}
		} ENDFOREACH
```

This causes some way continuous shifting between connected and disconnected states at idevicerestore irecv_event_cb:

irecv_event_cb: device 000e69681885e520 (udid: N/A) connected in Recovery mode
irecv_event_cb: device 000e69681885e520 (udid: N/A) disconnected
irecv_event_cb: device 000e69681885e520 (udid: N/A) connected in Recovery mode
irecv_event_cb: device 000e69681885e520 (udid: N/A) disconnected
irecv_event_cb: device 000e69681885e520 (udid: N/A) connected in Recovery mode
irecv_event_cb: device 000e69681885e520 (udid: N/A) disconnected

I'm not sure if this is causing some way that commit 3050e61588bfc79b3b360fa25db32c81e3264efb is not working on Windows.

taanders, 2019-11-24:
@nikias Hi, it seems that libirecovery is not working yet on Windows with latest codes on master.

I seems that new callback methods are changing states between 

```
irecv_event_cb: device 000e69681885e520 (udid: N/A) connected in Recovery mode
irecv_event_cb: device 000e69681885e520 (udid: N/A) disconnected
irecv_event_cb: device 000e69681885e520 (udid: N/A) connected in Recovery mode
irecv_event_cb: device 000e69681885e520 (udid: N/A) disconnected
irecv_event_cb: device 000e69681885e520 (udid: N/A) connected in Recovery mode
irecv_event_cb: device 000e69681885e520 (udid: N/A) disconnected
irecv_event_cb: device 000e69681885e520 (udid: N/A) connected in Recovery mode
irecv_event_cb: device 000e69681885e520 (udid: N/A) disconnected
```

I think there comes those events continuously:
_irecv_handle_device_add(&win_ctx);
_irecv_handle_device_remove(devinfo);

I'm not sure if there is some kind of situation that there comes removed event from GUID_DEVINTERFACE_DFU device list entries and connected event from GUID_DEVINTERFACE_IBOOT devices list entries.

This causes error on idevicerestore device mode detection:
ERROR: Device is in an invalid state
Getting device's ECID for TSS request
ERROR: Device is in an invalid state
ERROR: Unable to find device ECID

jkscheung, 2019-12-23:
Hi, I'm encountering this error too when running idevicerestore under Windows. Just wondering if there are any fixes on the way soon or if there is a workaround?

Many thanks.

haen7, 2020-08-10:
I solved this problem by uninstalling all Apple Mobile Device USB Drivers (version 423.36.0.0) from device manager and reinstalling iTunes (12.10.7.3) and Apple Mobile Device USB Driver (6.0.9999.69).

gabrc52, 2020-08-16:
In my case, this didn't work because I didn't run the command as super user. So you may need to prepend `sudo` to the command.

nikias, 2021-01-02:
> @nikias Hi, it seems that libirecovery is not working yet on Windows with latest codes on master.
>
> I seems that new callback methods are changing states between :
>
> ```
> irecv_event_cb: device [...]

@taanders this has been fixed by @DanyL and pushed to libirecovery master (commit https://github.com/libimobiledevice/libirecovery/commit/ea51265e251eacbff00b959d54bdad5f05ec9983).

taanders, 2021-01-05:
@nikias & @DanyL Hi, I tested this fix https://github.com/libimobiledevice/libirecovery/commit/ea51265e251eacbff00b959d54bdad5f05ec9983 committed by @DanyL. It seems that changing FOREACH loop position to earlier place eliminates this oscillating between states.

Parsing ecid from these lines is still not working properly I think.
```
	// Old driver result line:
	// \\?\usb#vid_05ac&pid_1281#cpid:8015_cprv:11_cpfm:03_scep:01_bdid:0a_ecid:0009299608384028_ibfl:3d_srnm:[c8pw28mfjc69]#{b8085869-feb9-404b-8cb1-1e5c14fa8c54}\0000

	// New driver result line:
	// \\?\usb#vid_05ac&pid_1281&mi_01#8&7eb9421&3&0001#{b8085869-feb9-404b-8cb1-1e5c14fa8c54}\usb#vid_05ac&pid_1281#cpid:8015_cprv:11_cpfm:03_scep:01_bdid:0a_ecid:0009299608384028_ibfl:3d_srnm:[c8pw28mfjc69]#mux1

	// DFU mode result line:
	// \\?\usb#vid_05ac&pid_1227#cpid:8000_cprv:20_cpfm:03_scep:01_bdid:04_ecid:000004f7f2c71227_ibfl:1c_srtg:[iboot-2234.0.0.3.3]#{b8085869-feb9-404b-8cb1-1e5c14fa8c54}\0000
	// \\?\usb#vid_05ac&pid_1281&mi_00#6&1e8f7b54&0&0000#{ed82a167-d61a-4af6-9ab6-11e52236c576}\usb#vid_05ac&pid_1281#sdom:01_cpid:8010_cprv:11_cpfm:03_scep:01_bdid:0c_ecid:000e69681885e527_ibfl:3d_srnm:[fk1sf318hg7f]#mux1

	// DFU mode new driver result line:
	// \\?\usb#vid_05ac&pid_1227#cpid:8000_cprv:20_cpfm:03_scep:01_bdid:04_ecid:00000175f501dd27_ibfl:1c_srtg:[iboot-2234.0.0.3.3]#{b8085869-feb9-404b-8cb1-1e5c14fa8c54}\usb#vid_8087&pid_0024#5&2e566f15&0&1#mux1
```

Those error causing lines are coming with Windows Update/Store Apple drivers version 432.0.0.36.