libimobiledevice/idevicerestore #284: Stay stuck restoring iOS 12.4.2 on iPhone 6 and 6+

Orbif, 2019-10-02:
When I try to restore iOS 12.4.2 on older devices with latest sources, it stay stuck after the step "Done sending FDR Trust Data" with the error "Unable to receive message from FDR 0000000000B88420 (-7). 0/2 bytes"

[restore_stuck_ios_12.4.2.txt](https://github.com/libimobiledevice/idevicerestore/files/3681727/restore_stuck_ios_12.4.2.txt)

Seems very similar to the issue https://github.com/libimobiledevice/idevicerestore/issues/268 

Does anyone have the same problem ?

Orbif, 2019-10-03:
I managed to fix this issue with taanders solution (https://github.com/libimobiledevice/idevicerestore/issues/280) by adding libimobiledevice include directory to CPPFLAGS instead of libimobiledevice_CFLAGS.

closed 2019-10-03 by Orbif

TheNegativeZero, 2020-06-08:
@Orbif Hi, would you care to explain your solution a bit more?  I have been troubleshooting this very same issue for days but to no avail.  

I have increased the timeout and retries to establish the connection and added a few checkpoints in fdr.c and restore.c just con confirm that indeed the program is not getting a successful connection before aborting.

Any hints are greatly appreciated.  Cheers.

yogendra123456, 2020-06-10:
Hi All,

i try to restore my device but it create error again and again on windows 10... i am using libmobiledevice code.

Extracting recoverymode@1334~iphone-lightning.im4p...
Personalizing IMG4 component RecoveryMode...
Extracting iBoot.n71.RELEASE.im4p...
Personalizing IMG4 component iBoot...
Extracting sep-firmware.n71.RELEASE.im4p...
Personalizing IMG4 component RestoreSEP...
Extracting sep-firmware.n71.RELEASE.im4p...
Personalizing IMG4 component SEP...
Sending NORData now...
Done sending NORData
Unmounting filesystems (29)
Unmounting filesystems (29)
Unmounting filesystems (29)
Unmounting filesystems (29)
Checking filesystems (15)
ERROR: Unable to receive message from FDR 00000124E29C2130 (-7). 0/2 bytes

Details in attached file..
[ErrorRelease.txt](https://github.com/libimobiledevice/idevicerestore/files/4757491/ErrorRelease.txt)

kindly provide the update what is the solution of this error, i am using latest libmobiledevice code with visual studio 2019...

thanks in advanced....