libimobiledevice/idevicerestore #285: make errors on ubuntu 18.04

timonline, 2019-10-03:
The autogen is successful with the following output:

```
$ ./autogen.sh

Configuration for idevicerestore 1.0:
-------------------------------------------

  Install prefix: .........: /usr/local

  Now type 'make' to build idevicerestore 1.0,
  and then 'make install' for installation.
```
But the make fails with these errors:

```
common.h:107:2:` error: unknown type name ‘irecv_device_event_context_t’
  irecv_device_event_context_t irecv_e_ctx;
  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
idevicerestore.c:226:34: error: unknown type name ‘irecv_device_event_t’
 static void irecv_event_cb(const irecv_device_event_t* event, void *userdata)
                                  ^~~~~~~~~~~~~~~~~~~~
idevicerestore.c: In function ‘irecv_event_cb’:
idevicerestore.c:229:11: error: request for member ‘type’ in something not a structure or union
  if (event->type == IRECV_DEVICE_ADD) {
           ^~
idevicerestore.c:229:21: error: ‘IRECV_DEVICE_ADD’ undeclared (first use in this function); did you mean ‘IDEVICE_DEVICE_ADD’?
  if (event->type == IRECV_DEVICE_ADD) {
                     ^~~~~~~~~~~~~~~~
                     IDEVICE_DEVICE_ADD
idevicerestore.c:229:21: note: each undeclared identifier is reported only once for each function it appears in
idevicerestore.c:231:24: error: request for member ‘device_info’ in something not a structure or union
    client->ecid = event->device_info->ecid;
                        ^~
idevicerestore.c:233:28: error: request for member ‘device_info’ in something not a structure or union
   if (client->ecid && event->device_info->ecid == client->ecid) {
                            ^~
idevicerestore.c:234:17: error: request for member ‘mode’ in something not a structure or union
    switch (event->mode) {
                 ^~
idevicerestore.c:252:18: error: request for member ‘type’ in something not a structure or union
  } else if (event->type == IRECV_DEVICE_REMOVE) {
                  ^~
idevicerestore.c:252:28: error: ‘IRECV_DEVICE_REMOVE’ undeclared (first use in this function); did you mean ‘IDEVICE_DEVICE_REMOVE’?
  } else if (event->type == IRECV_DEVICE_REMOVE) {
                            ^~~~~~~~~~~~~~~~~~~
                            IDEVICE_DEVICE_REMOVE
idevicerestore.c:253:28: error: request for member ‘device_info’ in something not a structure or union
   if (client->ecid && event->device_info->ecid == client->ecid) {
                            ^~
```

nikias, 2019-10-03:
Please update libirecovery.

timonline, 2019-10-04:
> Please update libirecovery.

Was using the latest version, just had to export the path.

Thanks

closed 2019-10-04 by timonline

Adrenokrome82, 2020-03-27:
> > Please update libirecovery.
> 
> Was using the latest version, just had to export the path.
> 
> Thanks

What is the path that you export and how? I have the same error. Thank