libimobiledevice/idevicerestore #286: iphone11/iphone11 pro Unknown data request 'BasebandUpdaterOutputData' received

barluc, 2019-10-09:
[plist.log](https://github.com/libimobiledevice/idevicerestore/files/3707037/plist.log)

`		else if (!strcmp(type, "FUDData")) {
			if(restore_send_fud_data(restore, client, build_identity, message) < 0) {
				error("ERROR: Unable to send FUD data\n");
				return -1;
			}
		}

		else if (!strcmp(type, "FirmwareUpdaterData")) {
			if(restore_send_firmware_updater_data(restore, client, build_identity, message) < 0) {
				error("ERROR: Unable to send FirmwareUpdater data\n");
				return -1;
			}
		}
       
        //this msg maybe only iphone11/iphone12 can receive.
        //if you not handle this msg, device progress will stay 95% forever.
        else if (!strcmp(type, "BasebandUpdaterOutputData")) {     
            //
            error("recvUnknownMessage");
        }
`

barluc, 2019-10-09:
`	else if (!strcmp(type, "FirmwareUpdaterData")) {
		if(restore_send_firmware_updater_data(restore, client, build_identity, message) < 0) {
			error("ERROR: Unable to send FirmwareUpdater data\n");
			return -1;
		}
	}
   
    //this msg maybe only iphone11/iphone12 can receive.
    //if you not handle this msg, device progress will stay 95% forever.
    else if (!strcmp(type, "BasebandUpdaterOutputData")) {     
        //
        error("recvUnknownMessage");
    }`

barluc, 2019-10-09:
[Recv 353 bytes,Count=185]:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>DataPort</key>
	<integer>49212</integer>
	<key>DataType</key>
	<string>BasebandUpdaterOutputData</string>    
	<key>MsgType</key>
	<string>DataRequestMsg</string>
</dict>
</plist>

---------------------------------------
when receive "BasebandUpdaterOutputData" msg ,the program will loop forever.

closed 2019-10-11 by barluc

zfuke164345, 2019-11-11:
how to handle it @barluc

mroger-tech, 2021-02-02:
i have  the same problem, how to handle it,please! @barluc

nikias, 2021-02-02:
I am working on a patch. The device wants "us" to connect to it and download a cpio archive.

nikias, 2021-02-02:
Actually, correction: The device wants us to send a cpio archive...

nikias, 2021-02-02:
The device doesn't seem to require any reply, but I still think we should handle this properly. Can someone of you guys test this patch, run a restore (no need for `-d`!!) and paste the hexdump output here? Thanks
https://gist.github.com/nikias/df2c8cdca3515985d1ec9bcb76cae291

mroger-tech, 2021-02-03:
> The device doesn't seem to require any reply, but I still think we should handle this properly. Can someone of you guys test this patch, run a restore (no need for `-d`!!) and paste the hexdump output here? Thanks
> https://gist.github.com/nikias/df2c8cdca3515985d1ec9bcb76cae291

I test the patch,the hexdump as bellow:

0000: 30 37 30 37 30 37 30 30 30 30 30 30 30 30 30 30   | 0707070000000000
0010: 30 30 30 34 30 37 35 35 30 30 30 30 33 31 30 30   | 0004075500003100
0020: 30 30 33 31 30 30 30 30 30 34 30 30 30 30 30 30   | 0031000004000000
0030: 31 34 30 30 36 34 31 33 30 35 30 30 30 30 30 30   | 1400641305000000
0040: 32 30 30 30 30 30 30 30 30 30 30 30 2e 00         | 200000000000..

0000: 30 37 30 37 30 37 30 30 30 30 30 30 30 30 30 30   | 0707070000000000
0010: 30 31 30 34 30 37 35 35 30 30 30 30 33 31 30 30   | 0104075500003100
0020: 30 30 33 31 30 30 30 30 30 32 30 30 30 30 30 30   | 0031000002000000
0030: 31 34 30 30 36 34 31 33 30 33 36 30 30 30 30 31   | 1400641303600001
0040: 31 30 30 30 30 30 30 30 30 30 30 30 2e 2f 53 61   | 100000000000./Sa
0050: 76 61 67 65 00 30 37 30 37 30 37 30 30 30 30 30   | vage.07070700000
0060: 30 30 30 30 30 30 32 30 34 30 37 35 35 30 30 30   | 0000002040755000
0070: 30 33 31 30 30 30 30 33 31 30 30 30 30 30 33 30   | 0310000310000030
0080: 30 30 30 30 30 31 34 30 30 36 34 31 33 30 35 30   | 0000014006413050
0090: 30 30 30 30 30 37 30 30 30 30 30 30 30 30 30 30   | 0000070000000000
00a0: 30 2e 2f 52 6f 73 65 00 30 37 30 37 30 37 30 30   | 0./Rose.07070700
00b0: 30 30 30 30 30 30 30 30 30 33 30 34 30 37 35 35   | 0000000003040755
00c0: 30 30 30 30 30 30 30 30 30 30 33 31 30 30 30 30   | 0000000000310000
00d0: 30 34 30 30 30 30 30 30 31 34 30 30 36 34 31 33   | 0400000014006413
00e0: 30 37 36 30 30 30 30 35 32 30 30 30 30 30 30 30   | 0760000520000000
00f0: 30 30 30 30 2e 2f 52 6f 73 65 2f 52 6f 73 65 2d   | 0000./Rose/Rose-
0100: 32 30 32 31 2d 30 32 2d 30 33 2d 30 33 2d 31 39   | 2021-02-03-03-19
0110: 2d 30 34 2d 38 30 30 2d 44 65 62 75 67 00 30 37   | -04-800-Debug.07
0120: 30 37 30 37 30 30 30 30 30 30 30 30 30 30 30 34   | 0707000000000004
0130: 31 30 30 36 34 34 30 30 30 30 30 30 30 30 30 30   | 1006440000000000
0140: 33 31 30 30 30 30 30 31 30 30 30 30 30 30 31 34   | 3100000100000014
0150: 30 30 36 34 31 33 30 37 36 30 30 30 30 36 35 30   | 0064130760000650
0160: 30 30 30 30 30 30 34 35 31 32 2e 2f 52 6f 73 65   | 0000004512./Rose
0170: 2f 52 6f 73 65 2d 32 30 32 31 2d 30 32 2d 30 33   | /Rose-2021-02-03
0180: 2d 30 33 2d 31 39 2d 30 34 2d 38 30 30 2d 44 65   | -03-19-04-800-De
0190: 62 75 67 2f 74 69 63 6b 65 74 2e 64 65 72 00 30   | bug/ticket.der.0
01a0: 82 09 46 16 04 49 4d 34 4d 02 01 00 31 82 02 19   | ..F..IM4M...1...
01b0: ff 84 ea 85 9c 42 82 02 10 30 82 02 0c 16 04 4d   | .....B...0.....M
01c0: 41 4e 42 31 82 02 02 ff 84 ea 85 9c 50 81 e8 30   | ANB1........P..0
01d0: 81 e5 16 04 4d 41 4e 50 31 81 dc ff 84 92 b9 86   | ....MANP1.......
01e0: 48 2a 30 28 16 04 42 4e 43 48 04 20 78 c4 ed d0   | H*0(..BNCH. x...
01f0: b4 24 c0 ad 7e a2 e1 4b 45 66 a2 f6 6d 91 ad 44   | .$..~..KEf..m..D
0200: 8c 62 2b c3 8e 7b 30 84 88 a7 06 9c ff 84 92 bd   | .b+..{0.........
0210: a4 44 0c 30 0a 16 04 42 4f 52 44 02 02 01 08 ff   | .D.0...BORD.....
0220: 84 9a 95 a0 4f 0b 30 09 16 04 43 45 50 4f 02 01   | ....O.0...CEPO..
0230: 01 ff 84 9a a1 92 50 0c 30 0a 16 04 43 48 49 50   | ......P.0...CHIP
0240: 02 02 20 06 ff 84 9a c1 a4 4f 0b 30 09 16 04 43   | .. ......O.0...C
0250: 50 52 4f 01 01 ff ff 84 9a cd 8a 43 0b 30 09 16   | PRO........C.0..
0260: 04 43 53 45 43 01 01 ff ff 84 aa 8d 92 44 11 30   | .CSEC........D.0
0270: 0f 16 04 45 43 49 44 02 07 05 08 68 36 04 00 1e   | ...ECID....h6...
0280: ff 85 9a 91 9e 4d 0b 30 09 16 04 53 44 4f 4d 02   | .....M.0...SDOM.
0290: 01 01 ff 87 9b c9 ec 6e 1e 30 1c 16 04 73 72 76   | .......n.0...srv
02a0: 6e 04 14 6a d3 36 cb 74 ab 9e 0f da 02 f4 c0 13   | n..j.6.t........
02b0: 0f fe 67 8b a9 c8 0d ff 87 93 ad de 73 81 81 30   | ..g.........s..0
02c0: 7f 16 04 72 6b 6f 73 31 77 ff 84 a2 9d a6 54 3a   | ..rkos1w.....T:
02d0: 30 38 16 04 44 47 53 54 04 30 85 42 79 a9 ac d9   | 08..DGST.0.By...
02e0: d3 5c 65 4b c7 d6 b6 ef ab ca 6f 99 65 c3 0e 8c   | .\eK......o.e...
02f0: d8 5f 15 c0 0c 5a ff 1c 6a 17 40 94 2e 79 3f 55   | ._...Z..j.@..y?U
0300: 3c 02 2c 02 2e b4 8f 23 2a d2 ff 84 aa ad 8a 59   | <.,....#*......Y
0310: 0b 30 09 16 04 45 4b 45 59 01 01 ff ff 84 aa c1   | .0...EKEY.......
0320: a4 4f 0b 30 09 16 04 45 50 52 4f 01 01 ff ff 84   | .O.0...EPRO.....
0330: aa cd 8a 43 0b 30 09 16 04 45 53 45 43 01 01 ff   | ...C.0...ESEC...
0340: ff 87 9b 89 c8 31 81 81 30 7f 16 04 73 62 64 31   | .....1..0..sbd1
0350: 31 77 ff 84 a2 9d a6 54 3a 30 38 16 04 44 47 53   | 1w.....T:08..DGS
0360: 54 04 30 ee e4 b4 e3 32 32 a1 4f 9c 61 0a 96 c9   | T.0....22.O.a...
0370: 6d 4e ea 46 83 35 c8 cb 2a 22 56 9e 6c d2 bd 6a   | mN.F.5..*"V.l..j
0380: ab 93 ca dc c6 22 79 86 16 3e 15 59 a8 1e 00 19   | ....."y..>.Y....
0390: 85 f3 5b ff 84 aa ad 8a 59 0b 30 09 16 04 45 4b   | ..[.....Y.0...EK
03a0: 45 59 01 01 ff ff 84 aa c1 a4 4f 0b 30 09 16 04   | EY........O.0...
[... 232 lines cut ...]
1230: 41 4d 20 3d 20 7b 20 20 20 20 20 20 20 20 20 20   | AM = {
1240: 20 20 30 20 42 20 7d 5c 6e 22 2c 0a 20 20 20 20   |   0 B }\n",.
1250: 22 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20   | "
1260: 20 20 20 20 4c 6f 67 5f 44 72 70 64 5f 51 46 75   |     Log_Drpd_QFu
1270: 6c 20 3d 20 7b 20 20 20 20 20 20 20 20 20 20 20   | l = {
[... cut ...]
1350: 30 30 30 31 33 30 30 30 30 30 30 30 30 30 30 30   | 0001300000000000
1360: 54 52 41 49 4c 45 52 21 21 21 00 00 00 00 00 00   | TRAILER!!!......
1370: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   | ................
1380: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   | ................
1390: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   | ................
13a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   | ................
13b0: 00 00                                             | ..

mroger-tech, 2021-02-03:
> The device doesn't seem to require any reply, but I still think we should handle this properly. Can someone of you guys test this patch, run a restore (no need for `-d`!!) and paste the hexdump output here? Thanks
> https://gist.github.com/nikias/df2c8cdca3515985d1ec9bcb76cae291

I test the patch,the hexdump as bellow:

[hexdump.log](https://github.com/libimobiledevice/idevicerestore/files/5915924/hexdump.log)

nikias, 2021-02-03:
OK thanks. It's plain CPIO data, uncompressed, without any additional headers whatsoever, so I guess I will have to read until the remote side closes the connection and use a reasonable timeout.

mroger-tech, 2021-02-03:
Thanks too,hope to help solve this problem.

nikias, 2021-02-03:
It should actually restore already with the patch, just be really verbose while doing it. Or did it fail again?

mroger-tech, 2021-02-03:
fail again,error message as bellow(a part of):

...
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received Rose ticket
Extracting ftab.bin...
NOTE: Build identity does not have a 'Rap,RestoreRTKitOS' component.
Sending FirmwareResponse data now...
Done sending FirmwareUpdater data
Updating Rose (65)
restore_handle_baseband_updater_output_data: Initial read returned 78 bytes:
restore_handle_baseband_updater_output_data: Second read returned 5042 bytes:
Got status message
**Status: Verification Error**
Log is available:
[03:12:34.0946-GMT]{4>8} CHECKPOINT NOTICE: Image4 device: AP nonce clearable
entering ramrod_clear_ap_nonce
[03:12:34.0957-GMT]{4>8} CHECKPOINT NOTICE: AP nonce consumed
[03:12:34.0957-GMT]{4>8} CHECKPOINT NOTICE: Pre-existing NVRAM variable: auto-boot=false
[03:12:34.0957-GMT]{4>8} CHECKPOINT NOTICE: Pre-existing NVRAM variable: restore-outcome=initial_monitor_no_return
[03:12:34.0957-GMT]{4>8} CHECKPOINT PROGRESS: START (unknown) -> (initial_engine_no_return)
[03:12:34.0957-GMT]{4>8} CHECKPOINT NOTICE: NVRAM access available on initial check
restore-outcome = initial_engine_no_return
executing /usr/sbin/nvram restore-outcome=initial_engine_no_return
[03:12:34.0965-GMT]{4>8} CHECKPOINT BEGIN: MAIN:[0x0400] umask
restore-step-ids = {0x11030400:1}
restore-step-names = {0x11030400:umask}
restore-step-uptime = 0
restore-step-user-progress = -1

...

mroger-tech, 2021-02-03:
It seems that the device restore to normal mode successfully,,but  the device tell me failed finally ,error message as bellow(a part of):

...
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received Rose ticket
Extracting ftab.bin...
NOTE: Build identity does not have a 'Rap,RestoreRTKitOS' component.
Sending FirmwareResponse data now...
Done sending FirmwareUpdater data
Updating Rose (65)
restore_handle_baseband_updater_output_data: Initial read returned 78 bytes:
restore_handle_baseband_updater_output_data: Second read returned 5042 bytes:
Got status message
**Status: Verification Error**
Log is available:
[03:12:34.0946-GMT]{4>8} CHECKPOINT NOTICE: Image4 device: AP nonce clearable
entering ramrod_clear_ap_nonce
[03:12:34.0957-GMT]{4>8} CHECKPOINT NOTICE: AP nonce consumed
[03:12:34.0957-GMT]{4>8} CHECKPOINT NOTICE: Pre-existing NVRAM variable: auto-boot=false
[03:12:34.0957-GMT]{4>8} CHECKPOINT NOTICE: Pre-existing NVRAM variable: restore-outcome=initial_monitor_no_return
[03:12:34.0957-GMT]{4>8} CHECKPOINT PROGRESS: START (unknown) -> (initial_engine_no_return)
[03:12:34.0957-GMT]{4>8} CHECKPOINT NOTICE: NVRAM access available on initial check
restore-outcome = initial_engine_no_return
executing /usr/sbin/nvram restore-outcome=initial_engine_no_return
[03:12:34.0965-GMT]{4>8} CHECKPOINT BEGIN: MAIN:[0x0400] umask
restore-step-ids = {0x11030400:1}
restore-step-names = {0x11030400:umask}
restore-step-uptime = 0
restore-step-user-progress = -1

...

mroger-tech, 2021-02-03:
I'll  test it again later

mroger-tech, 2021-02-03:
I test again,and failed again,the error message the same as above.

The device is still in recovery mode.

mroger-tech, 2021-02-03:
[1.log](https://github.com/libimobiledevice/idevicerestore/files/5917549/1.log)
 
this is the complete log.

nikias, 2021-02-03:
@mroger-tech I updated the patch: https://gist.github.com/nikias/df2c8cdca3515985d1ec9bcb76cae291
I don't see that the host sends a reply after it received the cpio data... Maybe there's a different reason for it to fail, not really sure. Maybe the updated patch makes it work for some reason.

mroger-tech, 2021-02-04:
I used the new patch,but fail again.

...
About to send SystemImageRootHash...
Extracting 018-01796-018.dmg.root_hash...
Personalizing IMG4 component SystemVolume...
Sending SystemImageRootHash now...
Done sending SystemImageRootHash
Fixing up /var (17)
Creating system key bag (50)
Modifying persistent boot-args (25)
 ERROR: Unable to receive message from FDR 0359B7F8 (-2). 0/2 bytes
Updating Savage (60)
Updating Savage (60)
Updating Rose (65)
Sending Rose TSS request...
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received Rose ticket
Extracting ftab.bin...
NOTE: Build identity does not have a 'Rap,RestoreRTKitOS' component.
Sending FirmwareResponse data now...
Done sending FirmwareUpdater data
Updating Rose (65)
**Wrote baseband updater output data to updater_output-00008101-000508683604001E.cpi**
Got status message
**Status: Verification Error**
Log is available:
[01:20:10.0717-GMT]{4>7} CHECKPOINT NOTICE: Image4 device: AP nonce clearable
entering ramrod_clear_ap_nonce
[01:20:10.0728-GMT]{4>7} CHECKPOINT NOTICE: AP nonce consumed
[01:20:10.0729-GMT]{4>7} CHECKPOINT NOTICE: Pre-existing NVRAM variable: auto-boot=false
[01:20:10.0729-GMT]{4>7} CHECKPOINT NOTICE: Pre-existing NVRAM variable: restore-outcome=initial_monitor_no_return
[01:20:10.0729-GMT]{4>7} CHECKPOINT PROGRESS: START (unknown) -> (initial_engine_no_return)
[01:20:10.0729-GMT]{4>7} CHECKPOINT NOTICE: NVRAM access available on initial check
restore-outcome = initial_engine_no_return
executing /usr/sbin/nvram restore-outcome=initial_engine_no_return
...

nikias, 2021-02-04:
Can you attach the `updater_output-00008101-000508683604001E.cpi` file? (I made a small mistake in the patch, the filename should be `.cpio` but I allocate 1 byte less than required). Maybe that contains information on what actually went wrong.

mroger-tech, 2021-02-04:
[updater_output-00008101-000508683604001E.zip](https://github.com/libimobiledevice/idevicerestore/files/5922658/updater_output-00008101-000508683604001E.zip)

mroger-tech, 2021-02-04:
@nikias hi friend,i found out a strange problem(maybe related to this problem ),the tss request data maybe wrong(I captured the dataof USB transfers of 3utool for comparison,3utool can restore success),
for example,the Rose Tss request data is:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
        <key>@Locality</key>
        <string>en_US</string>
        <key>@HostPlatformInfo</key>
        <string>windows</string>
        <key>@VersionInfo</key>
        <string>libauthinstall-698.0.5</string>
        <key>@UUID</key>
        <string>24E22B57-124A-8D33-F504-E37E2E1431B0</string>
        <key>@BBTicket</key>
        <true/>
        <key>@Rap,Ticket</key>
        <true/>
        **<key>Rap,BoardID</key>
        <integer>2048</integer>
        <key>Rap,ChipID</key>
        <integer>1568</integer>
        <key>Rap,ECID</key>
        <integer>-7180548491729411072</integer>**
        <key>Rap,Nonce</key>
        <data>
        od8LFAdlvwFTKU+HNxYpxyJzo00aZe0x9QMITCnGV1Y=
        </data>
        <key>Rap,ProductionMode</key>
        <true/>
        <key>Rap,SecurityDomain</key>
        <integer>1</integer>
        <key>Rap,SecurityMode</key>
        <true/>
        <key>Rap,RTKitOS</key>
        <dict>
                <key>Digest</key>
                <data>
                Ojg8agH2lJFhPLy3vVUcPvoUqLD8Q9aFAandVcABDSVdtsH4PfUq6m0o9qNk
                xxNn
                </data>
                <key>Trusted</key>
                <true/>
                <key>EPRO</key>
                <true/>
                <key>ESEC</key>
                <true/>
        </dict>
        <key>Rap,SoftwareBinaryDsp1</key>
        <dict>
                <key>Digest</key>
                <data>
                Lal9KmBu1cKR13ccw1QYQpevyrLwIrQ3S6nDLdSCcnvbmhjwXXslAeb5y87C
                sCCD
                </data>
                <key>Trusted</key>
                <true/>
                <key>EPRO</key>
                <true/>
                <key>ESEC</key>
                <true/>
        </dict>
</dict>
</plist>

**but after i send this request ,the TSS server returned: STATUS=94&MESSAGE=This device isn't eligible for the requested build.
so,i changed the request data to:**
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
[... 139 lines cut ...]
        <integer>57601</integer>
        <key>BMU,Nonce</key>
        <data>
        BI31dyL9YbQxMDt+mlYa10BrZg5Cws/fhlO5W4pUIBQ=
        </data>
        <key>BMU,ProductionMode</key>
        <true/>
        <key>BMU,UniqueID</key>
        <integer>268440677</integer>
        <key>BMU,DigestMap</key>
        <dict>
                <key>Digest</key>
                <data>
                TdQ9KoFBR9G+fxYjFnsm4Y7/mcn9aBVwPafBUPxYMa0=
                </data>
        </dict>
        <key>BMU,FirmwareMap</key>
        <dict>
                <key>Digest</key>
                <data>
                vvGN/QKrcwjBBEldNOtqu4tMfwXzCiWCi1WIoqtSdcU=
                </data>
        </dict>
</dict>
</plist>

mroger-tech, 2021-02-04:
@nikias hi friend,i found out a strange problem(maybe related to this problem ),the tss request data maybe wrong(**I captured the data of USB transfers of 3utool for comparison,3utool can restore success**),
for example,the Rose Tss request data is:
```
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
        <key>@Locality</key>
        <string>en_US</string>
        <key>@HostPlatformInfo</key>
        <string>windows</string>
        <key>@VersionInfo</key>
        <string>libauthinstall-698.0.5</string>
        <key>@UUID</key>
        <string>24E22B57-124A-8D33-F504-E37E2E1431B0</string>
        <key>@BBTicket</key>
        <true/>
        <key>@Rap,Ticket</key>
        <true/>
        <key>Rap,BoardID</key>
        <integer>2048</integer>       //this value is strange
        <key>Rap,ChipID</key>
        <integer>1568</integer>       //this value is strange
        <key>Rap,ECID</key>
        <integer>-7180548491729411072</integer>    //this value is strange
        <key>Rap,Nonce</key>
        <data>
        od8LFAdlvwFTKU+HNxYpxyJzo00aZe0x9QMITCnGV1Y=
        </data>
        <key>Rap,ProductionMode</key>
        <true/>
        <key>Rap,SecurityDomain</key>
        <integer>1</integer>
        <key>Rap,SecurityMode</key>
        <true/>
        <key>Rap,RTKitOS</key>
        <dict>
                <key>Digest</key>
                <data>
                Ojg8agH2lJFhPLy3vVUcPvoUqLD8Q9aFAandVcABDSVdtsH4PfUq6m0o9qNk
                xxNn
                </data>
                <key>Trusted</key>
                <true/>
                <key>EPRO</key>
                <true/>
                <key>ESEC</key>
                <true/>
        </dict>
        <key>Rap,SoftwareBinaryDsp1</key>
        <dict>
                <key>Digest</key>
                <data>
                Lal9KmBu1cKR13ccw1QYQpevyrLwIrQ3S6nDLdSCcnvbmhjwXXslAeb5y87C
                sCCD
                </data>
                <key>Trusted</key>
                <true/>
                <key>EPRO</key>
                <true/>
                <key>ESEC</key>
                <true/>
        </dict>
</dict>
</plist>
```

**but after i send this request ,the TSS server returned: STATUS=94&MESSAGE=This device isn't eligible for the requested build.
so,i changed the request data to:**
```
[... 147 lines cut ...]
        <key>BMU,Nonce</key>
        <data>
        BI31dyL9YbQxMDt+mlYa10BrZg5Cws/fhlO5W4pUIBQ=
        </data>
        <key>BMU,ProductionMode</key>
        <true/>
        <key>BMU,UniqueID</key>
        <integer>268440677</integer>
        <key>BMU,DigestMap</key>
        <dict>
                <key>Digest</key>
                <data>
                TdQ9KoFBR9G+fxYjFnsm4Y7/mcn9aBVwPafBUPxYMa0=
                </data>
        </dict>
        <key>BMU,FirmwareMap</key>
        <dict>
                <key>Digest</key>
                <data>
                vvGN/QKrcwjBBEldNOtqu4tMfwXzCiWCi1WIoqtSdcU=
                </data>
        </dict>
</dict>
</plist>
```

nikias, 2021-02-04:
@mroger-tech are you running this on Windows by any chance? My guess is that libplist has an issue based on how it was compiled. Since the values are sent over by the device, they should be correct.
When building libplist, make sure to also run `make check` to have it run test cases that would (hopefully) catch any issues.