libimobiledevice/idevicerestore #297: Error while running make

SkywalkerSW5, 2019-11-08:
While I run make, it gives me this output:

`
make  all-recursive
make[1]: Entering directory '/home/pi/idevicerestore'
Making all in src
make[2]: Entering directory '/home/pi/idevicerestore/src'
  CC       idevicerestore-normal.o
normal.c:31:10: fatal error: libimobiledevice/preboard.h: No such file or directory
 #include <libimobiledevice/preboard.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [Makefile:665: idevicerestore-normal.o] Error 1
make[2]: Leaving directory '/home/pi/idevicerestore/src'
make[1]: *** [Makefile:433: all-recursive] Error 1
make[1]: Leaving directory '/home/pi/idevicerestore'
make: *** [Makefile:365: all] Error 2

`

Can anyone tell me how to fix this? I am using this on a Raspberry Pi running Raspbian.

Thanks in advance.

JoshGuyette, 2019-11-12:
I'm having the same issue, also on Raspberry Pi

JoshGuyette, 2019-11-12:
@SkywalkerSW5 You need to replace your libimobiledevice for idevicerestore to build...

Fix libimobiledevice build issue…

sudo apt-get install libplist++
sudo apt-get remove libimobiledevice-dev
git clone https://github.com/libimobiledevice/libimobiledevice.git
git revert -n 7dbc17f2c763a7b18bc1a867f39918c26bb5e9e5
git revert -n d6ca8fca11c6aa86ed5fbcf943e1580eb8587538
git revert -n fe3ef93359d9a1105b920d4a7a342706bd3b32c0
git revert -n 6a73100414f66d12287105a322529c488f32182e
git revert -n 74dfc6485b61c76775c332f0ae6c65a28bcced9c
git revert -n 953c13961071a9836430a814159aae33502e2e0c
git revert -n 5390060ec90bed0f565698a389cebc57c09c3ac8
./autogen.sh
make
sudo make install

closed 2019-11-15 by SkywalkerSW5

joshpeterson1, 2020-02-23:
@JoshGuyette I'm running into the same issue on Ubuntu 18.0.4.4 LTS, any ideas?

I tried the above, but ran into a git conflict...

Thanks in advance