libimobiledevice/idevicerestore #304: Stuck in Recovery Mode ( Updating NAND Firmware (58)  )

tharyckgusmao, 2019-12-24:
Hi friends, I need some help or some guidance.

My phone restarted unexpectedly, and got stuck on the recovery screen, I tested various methods to recover like itunes, reiboot and idevicerestore.

Itunes stuck in error 9.
Reiboot timeout.
idevicerestore Updating NAND Firmware (58)
No data to read (timeout)
FDR 0x563629f4aa20 timeout waiting for command
FDR 0x563629f4aa20 waiting for message ...
No data to read (timeout)
No data to read (timeout)
FDR 0x563629f4aa20 timeout waiting for command
FDR 0x563629f4aa20 waiting for message ...
No data to read (timeout)
ERROR: Could not read data (-256). Aborting.
ERROR: Unable to receive message from FDR 0x563629f4aa20

My device is an iphone 6s, was in version 13.2.3 with checkra1n, after the restart I got stuck in recovery mode ...

Someone went through this, or have any suggestions, follow my idevicerestore log. 
Thanks!!

[fulllog.txt](https://github.com/libimobiledevice/idevicerestore/files/3998854/fulllog.txt)

jroy2023, 2020-12-15:
Hello, did you find a solution to this? I have an iPhone 6s and now is stuck in DFU mode, iTunes constantly gives me Error 9 and 3UTools fails at 20% while doing 'Updating NAND firmware' and gives me the message "Error: Unable to restore iDevice(-1)"
Thanks!

tharyckgusmao, 2020-12-15:
hello friend, I didn't find a solution by software, apparently there is a problem in ipsone 6s, where the processor enters a cold soldering stage, I actually sold it to remove parts.

multihad, 2021-05-18:
MacBook Pro 
Found device in Recovery mode
Identified device as j680ap, iBridge2,3
Extracting BuildManifest from IPSW
Product Version: 5.3
Product Build: 18P4556 Major: 18
INFO: device serial number is  
Device supports Image4: true
Variant: Customer Upgrade Install (IPSW)
This restore will update your device without erasing user data.
Checking IPSW for required components...
All required components found in IPSW
Using cached filesystem from '/Users/macbookpro/Desktop/iBridge2,1,iBridge2,10,iBridge2,12,iBridge2,14,iBridge2,15,iBridge2,16,iBridge2,19,iBridge2,20,iBridge2,21,iBridge2,22,iBridge2,3,iBridge2,4,iBridge2,5,iBridge2,6,iBridge2,7,iBridge2,8_5.3_18P4556_Restore/038-69925-193.dmg'
Found ECID 5928812160352294
Getting ApNonce in recovery mode... 69 18 92 4d e1 13 f2 14 51 64 44 b9 c9 b8 92 a4 57 ac d6 62 54 3c 62 ef e8 b4 42 1b a6 c6 21 25 
Trying to fetch new SHSH blob
Getting SepNonce in recovery mode... 4a 77 13 e1 3d a6 01 62 ae 37 e5 7d 0d 38 f4 2d d1 a0 77 05 
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received SHSH blobs
Extracting iBEC.j680.RELEASE.im4p...
Personalizing IMG4 component iBEC...
Sending iBEC (1189528 bytes)...
Recovery Mode Environment:
iBoot build-version=iBoot-6723.101.4
iBoot build-style=RELEASE
Extracting 038-69900-193.dmg.trustcache...
Personalizing IMG4 component RestoreTrustCache...
Sending RestoreTrustCache (10182 bytes)...
ramdisk-size=0x8000000
Extracting 038-69900-193.dmg...
Personalizing IMG4 component RestoreRamDisk...
Sending RestoreRamDisk (98285100 bytes)...
Extracting DeviceTree.j680ap.im4p...
Personalizing IMG4 component RestoreDeviceTree...
Sending RestoreDeviceTree (35665 bytes)...
Extracting sep-firmware.j680.RELEASE.im4p...
Personalizing IMG4 component RestoreSEP...
Sending RestoreSEP (4114154 bytes)...
Extracting kernelcache.release.ibridge2p...
Personalizing IMG4 component RestoreKernelCache...
Sending RestoreKernelCache (10026706 bytes)...
Waiting for device to enter restore mode...
About to restore device... 
Connecting now...
Connected to com.apple.mobile.restored, version 15
Device 00008012-0015103926C20026 has successfully entered restore mode
Hardware Information:
BoardID: 11
ChipID: 32786
UniqueChipID: 5928812160352294
ProductionMode: true
Previous restore exit status: 0x100
Starting FDR listener thread
About to send RootTicket...
Sending RootTicket now...
Done sending RootTicket
Waiting for NAND (28)
Updating NAND Firmware (58)
Got status message
Status: Disk Failure
Log is available:
[01:21:23.0095-GMT]{3>8} CHECKPOINT NOTICE: Image4 device: AP nonce clearable
entering ramrod_clear_ap_nonce
[01:21:23.0348-GMT]{3>8} CHECKPOINT NOTICE: AP nonce consumed
[01:21:23.0348-GMT]{3>8} CHECKPOINT NOTICE: Pre-existing NVRAM variable: auto-boot=false
[01:21:23.0348-GMT]{3>8} CHECKPOINT NOTICE: Pre-existing NVRAM variable: restore-outcome=initial_monitor_no_return
[01:21:23.0349-GMT]{3>8} CHECKPOINT PROGRESS: START (unknown) -> (initial_engine_no_return)
[01:21:23.0349-GMT]{3>8} CHECKPOINT NOTICE: NVRAM access available on initial check
restore-outcome = initial_engine_no_return
[... 351 lines cut ...]
restore-step-user-progress = 0
[01:21:32.0617-GMT]{3>8} CHECKPOINT END: RESTORED:[0x1613] cleanup_recovery_os_volume
restore-step-ids = {0x1103060E:17}
restore-step-names = {0x1103060E:update_NAND_firmware}
restore-step-uptime = 10
restore-step-user-progress = 0
[01:21:32.0617-GMT]{3>8} CHECKPOINT BEGIN: RESTORED:[0x0647] cleanup_check_result
restore-step-ids = {0x1103060E:17;0x11030647:20}
restore-step-names = {0x1103060E:update_NAND_firmware;0x11030647:cleanup_check_result}
restore-step-uptime = 10
restore-step-user-progress = 0
[01:21:32.0618-GMT]{3>8} CHECKPOINT END: RESTORED:[0x0647] cleanup_check_result
restore-step-ids = {0x1103060E:17}
restore-step-names = {0x1103060E:update_NAND_firmware}
restore-step-uptime = 10
restore-step-user-progress = 0
[01:21:32.0618-GMT]{3>8} CHECKPOINT BEGIN: RESTORED:[0x0648] cleanup_send_final_status
restore-step-ids = {0x1103060E:17;0x11030648:21}
restore-step-names = {0x1103060E:update_NAND_firmware;0x11030648:cleanup_send_final_status}
restore-step-uptime = 10
restore-step-user-progress = 0

ERROR: Unable to successfully restore device
ERROR: Unable to restore device
macbookpro@MacBook-Pro-2 ~ %

xros, 2024-01-04:
Hi did you solve the problem? I had the similar issue recently when upgrading iOS from 16.6.1 to iOS 17.2.1 using iMazing.

Here's a short version of the error log

```
17:05:40.406: <Restore Device 012ACCF0>: operation 58 progress -1
17:05:40.406:  正在更新S3E固件
17:05:46.345: Completed checkpoint id: 0x60E (update_NAND_firmware)
17:05:46.345: Checkpoint 0x60E (update_NAND_firmware) failed with result: 6
17:05:46.345: Started checkpoint id: 0x67C (cleanup_boot_command)
17:05:46.345: Completed checkpoint id: 0x67C (cleanup_boot_command)
17:05:46.345: Started checkpoint id: 0x1613 (cleanup_recovery_os_volume)
17:05:46.345: Completed checkpoint id: 0x1613 (cleanup_recovery_os_volume)
17:05:46.345: Started checkpoint id: 0x647 (cleanup_check_result)
17:05:46.345: Completed checkpoint id: 0x647 (cleanup_check_result)
17:05:46.409: Started checkpoint id: 0x648 (cleanup_send_final_status)
17:05:46.409: device returned CFError with code 6
17:05:46.409: dumping CFError returned by restored:
17:05:46.409: CFError domain:AMRestoreErrorDomain code:6 description:failed to update NAND firmware
17:05:46.409: CFError domain:RamrodErrorDomain code:7 description:update_NAND: Update NAND - FTL version mismatch. Erase install required
17:05:46.409: Sending acknowledgement that final status was received
17:05:46.534: unable to parse regex '\"ApECID\"[^
17:05:46.534: ]*value = [+-][0-9<>IMEI]*': 4
```

In short, NAND firmware failed to update. **Update NAND - FTL version mismatch. Erase install required**

I assume this is a hardware issue since the iPhone XR 256G is originally an iPhone XR 64G. Because the NAND chip had been replaced (expanded)... But how to bypass this to upgrade the iOS? The iPhone didn't boot from iOS 16.6.1 before. It keeps go back between **recovery mode** and **dfu mode**.