libimobiledevice/idevicerestore #305: Device failing to enter restore mode

Duplicitious, 2019-12-25:
I'm trying to erase my current iphone 7 and reinstall the newer firmware but for some reason each time I try it says the device failed to enter restore mode.

```
$ sudo idevicerestore -e iPhone_4.7_P3_13.3_17C54_Restore_GSM.ipsw 
Found device in Recovery mode
Identified device as d101ap, iPhone9,3
Extracting BuildManifest from IPSW
Product Version: 13.3
Product Build: 17C54 Major: 17
INFO: device serial number is F4GTFCFXHG7H
Device supports Image4: true
Variant: Customer Erase Install (IPSW)
This restore will erase your device data.
################################ [ WARNING ] #################################
# You are about to perform an *ERASE* restore. ALL DATA on the target device #
# will be IRREVERSIBLY DESTROYED. If you want to update your device without  #
# erasing the user data, hit CTRL+C now and restart without -e or --erase    #
# command line switch.                                                       #
# If you want to continue with the ERASE, please type YES and press ENTER.   #
##############################################################################
> YES
Checking IPSW for required components...
All required components found in IPSW
Extracting filesystem from IPSW: 038-02762-069.dmg
[==================================================] 100.0%
Found ECID 6246952625865510
Getting ApNonce in recovery mode... 6f f6 00 f8 31 fe ae 58 ad 05 6b b9 6b 62 4a cd 78 6d 29 7c 4d 99 07 a2 35 3a 7a b3 2a c0 c5 27 
Trying to fetch new SHSH blob
Getting SepNonce in recovery mode... 17 a0 42 0e 30 46 b4 d3 f5 4a 9a 2f be 89 61 6e a4 4c aa b4 
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received SHSH blobs
Extracting iBEC.d10.RELEASE.im4p...
Personalizing IMG4 component iBEC...
Sending iBEC (474503 bytes)...
Recovery Mode Environment:
iBoot build-version=iBoot-5540.60.11
iBoot build-style=RELEASE
Sending RestoreLogo...
Extracting applelogo@2x~iphone.im4p...
Personalizing IMG4 component RestoreLogo...
Sending RestoreLogo (13090 bytes)...
Extracting 038-02406-072.dmg.trustcache...
Personalizing IMG4 component RestoreTrustCache...
Sending RestoreTrustCache (10949 bytes)...
ramdisk-size=0x20000000
Extracting 038-02406-072.dmg...
Personalizing IMG4 component RestoreRamDisk...
Sending RestoreRamDisk (102052293 bytes)...
Extracting DeviceTree.d101ap.im4p...
Personalizing IMG4 component RestoreDeviceTree...
Sending RestoreDeviceTree (162899 bytes)...
Extracting kernelcache.release.iphone9...
Personalizing IMG4 component RestoreKernelCache...
Sending RestoreKernelCache (14205135 bytes)...
Waiting for device to enter restore mode...
ERROR: Device failed to enter restore mode.
```

Codyp1996, 2019-12-30:
Im having the same issue with an Apple Tv 4. Have you figured anything out? Im running on linux.

nikias, 2019-12-30:
I bet that usbmuxd is not running. Try running it in foreground in another terminal to see if that helps: ```sudo usbmuxd -f```
I am seeing some issues with systemd recently...

Duplicitious, 2019-12-30:
Yep that fixed it @nikias . Also do you know how to reset the devices apple ID?

nikias, 2019-12-30:
If you are talking about iCloud lock aka activation lock it is something that I can't help you with.

JosephBellfixit, 2020-03-17:
> I bet that usbmuxd is not running. Try running it in foreground in another terminal to see if that helps: `sudo usbmuxd -U usbmux -f`
> I am seeing some issues with systemd recently...

> I bet that usbmuxd is not running. Try running it in foreground in another terminal to see if that helps: `sudo usbmuxd -U usbmux -f`
> I am seeing some issues with systemd recently...

Great help, running it in the background solved the issue straight away.

closed 2020-03-24 by Duplicitious