libimobiledevice/idevicerestore #308: Support recovery syslog for diagnostic reasons

rickmark, 2020-01-01:
During the restore process, while the ramdisk is loaded idevicerecovery will connect to `com.appple.restored`.  This misses out on the diagnostic logs that are available by first connecting to `com.apple.syslog_relay` in the recovery image (this is without lockdownd being present) to provide additional diagnostics during a device recovery/restore.

Proposal is to add `--syslog /some/path/log.txt` as a command line switch which will pipe output to that path under the advanced flags...

R

rickmark, 2020-01-01:
Work underway on this branch:

https://github.com/rickmark/idevicerestore/tree/syslog_support

nikias, 2020-01-01:
Oh nice find! Happy new year!

rickmark, 2020-01-04:
Happy new year, I can't seem to assign this to myself?  (Also planning to create an issue to validate T2 restore from ARM/RasbperryPi)