libimobiledevice/idevicerestore #315: iPhone 3,1 Custom IPSW - Unable to connect to restore mode device

cyckl, 2020-02-08:
I'm trying to downgrade but my iPhone 3,1 has failed to restore with any custom firmware (ch3rryflower, Pluvia). It seems to always fail to `connect to restore mode device`

The iPhone gets stuck [here](https://user-images.githubusercontent.com/26363850/74093649-abb59300-4a89-11ea-8277-82af4e4a9dbf.jpg)

I've pwned the DFU using limera1n, SHAtter, and kDFU to no avail.

[Here](https://github.com/libimobiledevice/idevicerestore/files/4175627/idevicerestore.txt) is the output from the restore

nikias, 2020-02-09:
Do you have usbmuxd set up and running? It is required so idevicerestore can connect in restore mode.

cyckl, 2020-02-09:
I do, I figured that it could possibly stop running during the process so I ran it in the foreground and it was running the entire time.

nikias, 2020-02-09:
if you run it in verbose mode you can see if it detects anything. Also try running `lsusb` to see if it shows any Apple device when it stalls like in the photo you sent

cyckl, 2020-02-10:
Sorry, I was a bit confused writing the previous comment. I've been experimenting on two systems
- A macOS 10.13 VM
- Linux on baremetal

The macOS VM spits out `Cannot connect to restore mode device`
On Linux, the output spits out the following

**idevicerestore output**
```
Sending iBSS (67888 bytes)...
[==================================================] 100.0%
Waiting for device to disconnect...
ERROR: Device did not disconnect. Possibly invalid iBSS. Reset device and try again.
ERROR: Unable to place device into recovery mode from DFU mode
```

**usbmuxd output**
```
[20:43:48.908][4] New client on fd 12 (pid 7301)
[20:43:48.908][5] Client command in fd 12 len 426 ver 1 msg 8 tag 1
[20:43:48.908][5] send_pkt fd 12 tag 1 msg 8 payload_length 278
[20:43:48.908][5] Client 12 now LISTENING
[20:44:09.643][4] Client 12 connection closed
[20:44:09.644][4] Disconnecting client fd 12
```
Apologies for the more complicated circumstances.