libimobiledevice/idevicerestore #320: iDevice can't enter restore mode (?)

TuxPepe1893, 2020-02-25:
When I use itunes everything works fine however in idevicerestore the following issue appears:

Issue Appeared on the iphone SE And Apple TV 3rd gen
log:

Found device in Recovery mode
Identified device as n69ap, iPhone8,4
Extracting BuildManifest from IPSW
Product Version: 13.3.1
Product Build: 17D50 Major: 17
INFO: device serial number is F17S6PUPH2Y7
Device supports Image4: true
Variant: Customer Erase Install (IPSW)
This restore will erase your device data.
################################ [ WARNING ] #################################
# You are about to perform an *ERASE* restore. ALL DATA on the target device #
# will be IRREVERSIBLY DESTROYED. If you want to update your device without  #
# erasing the user data, hit CTRL+C now and restart without -e or --erase    #
# command line switch.                                                       #
# If you want to continue with the ERASE, please type YES and press ENTER.   #
##############################################################################
> YES
Checking IPSW for required components...
All required components found in IPSW
Using cached filesystem from '/media/david1893/LocalBackup/ipsw/iPhone_4.0_64bit_13.3.1_17D50_Restore/038-13562-056.dmg'
Found ECID 6034309744026554
Getting ApNonce in recovery mode... 3a 88 b7 c3 80 2f 2f 05 10 ab c4 32 10 4a 15 eb d8 bd 71 54 
Trying to fetch new SHSH blob
Getting SepNonce in recovery mode... 20 2f fa cd 91 a4 06 d6 b2 75 db 35 3a ca e4 60 c2 df da 49 
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received SHSH blobs
Extracting iBEC.n69.RELEASE.im4p...
Personalizing IMG4 component iBEC...
Sending iBEC (389870 bytes)...
Recovery Mode Environment:
iBoot build-version=iBoot-5540.80.2
iBoot build-style=RELEASE
Sending RestoreLogo...
Extracting applelogo@2x~iphone.im4p...
Personalizing IMG4 component RestoreLogo...
Sending RestoreLogo (12426 bytes)...
Extracting 038-13993-058.dmg.trustcache...
Personalizing IMG4 component RestoreTrustCache...
Sending RestoreTrustCache (10285 bytes)...
ramdisk-size=0x20000000
Extracting 038-13993-058.dmg...
Personalizing IMG4 component RestoreRamDisk...
Sending RestoreRamDisk (102111533 bytes)...
Extracting DeviceTree.n69ap.im4p...
Personalizing IMG4 component RestoreDeviceTree...
Sending RestoreDeviceTree (143159 bytes)...
Extracting kernelcache.release.iphone8b...
Personalizing IMG4 component RestoreKernelCache...
Sending RestoreKernelCache (16867605 bytes)...
Waiting for device to enter restore mode...
ERROR: Device failed to enter restore mode.

TuxPepe1893, 2020-03-01:
i think i might close this issue... nobody replied lol

closed 2020-03-01 by TuxPepe1893

skylarmt, 2020-06-23:
I'm having this exact same issue with an iPad2,1.

```
skylar@skylar-Super-Server:~$ idevicerestore --latest --restore-mode
Found device in Recovery mode
Identified device as k93ap, iPad2,1
The following firmwares are currently being signed for iPad2,1:
  [1] 9.3.5 (build 13G36)
Select the firmware you want to restore: 1
Selected firmware 9.3.5 (build 13G36)
Verifying 'iPad2,1_9.3.5_13G36_Restore.ipsw'...
Checksum matches.
Extracting BuildManifest from IPSW
Product Version: 9.3.5
Product Build: 13G36 Major: 13
INFO: device serial number is CENSORED
Device supports Image4: false
Variant: Customer Upgrade Install (IPSW)
This restore will update your device without erasing user data.
Checking IPSW for required components...
All required components found in IPSW
Using cached filesystem from 'iPad2,1_9.3.5_13G36_Restore/058-49123-036.dmg'
Found ECID 159872040437
Getting ApNonce in recovery mode... 
Trying to fetch new SHSH blob
Getting SepNonce in recovery mode... 
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received SHSH blobs
Sending APTicket (2710 bytes)
Extracting iBEC.k93.RELEASE.dfu...
Personalizing IMG3 component iBEC...
reconstructed size: 297498
Sending iBEC (297498 bytes)...
Getting ApNonce in recovery mode... 38 6c af c3 89 2a 5d d9 9a 25 65 54 45 51 77 ba 14 68 dd 1f 
Trying to fetch new SHSH blob
Getting SepNonce in recovery mode... 
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received SHSH blobs
Sending APTicket (2732 bytes)
Recovery Mode Environment:
iBoot build-version=iBoot-2817.60.2
iBoot build-style=RELEASE
Sending RestoreLogo...
Extracting applelogo.s5l8940x.img3...
Not personalizing component RestoreLogo...
Sending RestoreLogo (5388 bytes)...
ramdisk-size=0x4000000
Extracting 058-49412-036.dmg...
Not personalizing component RestoreRamDisk...
Sending RestoreRamDisk (21602572 bytes)...
Extracting DeviceTree.k93ap.img3...
Not personalizing component RestoreDeviceTree...
Sending RestoreDeviceTree (78412 bytes)...
Extracting kernelcache.release.k93...
Not personalizing component RestoreKernelCache...
Sending RestoreKernelCache (10674764 bytes)...
Waiting for device to enter restore mode...
ERROR: Device failed to enter restore mode.
```

nikias, 2020-06-24:
Can you guys make sure that usbmuxd is running or configured to start when it sees a device in restore mode (usually happening through udev)?

randomize, 2020-06-27:
Same issue here, OS: Arch Linux
usbmuxd, libimobiledevice installed from Arch repositories, idevicerestore-git installed from source from AUR.

```
.......

Sending RestoreDeviceTree (206176 bytes)...
Extracting kernelcache.release.iphone11...
Personalizing IMG4 component RestoreKernelCache...
Sending RestoreKernelCache (14716513 bytes)...
Waiting for device to enter restore mode...
ERROR: Device failed to enter restore mode.
Please make sure that usbmuxd is running.

```

Tried running usbmuxd in foreground with `usbmuxd -f -v`

```
[09:51:42.017][3] usbmuxd v1.1.1 starting up
[09:51:42.017][4] Creating socket
[09:51:42.017][4] Initializing USB
[09:51:42.017][3] Using libusb 1.0.23
[09:51:42.024][4] Registering for libusb hotplug events
[09:51:42.024][4] 0 devices detected
[09:51:42.024][3] Initialization complete
^[[09:52:09.020][4] Client 11 accepted: idevicerestore[2254]
libusb: error [udev_hotplug_event] ignoring udev action bind
libusb: error [udev_hotplug_event] ignoring udev action bind
[09:52:40.852][4] Client 11 connection closed
[09:52:40.852][4] Client 11 is going to be disconnected: idevicerestore[2254]
[09:54:28.237][4] Client 11 accepted: idevicerestore[3350]
libusb: error [udev_hotplug_event] ignoring udev action bind
libusb: error [udev_hotplug_event] ignoring udev action bind
[09:55:10.283][4] Client 11 connection closed
[09:55:10.283][4] Client 11 is going to be disconnected: idevicerestore[3350]
[10:00:41.835][4] Client 11 accepted: idevicerestore[23026]
libusb: error [udev_hotplug_event] ignoring udev action bind
libusb: error [udev_hotplug_event] ignoring udev action bind
[10:01:09.964][4] Client 11 connection closed
[10:01:09.964][4] Client 11 is going to be disconnected: idevicerestore[23026]
libusb: error [udev_hotplug_event] ignoring udev action bind

```

Every time I connect device that is in restore mode this line appears
`libusb: error [udev_hotplug_event] ignoring udev action bind`

Also in logs you can see hom idevicerestore is connecting/disconnecting during some attempts to do `idevicerestore -l`

bmenrigh, 2020-07-29:
I'm having the same issue (with an iPhone 6s / Identified device as n71ap, iPhone8,1) and I've also run usbmuxd in foreground mode like @randomize did and I'm seeing basically the same output there.

I also ran idevicerestore in debug mode

>  $ sudo ./idevicerestore -l -d -e

```
[...]
Sending RestoreRamDisk (104186138 bytes)...
DEBUG: tss_response_get_path_by_entry: No entry 'RestoreDeviceTree' in TSS response
NOTE: No path for component RestoreDeviceTree in TSS, will fetch from build_identity
Extracting DeviceTree.n71ap.im4p...
Personalizing IMG4 component RestoreDeviceTree...
Tag found
Sending RestoreDeviceTree (143204 bytes)...
DEBUG: tss_response_get_path_by_entry: No entry 'RestoreKernelCache' in TSS response
NOTE: No path for component RestoreKernelCache in TSS, will fetch from build_identity
Extracting kernelcache.release.n71...
Personalizing IMG4 component RestoreKernelCache...
Tag found
Sending RestoreKernelCache (17217797 bytes)...
DEBUG: Waiting for device to disconnect...
irecv_event_cb: device 000007744b86e47e (udid: N/A) disconnected
Waiting for device to enter restore mode...
irecv_event_cb: device 000007744b86e47e (udid: N/A) connected in Recovery mode
ERROR: Device failed to enter restore mode.
Please make sure that usbmuxd is running.
```

Note at the end there it mentions `connected in Recovery mode` which is NOT "Restore" mode.

lsusb tells me the same:
`Bus 003 Device 040: ID 05ac:1281 Apple, Inc. Apple Mobile Device [Recovery Mode]`

I've looked through the code and it seems there is a distinction between MODE_RECOVER and MODE_RESTORE.  I don't know if it's unexpected or an issue that the phone says it's in recover mode, not restore mode.

UsamaSalafi, 2021-10-17:
my pc was infected and i was also experiencing same issue.
i've installed a fresh windows copy & redownloaded ipsw.
now my issue has been resolved.