libimobiledevice/idevicerestore #322: iPhone 11 models fail to update Rose Firmware data on iOS 13.4 beta

a-gunter, 2020-03-16:
Testing on Ubuntu 19.10 with latest commits in all relevant repositories. I've been able to reproduce this issue on both iPhone 11 and iPhone 11 Pro Max when trying to restore to iOS 13.4 beta 4 ([...]_13.4_17E5249a_Restore.ipsw). 

It seems like the Rap,RestoreRTKitOS node is missing from the Manifest.. 
> Received Rose ticket
> Extracting ftab.bin...
> ERROR: Unable to find component node for Rap,RestoreRTKitOS
> ERROR: Unable get path for 'Rap,RestoreRTKitOS' component
> common.c:supressed printing 102732 bytes plist...
> ERROR: restore_send_firmware_updater_data: Couldn't get Rose firmware data
> ERROR: Unable to send FirmwareUpdater data
> ERROR: Unable to successfully restore device
> FDR 0x557213e9e960 timeout waiting for command
> FDR 0x557213e9e960 waiting for message...
> 11:22:20 property_list_service.c:196 internal_plist_receive_timeout(): initial read failed!
> No data to read (timeout)
> FDR 0x557213e9e960 timeout waiting for command
> FDR 0x557213e9e960 terminating...
> 11:22:33 restore.c:348 restored_goodbye(): called

I compared the difference between the build identities on iOS 13.3.1 vs iOS 13.4 beta 4. Here's a diff comparing the two build identities if you're interested: [https://diffchecker.com/E7eL2GxY](https://diffchecker.com/E7eL2GxY)

There's some extra nodes in the new manifest, but I'm not sure how relevant they are.

a-gunter, 2020-03-24:
Issue persists with customer .ipsw version 13.4.

Removing the check for component Rap,RestoreRTKitOS in restore_get_rose_firmware_data seems to alleviate the issue, but I'm not sure if it's the correct fix.

Not sure what to do with the extra SE / Rose firmware nodes in the new manifest, so I left them as is.

nikias, 2020-04-07:
I guess this is what PR #323 does. But also I am not sure about the `SE,RapRTKitOS ` and `SE,RapSwBinDsp` entries, they have the same hash and mention the same file as `Rap,SoftwareBinaryDsp1 ` and `Rap,SoftwareBinaryDsp1` but lack the `Trusted: true` and the `RestoreRequestRules`, and also have the `Personalize` set to `false`.

nikias, 2020-04-12:
Fixed with f27d19e1c2e69dba271834d0c7a3a5be3a83e275. Reopen if still failing.

closed 2020-04-12 by nikias