libimobiledevice/idevicerestore #324: ERROR: Device did not reconnect in recovery mode. Possibly invalid iBEC. Reset device and try again.

7iswin, 2020-04-02:
root@Hacker:~# idevicerestore -e '/root/Downloads/iPhone_4.0_64bit_12.4.6_16G183_Restore.ipsw' 
Found device in Recovery mode
Identified device as n53ap, iPhone6,2
Extracting BuildManifest from IPSW
Product Version: 12.4.6
Product Build: 16G183 Major: 16
INFO: device serial number is DQGQRMFAFRC6
Device supports Image4: true
Variant: Customer Erase Install (IPSW)
This restore will erase your device data.
################################ [ WARNING ] #################################
# You are about to perform an *ERASE* restore. ALL DATA on the target device #
# will be IRREVERSIBLY DESTROYED. If you want to update your device without  #
# erasing the user data, hit CTRL+C now and restart without -e or --erase    #
# command line switch.                                                       #
# If you want to continue with the ERASE, please type YES and press ENTER.   #
##############################################################################
> YES
Checking IPSW for required components...
All required components found in IPSW
Using cached filesystem from '/root/Downloads/iPhone_4.0_64bit_12.4.6_16G183_Restore/038-36938-005.dmg'
Found ECID 4355337108964
Getting ApNonce in recovery mode... 8a ce e9 97 a2 ef dd 54 be aa 64 99 ed ed 46 02 62 96 82 9b 
Trying to fetch new SHSH blob
Getting SepNonce in recovery mode... 29 63 83 d9 81 20 0f ef 95 bb c2 da 10 26 54 93 0e 09 a4 6a 
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received SHSH blobs
Extracting iBEC.iphone6.RELEASE.im4p...
Personalizing IMG4 component iBEC...
Sending iBEC (731057 bytes)...
ERROR: Device did not reconnect in recovery mode. Possibly invalid iBEC. Reset device and try again.

leukimi, 2020-05-24:
Confirming same error, same halting point, same error messages on iPad Mini 2 A1489

yogendra123456, 2020-05-26:
hi all,

i also received same error such as below.....
[image]

kindly suggest what is the update in code for that error , i used latest libmobiledevice code for windows 10......
 Thanks in advanced..

lukepasek, 2020-05-30:
Same here.
`Found device in Recovery mode
Identified device as k93ap, iPad2,1
NOTE: Running non-interactively, automatically selecting latest available version
Selected firmware 9.3.5 (build 13G36)
Verifying 'iPad2,1_9.3.5_13G36_Restore.ipsw'...
Checksum matches.
Extracting BuildManifest from IPSW
Product Version: 9.3.5
Product Build: 13G36 Major: 13
INFO: device serial number is [...]
Device supports Image4: false
Variant: Customer Erase Install (IPSW)
This restore will erase your device data.
Checking IPSW for required components...
All required components found in IPSW
Using cached filesystem from 'iPad2,1_9.3.5_13G36_Restore/058-49123-036.dmg'
Found ECID [...]
Getting ApNonce in recovery mode... [...]
Trying to fetch new SHSH blob
Getting SepNonce in recovery mode...
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received SHSH blobs
Sending APTicket (2732 bytes)
Extracting iBEC.k93.RELEASE.dfu...
Personalizing IMG3 component iBEC...
reconstructed size: 297498
Sending iBEC (297498 bytes)...
ERROR: Device did not disconnect. Possibly invalid iBEC. Reset device and try again.
`

lukepasek, 2020-05-31:
Ok, after some investigation on a docker container, I found out that this happens when there are no udev events. To diagnose if the udev subsystem (or libusb hotplug events) works properly with your device, run:
```
udevmuxd -f -v
```
And try resetting or disconnecting your apple device, multiple events should be visible as follows:
```
[00:10:38.582][3] usbmuxd v1.1.1 starting up
[00:10:38.582][4] Creating socket
[00:10:38.582][4] Initializing USB
[00:10:38.591][4] Registering for libusb hotplug events
[00:10:38.591][4] Found new device with v/p 05ac:129f at 3-52
[00:10:38.591][4] Setting configuration for device 3-52, from 1 to 4
[00:10:38.601][4] Found interface 1 with endpoints 04/85 for device 3-52
[00:10:38.601][4] Using wMaxPacketSize=512 for device 3-52
[00:10:38.601][4] USB Speed is 480 MBit/s for device 3-52
[00:10:38.601][4] 1 device detected
[00:10:38.601][3] Initialization complete
[00:10:38.603][4] Got lang ID 1033 for device 3-52
[00:10:38.603][4] Got serial 'XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX' for device 3-52
[00:10:38.604][3] Connecting to new device on location 0x30034 as ID 1
[00:10:38.604][3] Connected to v2.0 device 1 on location 0x30034 with serial number XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
[...CUT...]
[00:12:20.891][4] Device 3-52 RX aborted due to error or disconnect
[00:12:20.891][3] Removed device 1 on location 0x30034
[00:12:20.894][4] Device 3-52 RX aborted due to error or disconnect
[00:12:26.057][4] Found new device with v/p 05ac:129f at 3-53
[00:12:26.057][4] Setting configuration for device 3-53, from 1 to 4
[00:12:26.116][4] Found interface 1 with endpoints 04/85 for device 3-53
[00:12:26.116][4] Using wMaxPacketSize=512 for device 3-53
[00:12:26.116][4] USB Speed is 480 MBit/s for device 3-53
[00:12:26.117][4] Got lang ID 1033 for device 3-53
[00:12:26.118][4] Got serial 'XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX' for device 3-53
[00:12:26.118][3] Connecting to new device on location 0x30035 as ID 2
[00:12:26.118][3] Connected to v2.0 device 2 on location 0x30035 with serial number XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
```

**Note** that the device must be in **normal** mode. 
To switch to normal mode form the recovery mode one can use:
```
irecovery -n
```

To enable udev events in docker the container must be run with `--net=host`.

nikias, 2020-05-31:
I can see different events here. One is
```
ERROR: Device did not reconnect in recovery mode. Possibly invalid iBEC. Reset device and try again.
```
and the other is
```
ERROR: Device did not disconnect. Possibly invalid iBEC. Reset device and try again.
```
The first one (@7iswin and @yogendra123456) means the device went to recovery mode, was detected fine there, and after idevicerestore sent the iBEC image and told the device to run it, the device disconnects as expected, but for some reason it never gets an event for the reconnection of a device.

In the second case (@lukepasek) the device doesn't even disconnect apparently.

Can you run idevicerestore with `-d`? This gives a little more output. You will see messages like
```
Waiting for device to disconnect...
```
and then either it goes to the error case (which we see in @lukepasek's case)
```
ERROR: Device did not disconnect. Possibly invalid iBEC. Reset device and try again.
```
or it continues and says
```
Waiting for device to reconnect in recovery mode...
```
which then leads to the other error case
```
ERROR: Device did not reconnect in recovery mode. Possibly invalid iBEC. Reset device and try again.
```
or it continues.
Unfortunately I didn't add timestamps to the debug messages, but please could you check the timings from the `Waiting for device` messages to the following error? Is it instantly? Or does it happen after about 10 seconds?

yogendra123456, 2020-06-01:
Hi nikias and all,

i also run command with run idevicerestore with -d but given below error is get again.
hi lukepasek how i check udevmuxd in window system. it is combined with libmobiledevice code.
[Debug.txt](https://github.com/libimobiledevice/idevicerestore/files/4710270/Debug.txt)

thanks in advanced..

lukepasek, 2020-06-01:
@yogendra123456 I don't know how usbmusd should be run in Windows, but you could try with the docker image I made.
You need to install docker host for Windows: https://docs.docker.com/docker-for-windows/
Set it up to run Linux containers, do `docker pull lukepasek/idevicerestore`, and then run the container as described on docker hub readme: https://hub.docker.com/r/lukepasek/idevicerestore.
This will run the latest (as of 01 June 2020) Linux version of the idevicerestore.

nikias, 2020-06-01:
@yogendra123456 usbmuxd is part of the iTunes (more precise AppleMobileDevice driver) installation package. The issues we see here are not related to usbmuxd though, but more an issue with recovery mode where usbmuxd is not involved at all.

nikias, 2020-06-01:
@yogendra123456 thanks for the debug output it does look different *again*. Something else must be totally wrong on your end, but I can't really tell what. It doesn't even try to send the iBEC this time and I see a lot of connected/disconnect messages there too...
Unfortunately Windows isn't officially supported. The code is adapted to somehow work with Windows, but it's mostly untested...

yogendra123456, 2020-06-02:
hi @nikias and @taanders,

Can you provide me libmobiledevice code which is supported to window and update the firmware on iPhone device any suggestion how to restore device using libmobiledevice idevicerestore.exe ...on window system.....
[Debug.txt](https://github.com/libimobiledevice/idevicerestore/files/4714522/Debug.txt)
[Only debug.txt](https://github.com/libimobiledevice/idevicerestore/files/4714523/Only.debug.txt)

thanks in advanced...

yogendra123456, 2020-06-05:
hi @nikias and @taanders,
kindly give me some instruction about libmobiledevice for windows i found some error which is given below, what happen on window iphone is not placed in restore mode. while iTunes update the firmware at this stage and also 3uTools what happen with libmobiledevice code, it happen with usbmuxd or others class which is not connect device to properly to window system...
kindly provide me some update.......
[Release.txt](https://github.com/libimobiledevice/idevicerestore/files/4734767/Release.txt)
 thanks in advanced @nikias and @taanders,

yogendra123456, 2020-06-10:
Hi All,

i restore my device but it create error again and again on windows 10...

Extracting recoverymode@1334~iphone-lightning.im4p...
Personalizing IMG4 component RecoveryMode...
Extracting iBoot.n71.RELEASE.im4p...
Personalizing IMG4 component iBoot...
Extracting sep-firmware.n71.RELEASE.im4p...
Personalizing IMG4 component RestoreSEP...
Extracting sep-firmware.n71.RELEASE.im4p...
Personalizing IMG4 component SEP...
Sending NORData now...
Done sending NORData
Unmounting filesystems (29)
Unmounting filesystems (29)
Unmounting filesystems (29)
Unmounting filesystems (29)
Checking filesystems (15)
ERROR: Unable to receive message from FDR 00000124E29C2130 (-7). 0/2 bytes

Details in attached file..
[ErrorRelease.txt](https://github.com/libimobiledevice/idevicerestore/files/4757526/ErrorRelease.txt)

kindly provide the update what is the solution of this error, i am using latest libmobiledevice code with visual studio 2019...

thanks in advanced....

Bios597407, 2020-06-29:
> @yogendra123456 PLEASE STOP SPAMMING MULTIPLE TICKETS WITH THE SAME QUESTION!
> I told you already last time and YOU DO IT AGAIN, ON 8 DIFFERENT TICKETS! This is really annoying and not going to get you any help at all.
> 
> As I said before your problem is most likely Windows which is still only experimentally supported, and _ESPECIALLY_ VISUAL STUDIO IS **NOT** considered supported at all.
> 
> Get yourself a macOS or Linux system and I guarantee you it will just work.

sure 100%

chascoso, 2020-07-03:
hi @nikias  i have the same problem, but i am running mint tricia, but i dont know code at all... just starting now, following instructions... could you help ? i have a ipad mini 2, and i receive this error when i try to restore

 sudo idevicerestore --restore-mode '/home/chascoso/Desktop/iPad_64bit_12.4.7_16G192_Restore.ipsw' 
Found device in Recovery mode
Identified device as j86ap, iPad4,5
Extracting BuildManifest from IPSW
Product Version: 12.4.7
Product Build: 16G192 Major: 16
INFO: device serial number is DLXM428AFLMN
Device supports Image4: true
Variant: Customer Upgrade Install (IPSW)
This restore will update your device without erasing user data.
Checking IPSW for required components...
All required components found in IPSW
Using cached filesystem from '/home/chascoso/Desktop/iPad_64bit_12.4.7_16G192_Restore/038-53296-004.dmg'
Found ECID 1328475922452
Getting ApNonce in recovery mode... 6b b3 f1 14 af 9d 99 73 e2 44 43 bc e9 78 65 e0 d0 56 d2 8d 
Trying to fetch new SHSH blob
Getting SepNonce in recovery mode... eb e4 75 0d 7d f0 4a 5e 28 a1 e3 96 33 50 e6 96 5f 68 b3 1a 
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received SHSH blobs
Extracting iBEC.ipad4b.RELEASE.im4p...
Personalizing IMG4 component iBEC...
Sending iBEC (747441 bytes)...
ERROR: Device did not reconnect in recovery mode. Possibly invalid iBEC. Reset device and try again.

ive tried everything up and still get stucked....

chascoso, 2020-07-03:
> @yogendra123456 I don't know how usbmusd should be run in Windows, but you could try with the docker image I made.
> You need to install docker host for Windows: [...]

also tried this, but 
Got permission denied while trying to connect to the Docker daemon socket at unix:///var/run/docker.sock: Post http://%2Fvar%2Frun%2Fdocker.sock/v1.40/images/create?fromImage=lukepasek%2Fidevicerestore&tag=latest: dial unix /var/run/docker.sock: connect: permission denied

TestingPlant, 2020-10-23:
I found a solution to this problem, at least for my iPad Air.
You should first apply the below patch:
1. Clone this repository (`git clone https://github.com/libimobiledevice/idevicerestore; cd idevicerestore`)
2. Download [patch.txt](https://github.com/libimobiledevice/idevicerestore/files/5426569/patch.txt)
3. Apply it (`patch --strip 1 --input patch.patch`)
4. Compile and install the repository

Afterwards, you should run `idevicerestore` and add `--debug`. Once you see `Waiting for device to reconnect in recovery mode...` outputted, turn on the device by holding the power button.

TonyKutunio, 2020-10-28:
> @yogendra123456 PLEASE STOP SPAMMING MULTIPLE TICKETS WITH THE SAME QUESTION!
> I told you already last time and YOU DO IT AGAIN, ON 8 DIFFERENT TICKETS! This is really annoying and not going to get [...]

Have the same problem on Ubuntu

kenkit, 2020-12-27:
This worked iphonde 5s

kenkit, 2020-12-27:
This worked iphone 5s

rodarima, 2022-04-23:
Reproduced in Arch Linux with an old iPad4,2, it takes around 20 seconds to enter the recovery mode:

```
[ 5115.439383] apple-mfi-fastcharge 3-2.2: USB disconnect, device number 20
[ 5134.587015] usb 3-2.2: new high-speed USB device number 21 using xhci_hcd
[ 5134.677485] usb 3-2.2: New USB device found, idVendor=05ac, idProduct=1281, bcdDevice= 0.00
[ 5134.677495] usb 3-2.2: New USB device strings: Mfr=2, Product=3, SerialNumber=4
[ 5134.677499] usb 3-2.2: Product: Apple Mobile Device (Recovery Mode)
[ 5134.677502] usb 3-2.2: Manufacturer: Apple Inc.
[ 5134.677504] usb 3-2.2: SerialNumber: CPID:8960 CPRV:11 CPFM:03 SCEP:01 BDID:12 ECID:XXXX IBFL:1C SRNM:[XXXX]
...
[ 5609.004028] apple-mfi-fastcharge 3-2.2: USB disconnect, device number 21
[ 5628.153349] usb 3-2.2: new high-speed USB device number 22 using xhci_hcd
[ 5628.243818] usb 3-2.2: New USB device found, idVendor=05ac, idProduct=1281, bcdDevice= 0.00
[ 5628.243835] usb 3-2.2: New USB device strings: Mfr=2, Product=3, SerialNumber=4
[ 5628.243843] usb 3-2.2: Product: Apple Mobile Device (Recovery Mode)
[ 5628.243848] usb 3-2.2: Manufacturer: Apple Inc.
[ 5628.243853] usb 3-2.2: SerialNumber: CPID:8960 CPRV:11 CPFM:03 SCEP:01 BDID:12 ECID:XXXX IBFL:1C SRNM:[XXXX]
```

But the timeout is set to 10 seconds:

https://github.com/libimobiledevice/idevicerestore/blob/master/src/idevicerestore.c#L1281

Setting it to a larger value (I used 60 seconds), as the @TestingPlant patch suggests, worked fine.

nikias, 2022-04-24:
I merged the PR from @rodarima, kindly check if the issues persists.

chascoso, 2022-04-24:
Thanks for the help! Unfortunately I don't have the device anymore.

Thanks anyway!

Sonico98, 2022-07-28:
> I merged the PR from @rodarima, kindly check if the issues persists.

It's working fine now on iPad 2,2. Thanks!

Core2Quad-779, 2026-09-20:
> ```
> udevmuxd -f -v
> ```
@lukepasek How do I run that command? This page is the first result for it...