libimobiledevice/idevicerestore #325: ERROR: Unable to find component node for Rap,RestoreRTKitOS

ITAD-Development, 2020-04-02:
ERROR: Unable to find component node for Rap,RestoreRTKitOS
ERROR: Unable get path for 'Rap,RestoreRTKitOS' component
ERROR: restore_send_firmware_updater_data: Couldn't get Rose firmware data
ERROR: Unable to send FirmwareUpdater data
ERROR: Unable to successfully restore device
ERROR: Unable to restore device

nikias, 2020-04-07:
Can you test with the patch from PR #323

nikias, 2020-04-07:
Duplicate of #322

closed 2020-04-07 by nikias

iM4CH3T3, 2020-04-10:
ok it worked with the fix from PR #323 with this output
Received Rose ticket
Extracting ftab.bin...
ERROR: Unable to find component node for Rap,RestoreRTKitOS
WARNING: Could not find 'Rap,RestoreRTKitOS' node. The entry is not in the build manifest.
Sending FirmwareResponse data now...
16:59:20 property_list_service.c:132 internal_plist_send(): sending 819217 bytes
16:59:20 service.c:144 service_send(): sending 4 bytes
16:59:20 service.c:144 service_send(): sending 819217 bytes
16:59:20 property_list_service.c:137 internal_plist_send(): sent 819217 bytes
16:59:20 property_list_service.c:138 internal_plist_send(): supress printing 819217 bytes plist...

Done sending FirmwareUpdater data
16:59:23 property_list_service.c:205 internal_plist_receive_timeout(): initial read=4
16:59:23 property_list_service.c:211 internal_plist_receive_timeout(): 328 bytes following
16:59:23 property_list_service.c:224 internal_plist_receive_timeout(): received 328 bytes
16:59:23 property_list_service.c:250 internal_plist_receive_timeout(): printing 328 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>MsgType</key>
	<string>ProgressMsg</string>
	<key>Operation</key>
	<integer>65</integer>
	<key>Progress</key>
	<integer>100</integer>
</dict>
</plist>
16:59:23 property_list_service.c:205 internal_plist_receive_timeout(): initial read=4
16:59:23 property_list_service.c:211 internal_plist_receive_timeout(): 327 bytes following
16:59:23 property_list_service.c:224 internal_plist_receive_timeout(): received 327 bytes
16:59:23 property_list_service.c:250 internal_plist_receive_timeout(): printing 327 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>MsgType</key>
	<string>ProgressMsg</string>
	<key>Operation</key>
	<integer>29</integer>
	<key>Progress</key>
	<integer>-1</integer>
</dict>
</plist>
Unmounting filesystems (29)
16:59:23 property_list_service.c:205 internal_plist_receive_timeout(): initial read=4
16:59:23 property_list_service.c:211 internal_plist_receive_timeout(): 327 bytes following
16:59:23 property_list_service.c:224 internal_plist_receive_timeout(): received 327 bytes
16:59:23 property_list_service.c:250 internal_plist_receive_timeout(): printing 327 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>MsgType</key>
	<string>ProgressMsg</string>
	<key>Operation</key>
	<integer>29</integer>
	<key>Progress</key>
	<integer>-1</integer>
</dict>
</plist>
Unmounting filesystems (29)
16:59:23 property_list_service.c:205 internal_plist_receive_timeout(): initial read=4
16:59:23 property_list_service.c:211 internal_plist_receive_timeout(): 327 bytes following
16:59:23 property_list_service.c:224 internal_plist_receive_timeout(): received 327 bytes
16:59:23 property_list_service.c:250 internal_plist_receive_timeout(): printing 327 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
[... 51 lines cut ...]
16:59:23 property_list_service.c:138 internal_plist_send(): printing 249 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>MsgType</key>
	<string>ReceivedFinalStatusMsg</string>
</dict>
</plist>
idevice_event_cb: device 000d59992290802e (udid: 00008030-000D59992290802E) disconnected
16:59:29 idevice.c:552 internal_connection_receive_timeout(): ERROR: usbmuxd_recv_timeout returned -104 (Connection reset by peer)
FDR 0x55dab16de0d0 terminating...
16:59:29 restore.c:351 restored_goodbye(): called
16:59:29 property_list_service.c:132 internal_plist_send(): sending 285 bytes
16:59:29 service.c:144 service_send(): sending 4 bytes
16:59:29 idevice.c:479 internal_connection_send(): ERROR: usbmuxd_send returned -32 (Broken pipe)
16:59:29 service.c:147 service_send(): ERROR: sending to device failed.
16:59:29 property_list_service.c:147 internal_plist_send(): ERROR: sending to device failed.
16:59:29 idevice.c:552 internal_connection_receive_timeout(): ERROR: usbmuxd_recv_timeout returned -104 (Connection reset by peer)
16:59:29 service.c:167 service_receive_with_timeout(): could not read data
16:59:29 property_list_service.c:196 internal_plist_receive_timeout(): initial read failed!
16:59:29 restore.c:359 restored_goodbye(): did not get goodbye response back
Cleaning up...
DONE
im4ch3t3@im4ch3t3-ubuntu:~$

iM4CH3T3, 2020-04-10:
thats the whole output 

https://pastebin.com/wUDHgnNX

nikias, 2020-04-12:
Fixed it with commit f27d19e1c2e69dba271834d0c7a3a5be3a83e275.