libimobiledevice/idevicerestore #326: unable to successfully restore device

iM4CH3T3, 2020-04-07:
im4ch3t3@im4ch3t3-ubuntu:~$ idevicerestore -e '/home/im4ch3t3/Downloads/iPhone11,8,iPhone12,1_13.4.1_17E262_Restore.ipsw' 
Found device in Recovery mode
Identified device as n104ap, iPhone12,1
Extracting BuildManifest from IPSW
Product Version: 13.4.1
Product Build: 17E262 Major: 17
INFO: device serial number is C8PZ8BMRN735
Device supports Image4: true
Variant: Customer Erase Install (IPSW)
This restore will erase your device data.
################################ [ WARNING ] #################################
# You are about to perform an *ERASE* restore. ALL DATA on the target device #
# will be IRREVERSIBLY DESTROYED. If you want to update your device without  #
# erasing the user data, hit CTRL+C now and restart without -e or --erase    #
# command line switch.                                                       #
# If you want to continue with the ERASE, please type YES and press ENTER.   #
##############################################################################
> YES
Checking IPSW for required components...
All required components found in IPSW
Using cached filesystem from '/home/im4ch3t3/Downloads/iPhone11,8,iPhone12,1_13.4.1_17E262_Restore/038-03660-158.dmg'
Found ECID 3757688942002222
Getting ApNonce in recovery mode... 6a 28 71 6c 9c e5 88 68 9f fd 73 34 8d 84 5d 41 24 d5 64 9f c2 8b 4a 32 d6 55 6d de fa 53 2f 1f 
Trying to fetch new SHSH blob
Getting SepNonce in recovery mode... 8c 6e 81 5a 98 72 cc f9 6b 72 3a 4a e2 1e 34 ee fa dc 46 8f 
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received SHSH blobs
Extracting iBEC.n104.RELEASE.im4p...
Personalizing IMG4 component iBEC...
Sending iBEC (1192123 bytes)...
Recovery Mode Environment:
iBoot build-version=iBoot-5540.102.4
iBoot build-style=RELEASE
Sending RestoreLogo...
Extracting applelogo@1792~iphone.im4p...
Personalizing IMG4 component RestoreLogo...
Sending RestoreLogo (20164 bytes)...
Extracting h12_ane_fw_metis.im4p...
Personalizing IMG4 component ANE...
Sending ANE (12347740 bytes)...
Extracting aopfw-iphone12baop.im4p...
Personalizing IMG4 component AOP...
Sending AOP (1663404 bytes)...
Extracting AppleAVE2FW.im4p...
Personalizing IMG4 component AVE...
Sending AVE (1113324 bytes)...
Extracting armfw_g12p.im4p...
Personalizing IMG4 component GFX...
Sending GFX (3470444 bytes)...
Extracting adc-zelus-n104.im4p...
Personalizing IMG4 component ISP...
Sending ISP (9742284 bytes)...
Extracting t8030pmp.im4p...
Personalizing IMG4 component PMP...
Sending PMP (340396 bytes)...
Extracting 048-65173-372.dmg.trustcache...
Personalizing IMG4 component RestoreTrustCache...
Sending RestoreTrustCache (13085 bytes)...
Extracting SmartIOFirmwareT8030.im4p...
Personalizing IMG4 component SIO...
Sending SIO (1161264 bytes)...
Extracting WirelessPower.iphone12b.im4p...
Personalizing IMG4 component WCHFirmwareUpdater...
Sending WCHFirmwareUpdater (56715 bytes)...
ramdisk-size=0x20000000
Extracting 048-65173-372.dmg...
Personalizing IMG4 component RestoreRamDisk...
Sending RestoreRamDisk (104524207 bytes)...
Extracting DeviceTree.n104ap.im4p...
[... 186 lines cut ...]
Done sending BasebandData
Updating Baseband in progress...
About to send BasebandData...
Sending Baseband TSS request...
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received Baseband SHSH blobs
WARNING: size mismatch when parsing MBN file. Continuing anyway.
WARNING: size mismatch when parsing MBN file. Continuing anyway.
Sending BasebandData now...
Done sending BasebandData
Updating Baseband completed.
Updating Rose (65)
Sending Rose TSS request...
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received Rose ticket
Extracting ftab.bin...
ERROR: Unable to find component node for Rap,RestoreRTKitOS
ERROR: Unable get path for 'Rap,RestoreRTKitOS' component
ERROR: restore_send_firmware_updater_data: Couldn't get Rose firmware data
ERROR: Unable to send FirmwareUpdater data
ERROR: Unable to successfully restore device
ERROR: Unable to restore device
im4ch3t3@im4ch3t3-ubuntu:~$

nikias, 2020-04-07:
Can you try the patch of this PR: https://github.com/libimobiledevice/idevicerestore/pull/323

nikias, 2020-04-07:
Also duplicate of #325.

closed 2020-04-07 by nikias

iM4CH3T3, 2020-04-08:
Will try pr323 when back home ...

iM4CH3T3, 2020-04-10:
> Can you try the patch of this PR: https://github.com/libimobiledevice/idevicerestore/pull/323

Worked with the fix of PR #323
[image]

iM4CH3T3, 2020-04-10:
> > Can you try the patch of this PR: https://github.com/libimobiledevice/idevicerestore/pull/323
> 
> Worked with the fix of PR #323
> [image]

Do u need the whole output ?

nikias, 2020-04-12:
Fixed it with commit f27d19e1c2e69dba271834d0c7a3a5be3a83e275.