libimobiledevice/idevicerestore #331: ERROR: Unable to restore device

Lodifice, 2020-04-14:
Hi, after running out of charge my iPod 4G (iOS 6.1, jailbroken years ago) now crashes during boot and enters recovery mode. `irecovery -n` doesn't help. So I decided to restore which failed as well. Find the log here: https://pastebin.com/PZ8TXsCX.

At this point, is there anything that can be done? I don't care much for the device itself, but I'd rather not lose the data. Since your tool is somehow able to print the partitions, maybe there is a way to mount them and copy the data? Any help is appreciated.

nikias, 2020-04-14:
There is *some* hope but I can't promise you anything here. Firstly, idevicerestore doesn't access the device directly, but rather helps booting it into the restore environment. This is completely device driven, and you can't directly interfere with it. Your choice to try an update restore was the right one but unfortunately, as you can see from the log, the user data partition is corrupted and cannot be repaired.
So this means that idevicerestore cannot help you with your problem.

Now, the good news is that the iPod Touch 4G has a A4 CPU which is vulnerable to limera1n exploit (like the iPhone 4) which in turn allows you to boot a custom ramdisk with SSH on it. Check out this page here which *might* help you recover at least some of your data:
http://msftguy.blogspot.com/2012/01/automatic-ssh-ramdisk-creation-and.html
There's a video showing the usage. After mounting (given/hoping that it will mount your corrupted fs) you should be able to copy over the data via SSH from the /mnt2 mount.
Good luck!

Lodifice, 2020-04-14:
Thank you already for your suggestion, I'll definitly give it a try.

However, there is already a problem with putting my device in DFU mode: the Home key does not work anymore. If you don't mind the question: is there any way of putting it into DFU mode from software?

nikias, 2020-04-19:
This would involve flashing a corrupted LLB image which would work through a restore, but you can't perform a restore in your case, so that's not an option. Maybe take it apart and short the pins of the home button on the board to enter DFU?

Lodifice, 2020-04-19:
Thanks again for the reply. I see dismantling the device as my last resort option, but it seems that it will come down to this.

In the meantime, I have found https://forums.modmy.com/general-f201/710786-recovery-ramdisk-ssh.html which appears to be the same ramdisk thing by msftguy but explicitly mentions Recovery mode. Do you think this can work? For me the creation of the ramdisk didn't work (error: untar failed), so I'd like to know if this might work at all before investing more time into it.

nikias, 2020-04-19:
No, that's for an even older device. For iPhone 4 / iPod Touch 4th gen you need to get it into DFU mode for the limera1n exploit to work.