libimobiledevice/idevicerestore #332: [Question]ASR: Could not restore - No space left on device

duke-cliff, 2020-04-24:
What could be the reason the device is running out of space, is there something we can do to free some space to continue?

Or should we wait for a smaller upgrade in the future?

```
ASR STATUS: start	512	multicast-client
ASR: Waiting for connection attempt from server
ASR STATUS: setup
ASR: Validating target...
ASR: done
ASR STATUS: metadata
ASR: Validating source...
ASR: done
ASR: Retrieving scan information...
ASR: done
ASR: Validating sizes...
ASR: done
ASR STATUS: restore
ASR RESTORE PROGRESS: 2%
ramrod_display_set_granular_progress_forced: 6.000000
ASR RESTORE PROGRESS: 4%
ramrod_display_set_granular_progress_forced: 7.000000
ASR RESTORE PROGRESS: 6%
ramrod_display_set_granular_progress_forced: 7.000000
ASR RESTORE PROGRESS: 8%
ramrod_display_set_granular_progress_forced: 8.000000
ASR RESTORE PROGRESS: 10%
ramrod_display_set_granular_progress_forced: 8.000000
ASR RESTORE PROGRESS: 12%
ramrod_display_set_granular_progress_forced: 9.000000
ASR RESTORE PROGRESS: 14%
ramrod_display_set_granular_progress_forced: 10.000000
ASR RESTORE PROGRESS: 16%
ramrod_display_set_granular_progress_forced: 10.000000
ASR RESTORE PROGRESS: 18%
ramrod_display_set_granular_progress_forced: 11.000000
ASR RESTORE PROGRESS: 20%
ramrod_display_set_granular_progress_forced: 11.000000
ASR RESTORE PROGRESS: 22%
ramrod_display_set_granular_progress_forced: 12.000000
ASR RESTORE PROGRESS: 24%
ramrod_display_set_granular_progress_forced: 12.000000
ASR RESTORE PROGRESS: 26%
ramrod_display_set_granular_progress_forced: 13.000000
ASR RESTORE PROGRESS: 28%
ramrod_display_set_granular_progress_forced: 13.000000
ASR RESTORE PROGRESS: 30%
ramrod_display_set_granular_progress_forced: 14.000000
ASR RESTORE PROGRESS: 32%
ramrod_display_set_granular_progress_forced: 15.000000
ASR RESTORE PROGRESS: 34%
ramrod_display_set_granular_progress_forced: 15.000000
ASR RESTORE PROGRESS: 36%
ramrod_display_set_granular_progress_forced: 16.000000
ASR RESTORE PROGRESS: 38%
ramrod_display_set_granular_progress_forced: 16.000000
ASR RESTORE PROGRESS: 40%
ramrod_display_set_granular_progress_forced: 17.000000
ASR RESTORE PROGRESS: 42%
ramrod_display_set_granular_progress_forced: 17.000000
ASR RESTORE PROGRESS: 44%
ramrod_display_set_granular_progress_forced: 18.000000
ASR RESTORE PROGRESS: 46%
ramrod_display_set_granular_progress_forced: 18.000000
ASR RESTORE PROGRESS: 48%
ramrod_display_set_granular_progress_forced: 19.000000
ASR RESTORE PROGRESS: 50%
ramrod_display_set_granular_progress_forced: 20.000000
ASR RESTORE PROGRESS: 52%
ramrod_display_set_granular_progress_forced: 20.000000
ASR RESTORE PROGRESS: 54%
ramrod_display_set_granular_progress_forced: 21.000000
ASR RESTORE PROGRESS: 56%
ramrod_display_set_granular_progress_forced: 21.000000
ASR RESTORE PROGRESS: 58%
ramrod_display_set_granular_progress_forced: 22.000000
ASR RESTORE PROGRESS: 60%
ramrod_display_set_granular_progress_forced: 22.000000
ASR RESTORE PROGRESS: 62%
ramrod_display_set_granular_progress_forced: 23.000000
ASR RESTORE PROGRESS: 64%
ramrod_display_set_granular_progress_forced: 23.000000
ASR RESTORE PROGRESS: 66%
ramrod_display_set_granular_progress_forced: 24.000000
ASR RESTORE PROGRESS: 68%
ramrod_display_set_granular_progress_forced: 24.000000
ASR RESTORE PROGRESS: 70%
ramrod_display_set_granular_progress_forced: 25.000000
ASR RESTORE PROGRESS: 72%
ramrod_display_set_granular_progress_forced: 26.000000
ASR RESTORE PROGRESS: 74%
ramrod_display_set_granular_progress_forced: 26.000000
ASR RESTORE PROGRESS: 76%
ramrod_display_set_granular_progress_forced: 27.000000
ASR RESTORE PROGRESS: 78%
ramrod_display_set_granular_progress_forced: 27.000000
ASR RESTORE PROGRESS: 80%
ramrod_display_set_granular_progress_forced: 28.000000
ASR RESTORE PROGRESS: 82%
ramrod_display_set_granular_progress_forced: 28.000000
ASR RESTORE PROGRESS: 84%
ramrod_display_set_granular_progress_forced: 29.000000
ASR RESTORE PROGRESS: 86%
ramrod_display_set_granular_progress_forced: 29.000000
ASR RESTORE PROGRESS: 88%
ramrod_display_set_granular_progress_forced: 30.000000
ASR RESTORE PROGRESS: 90%
ramrod_display_set_granular_progress_forced: 31.000000
ASR RESTORE PROGRESS: 92%
ramrod_display_set_granular_progress_forced: 31.000000
ASR RESTORE PROGRESS: 94%
ramrod_display_set_granular_progress_forced: 32.000000
ASR RESTORE PROGRESS: 96%
ramrod_display_set_granular_progress_forced: 32.000000
ASR STATUS: fail
**ASR: Could not restore - No space left on device**
ASR STATUS: fail
restore_apfs_image : failed to restore an APFS image, error = 14
[04:52:13.0617-GMT]{3>4} CHECKPOINT FAILURE:(FAILURE:14) (null):[0x065B] asr_and_invert_image [0]D(failed to restore APFS image)
restore-step-results = {0x1107065B:{0:14}}
restore-step-codes = {0x1107065B:{0:14}}
restore-step-domains = {0x1107065B:{0:"AMRestoreErrorDomain"}}
restore-step-error = {0x1107065B:"[0]D(failed to restore APFS image)"}
restore-step-uptime = 214
restore-step-user-progress = 32
[04:52:13.0617-GMT]{3>4} CHECKPOINT NOTICE: (NVRAM set) restore-step-user-progress=32 [sync=true] (first failure)
[04:52:13.0617-GMT]{3>4} CHECKPOINT FAILURE:(FAILURE:14) RESTORED:[0x0677] perform_main_os_prepare [0]D(failed to restore APFS image)
restore-step-results = {0x11070677:{0:14};0x1107065B:{0:14}}
restore-step-codes = {0x11070677:{0:14};0x1107065B:{0:14}}
restore-step-domains = {0x11070677:{0:"AMRestoreErrorDomain"};0x1107065B:{0:"AMRestoreErrorDomain"}}
restore-step-error = {0x11070677:"[0]D(failed to restore APFS image)"}
restore-step-uptime = 214
restore-step-user-progress = 32
[04:52:13.0618-GMT]{3>4} CHECKPOINT BEGIN: RESTORED:[0x067C] cleanup_boot_command
restore-step-ids = {0x11030677:40;0x1103065B:48;0x1103067C:49}
restore-step-names = {0x11030677:perform_main_os_prepare;0x1103065B:asr_and_invert_image;0x1103067C:cleanup_boot_command}
restore-step-uptime = 214
restore-step-user-progress = 32
entering reset_boot_command_if_value
[04:52:13.0618-GMT]{3>4} CHECKPOINT END: RESTORED:[0x067C] cleanup_boot_command
restore-step-ids = {0x11030677:40;0x1103065B:48}
restore-step-names = {0x11030677:perform_main_os_prepare;0x1103065B:asr_and_invert_image}
restore-step-uptime = 214
restore-step-user-progress = 32
[04:52:13.0618-GMT]{3>4} CHECKPOINT BEGIN: RESTORED:[0x1614] cleanup_recovery_os_volume
restore-step-ids = {0x11030677:40;0x1103065B:48;0x11031614:50}
restore-step-names = {0x11030677:perform_main_os_prepare;0x1103065B:asr_and_invert_image;0x11031614:cleanup_recovery_os_volume}
restore-step-uptime = 214
restore-step-user-progress = 32
[04:52:13.0618-GMT]{3>4} CHECKPOINT END: RESTORED:[0x1614] cleanup_recovery_os_volume
restore-step-ids = {0x11030677:40;0x1103065B:48}
restore-step-names = {0x11030677:perform_main_os_prepare;0x1103065B:asr_and_invert_image}
restore-step-uptime = 214
restore-step-user-progress = 32
[04:52:13.0618-GMT]{3>4} CHECKPOINT BEGIN: RESTORED:[0x0647] cleanup_check_result
restore-step-ids = {0x11030677:40;0x1103065B:48;0x11030647:51}
restore-step-names = {0x11030677:perform_main_os_prepare;0x1103065B:asr_and_invert_image;0x11030647:cleanup_check_result}
restore-step-uptime = 214
restore-step-user-progress = 32
[04:52:13.0618-GMT]{3>4} CHECKPOINT END: RESTORED:[0x0647] cleanup_check_result
restore-step-ids = {0x11030677:40;0x1103065B:48}
restore-step-names = {0x11030677:perform_main_os_prepare;0x1103065B:asr_and_invert_image}
restore-step-uptime = 214
restore-step-user-progress = 32
[04:52:13.0618-GMT]{3>4} CHECKPOINT BEGIN: RESTORED:[0x0648] cleanup_send_final_status
restore-step-ids = {0x11030677:40;0x1103065B:48;0x11030648:52}
restore-step-names = {0x11030677:perform_main_os_prepare;0x1103065B:asr_and_invert_image;0x11030648:cleanup_send_final_status}
restore-step-uptime = 214
restore-step-user-progress = 32

```

duke-cliff, 2020-04-24:
Or is there anyway to use a smaller unsigned ipw?

nikias, 2020-04-24:
You're not the first to encounter such an error and it seems like that between certain iOS versions the root filesystem changed in size so much that with an update restore it can't write the full root filesystem. Only solution is to perform an erase restore but then you'll lose the user data  if you don't have a backup.

duke-cliff, 2020-04-27:
Thanks! I will keep the phone as-is for now. so many important photos for our new son on that phone. Can't afford to lose all the photos. Fxck Apple!

lzl124631x, 2020-08-20:
Mark. Having same issue. :(

pto2k, 2020-10-08:
Mark, same issue...
A visit to the Genius Bar didn't help.

bezuiden, 2020-11-05:
Same here ... guess we are all in the same boat ...

Couple of questions:
1) Is there a way to get data off the "data" partition?  Or is that no accessible while in recovery?  (example ... copy the sandbox of a specific application away - or backup the data partition and restore after an erase restore?
2) Can a different version of the OS be installed that is smaller and that would fit on the root file system? Or would that cause a data loss too?

Thanks!

wayne-yeung, 2020-11-28:
> Same here ... guess we are all in the same boat ...
>
> Couple of questions:
>
> 1. Is there a way to get data off the "data" partition?  Or is that no accessible while in recovery?  (example ... copy the [...]

Same here, the exact same questions in my mind.
It sucks that there is no way to free up iPhone disk space in recovery mode. 
It is a design flaw that lack of disk space can brick the device.

xslidian, 2021-01-06:
Hello everyone, 
if your ASR restore process passes 90%, you may want to try the iOS 14.4 beta (`18D5030e`):
https://www.theiphonewiki.com/wiki/Beta_Firmware/iPhone/14.x

Download the correct IPSW and run:
`idevicerestore your.ipsw`

singulart, 2021-01-23:
I'm seeing the same on 68%. Any hope?

glyh, 2023-07-11:
It's been a while, is there any new information on this? I brick my phone as well and I can't even do a erase restore.