libimobiledevice/idevicerestore #333: i can't restore iphone 4S

Ninoh-FOX, 2020-04-24:
The phone is of the a friend, and I think that he break the phone tried does jailbreak, the phone always show recovery mode.

I tried flash the last firmware, but this show me an error in iboot and baseband.

How I can fix this?

nikias, 2020-04-24:
Paste some error output.

Ninoh-FOX, 2020-04-24:
`ninoh-fox@debian:~/Descargas/libi$ idevicerestore -e /home/ninoh-fox/Descargas/libi/iPhone4,1_9.3.6_13G37_Restore.ipsw
Found device in Recovery mode
Identified device as n94ap, iPhone4,1
Extracting BuildManifest from IPSW
Product Version: 9.3.6
Product Build: 13G37 Major: 13
INFO: device serial number is C5FLL3RQDTC0
Device supports Image4: false
Variant: Customer Erase Install (IPSW)
This restore will erase your device data.
################################ [ WARNING ] #################################
# You are about to perform an *ERASE* restore. ALL DATA on the target device #
# will be IRREVERSIBLY DESTROYED. If you want to update your device without  #
# erasing the user data, hit CTRL+C now and restart without -e or --erase    #
# command line switch.                                                       #
# If you want to continue with the ERASE, please type YES and press ENTER.   #
##############################################################################
> YES
Checking IPSW for required components...
All required components found in IPSW
Using cached filesystem from '/home/ninoh-fox/Descargas/libi/iPhone4,1_9.3.6_13G37_Restore/058-48374-040.dmg'
Found ECID 4251625202574
Getting ApNonce in recovery mode... 7b d2 00 d8 51 18 bc 6c a4 90 10 d6 3e c6 ec 90 d4 e8 8d bb 
Trying to fetch new SHSH blob
Getting SepNonce in recovery mode... 
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received SHSH blobs
Sending APTicket (2732 bytes)
Extracting iBEC.n94.RELEASE.dfu...
Not personalizing component iBEC...
Sending iBEC (278924 bytes)...
Getting ApNonce in recovery mode... 7b d2 00 d8 51 18 bc 6c a4 90 10 d6 3e c6 ec 90 d4 e8 8d bb 
Sending APTicket (2732 bytes)
Recovery Mode Environment:
iBoot build-version=iBoot-2817.60.2
iBoot build-style=RELEASE
Sending RestoreLogo...
Extracting applelogo@2x~iphone.s5l8940x.img3...
Not personalizing component RestoreLogo...
Sending RestoreLogo (6796 bytes)...
ramdisk-size=0x4000000
Extracting 058-49032-040.dmg...
Not personalizing component RestoreRamDisk...
Sending RestoreRamDisk (20775180 bytes)...
Extracting DeviceTree.n94ap.img3...
Not personalizing component RestoreDeviceTree...
Sending RestoreDeviceTree (83532 bytes)...
Extracting kernelcache.release.n94...
Not personalizing component RestoreKernelCache...
Sending RestoreKernelCache (10718924 bytes)...
Waiting for device to enter restore mode...
About to restore device... 
Connecting now...
Connected to com.apple.mobile.restored, version 13
Device ################################################
Hardware Information:
BoardID: 8
ChipID: 35136
UniqueChipID: #############
ProductionMode: true
Previous restore exit status: 0x100
Starting FDR listener thread
ERROR: Unable to connect to FDR client (-2)
ERROR: Failed to start FDR Ctrl channel
About to send RootTicket...
Sending RootTicket now...
Done sending RootTicket
Waiting for NAND (28)
Previous Restore Log Received:
[... 676 lines cut ...]
restore failed with CFError:
0: AMRestoreErrorDomain/ffffffff: failed to update device firmware
1: AMRestoreErrorDomain/ffffffff: ramrod firmware update failed
2: RamrodErrorDomain/3ec: update_baseband: failed querying baseband info
3: BBUpdater/32: Failed creating a communication transport to baseband.
waiting for host to acknowledge final status received...
final response is missing message typeattempting to dump restore log
writing log file: /mnt1/restore.log

Unmounting filesystems (29)
Unmounting filesystems (29)
Creating partition map (11)
Creating filesystem (12)
Creating filesystem (12)
Mounting filesystems (16)
Mounting filesystems (16)
Unmounting filesystems (29)
Unmounting filesystems (29)
About to send filesystem...
Connected to ASR
Validating the filesystem
Filesystem validated
Sending filesystem now...
[======================================            ]  74.0%^A^E
`