libimobiledevice/idevicerestore #336: Unable to restore iPhone X

jduranc, 2020-05-22:
Hi,

I'm facing one problem with iPhone X devices, where is unable to restore, the process shows the error message:

Found device in Recovery mode
Identified device as d20ap, iPhone10,1
Latest firmware is iPhone_4.7_P3_13.4.1_17E262_Restore.ipsw
Firmware path C:/iOSRestore/iPhone_4.7_P3_13.4.1_17E262_Restore.ipsw
Extracting BuildManifest from IPSW C:/iOSRestore/iPhone_4.7_P3_13.4.1_17E262_Restore.ipsw
INFO: device serial number is FFNVLKQZJC6C
Identified device as iPhone10,1
Variant: Customer Erase Install (IPSW)
This restore will erase your device data.
Checking IPSW for required components...
All required components found in IPSW
Getting device's ECID for TSS request
Found ECID 6254417692065838
Getting ApNonce Request URL set to http://gs.apple.com/TSS/controller?action=2
Enable Proxy: 10.2.0.116:20002
Sending TSS request attempt 1... response successfully received
Received SHSH blobs
Extracting iBEC.d20.RELEASE.im4p...file:Firmware/dfu/iBEC.d20.RELEASE.im4p
Personalizing IMG4 component iBEC...
Sending iBEC (1052560 bytes)...
Recovery Mode Environment:
Sending RestoreLogo...
Extracting applelogo@2x~iphone.im4p...file:Firmware/all_flash/applelogo@2x~iphone.im4p
Personalizing IMG4 component RestoreLogo...
Sending RestoreLogo (13351 bytes)...
DEBUG: ISP is loaded by iBoot.
Extracting adc-nike-d20.im4p...file:Firmware/isp_bni/adc-nike-d20.im4p
Personalizing IMG4 component ISP...
Sending ISP (8985713 bytes)...
DEBUG: RestoreTrustCache is loaded by iBoot.
Extracting 048-64389-370.dmg.trustcache...file:Firmware/048-64389-370.dmg.trustcache
Personalizing IMG4 component RestoreTrustCache...
Sending RestoreTrustCache (11226 bytes)...
ramdisk-size=0x20000000
Extracting 048-64389-370.dmg...file:048-64389-370.dmg
Personalizing IMG4 component RestoreRamDisk...
Sending RestoreRamDisk (104220868 bytes)...
Extracting DeviceTree.d20ap.im4p...file:Firmware/all_flash/DeviceTree.d20ap.im4p
Personalizing IMG4 component RestoreDeviceTree...
Sending RestoreDeviceTree (183100 bytes)...
Extracting kernelcache.release.iphone10...file:kernelcache.release.iphone10
Personalizing IMG4 component RestoreKernelCache...
Sending RestoreKernelCache (13953734 bytes)...
About to restore device...
Waiting for device...
Connect to device in 14.02 seconds
device connected.
Connected to com.apple.mobile.restored, version 15
Restore client success
Device (null) has successfully entered restore mode
UniqueChipID: 6254417692065838
Previous restore exit status: 0x100
Starting FDR listener thread
**ERROR: FDR could not send Begin command.
ERROR: Failed to start FDR Ctrl channel**
About to send RootTicket...
Sending RootTicket now...
Done sending RootTicket
About to send NORData...
Found firmware path Firmware/all_flash
Getting firmware manifest from build identity
Extracting LLB.d20.RELEASE.im4p...file:Firmware/all_flash/LLB.d20.RELEASE.im4p
Personalizing IMG4 component LLB...
Extracting applelogo@2x~iphone.im4p...file:Firmware/all_flash/applelogo@2x~iphone.im4p
Personalizing IMG4 component AppleLogo...
Extracting batterycharging0@2x~iphone.im4p...file:Firmware/all_flash/batterycharging0@2x~iphone.im4p
Personalizing IMG4 component BatteryCharging0...
Extracting batterycharging1@2x~iphone.im4p...file:Firmware/all_flash/batterycharging1@2x~iphone.im4p
Personalizing IMG4 component BatteryCharging1...
Extracting batteryfull@2x~iphone.im4p...file:Firmware/all_flash/batteryfull@2x~iphone.im4p
Personalizing IMG4 component BatteryFull...
Extracting batterylow0@2x~iphone.im4p...file:Firmware/all_flash/batterylow0@2x~iphone.im4p
Personalizing IMG4 component BatteryLow0...
Extracting batterylow1@2x~iphone.im4p...file:Firmware/all_flash/batterylow1@2x~iphone.im4p
Personalizing IMG4 component BatteryLow1...
Extracting glyphplugin@1334~iphone-lightning.im4p...file:Firmware/all_flash/glyphplugin@1334~iphone-lightning.im4p
Personalizing IMG4 component BatteryPlugin...
Extracting DeviceTree.d20ap.im4p...file:Firmware/all_flash/DeviceTree.d20ap.im4p
Personalizing IMG4 component DeviceTree...
Extracting liquiddetect@1334~iphone-lightning.im4p...file:Firmware/all_flash/liquiddetect@1334~iphone-lightning.im4p
Personalizing IMG4 component Liquid...
Extracting recoverymode@1334~iphone-lightning.im4p...file:Firmware/all_flash/recoverymode@1334~iphone-lightning.im4p
Personalizing IMG4 component RecoveryMode...
Extracting iBoot.d20.RELEASE.im4p...file:Firmware/all_flash/iBoot.d20.RELEASE.im4p
Personalizing IMG4 component iBoot...
Extracting sep-firmware.d20.RELEASE.im4p...file:Firmware/all_flash/sep-firmware.d20.RELEASE.im4p
Personalizing IMG4 component RestoreSEP...
Extracting sep-firmware.d20.RELEASE.im4p...file:Firmware/all_flash/sep-firmware.d20.RELEASE.im4p
Personalizing IMG4 component SEP...
NORData Sending now...
Done sending NORData
About to send FDR Trust data...
Sending FDR Trust data now...
Done sending FDR Trust Data
About to send filesystem...
Connecting to ASR
Retrying connection...
Connected to ASR
Validating the filesystem
Filesystem validated
Sending filesystem now...
Done sending filesystem
About to send KernelCache...
Extracting kernelcache.release.iphone10...file:kernelcache.release.iphone10
Personalizing IMG4 component KernelCache...
Sending KernelCache now...
Done sending KernelCache
About to send DeviceTree...
Extracting DeviceTree.d20ap.im4p...file:Firmware/all_flash/DeviceTree.d20ap.im4p
Personalizing IMG4 component DeviceTree...
Sending DeviceTree now...
Done sending DeviceTree
Got status message
Status: Failed to recover FDR data.
ERROR: StatusMsg -53
ERROR: Unable to successfully restore device -53

Here the restore log retrieved from device.
[restore_log_debug.log](https://github.com/libimobiledevice/idevicerestore/files/4670605/restore_log_debug.log)

Someone is facing the same issue? Any help is appreciated.
regards,
J. Luis

nikias, 2020-05-22:
Hi, looks like FDR proxy could not communicate through the reverse proxy:
```
_AMFDRHttpCopyPurpleReverseProxyInformation: Failed to get proxy info for URL 'https://gg.apple.com/ca/authorize'
AMFDRCreateError: _AMFDRHttpMessageSendSync: Failed to copy proxy information and proxy is enabled.: code=10
AMFDRCreateError: _AMFDRHttpRequestSendSyncNoRetry: _AMFDRHttpMessageSendSync failed: code=8
AMFDRCreateError: _AMFDRPermissionsRequestWithString: AMFDRPermissionsRequest failed: code=8
AMFDRCreateError: AMFDRPermissionsRequest: _AMFDRSavePersistentData failed: code=8
AMFDRCreateError: AMFDRDataHTTPLoadPersistent: Permissions request failed.: code=12```

nikias, 2020-05-22:
Ah right I just saw this in the above log messages:
```ERROR: FDR could not send Begin command.
ERROR: Failed to start FDR Ctrl channel```
Unfortunately I can't see why this failed. Have you tried multiple times? I wouldn't be surprised somehow if this has something todo with the fact that this runs under Windows...

jduranc, 2020-05-22:
Hi @nikias 

Thanks for your fast response. Yes, I try several times with same result, and yes I'm afraid that its something related to windows port or drivers. I will continue researching. Thanks for your help.

regards,
J. Luis

yogendra123456, 2020-06-01:
Hi nikias and all,

i also run command with run idevicerestore with -d but given below error is get again.
hi lukepasek how i check udevmuxd in window system. it is combined with libmobiledevice code.
Debug.txt

thanks in advanced..

yogendra123456, 2020-06-05:
hi @nikias and @taanders and all,
kindly give me some instruction about libmobiledevice for windows i found some error which is given below, what happen on window iphone is not placed in restore mode. while iTunes update the firmware at this stage and also 3uTools what happen with libmobiledevice code, it happen with usbmuxd or others class which is not connect device to properly to window system...
kindly provide me some update.......
[Release.txt](https://github.com/microsoft/vcpkg/files/4734811/Release.txt)

thanks in advanced @nikias and @taanders and all,

yogendra123456, 2020-06-10:
Hi All,

i restore my device but it create error again and again on windows 10...

Extracting recoverymode@1334~iphone-lightning.im4p...
Personalizing IMG4 component RecoveryMode...
Extracting iBoot.n71.RELEASE.im4p...
Personalizing IMG4 component iBoot...
Extracting sep-firmware.n71.RELEASE.im4p...
Personalizing IMG4 component RestoreSEP...
Extracting sep-firmware.n71.RELEASE.im4p...
Personalizing IMG4 component SEP...
Sending NORData now...
Done sending NORData
Unmounting filesystems (29)
Unmounting filesystems (29)
Unmounting filesystems (29)
Unmounting filesystems (29)
Checking filesystems (15)
ERROR: Unable to receive message from FDR 00000124E29C2130 (-7). 0/2 bytes

Details in attached file..
[ErrorRelease.txt](https://github.com/libimobiledevice/idevicerestore/files/4757542/ErrorRelease.txt)

kindly provide the update what is the solution of this error, i am using latest libmobiledevice code with visual studio 2019...

thanks in advanced....

yogendra123456, 2020-06-11:
Hi @nikias,

thanks for concern, and sorry for multiple tickets for same question, if you find any update of my comment on window system them comment me....
Thanks