libimobiledevice/idevicerestore #337: Cheksum mismatch

Socially-disabled, 2020-05-29:
As title says. Simple as that yet still annoying

Iphone se ios 13.5
Linux manjaro

nikias, 2020-05-29:
How about giving just a little bit more details here?
What iPhone SE, the old or the 2020 one?
Did you download the .ipsw myself or did you use `idevicerestore -l` ?
What *is* the actual checksum of the downloaded file?

Socially-disabled, 2020-05-30:
> How about giving just a little bit more details here?
> 
> What iPhone SE, the old or the 2020 one?
> 
> Did you download the .ipsw yourself or did you use `idevicerestore -l` ?
> 
> What *is* the actual checksum of the downloaded file?

Ohh sorry

It’s an old se. I used a command and I don’t know how to check the sum.
I’m gonna try using downloaded ipsw.

nikias, 2020-05-30:
Ok I just need to be sure that ipsw.me site gives me the correct checksum

Socially-disabled, 2020-05-30:
So I tried using ipsw from ipsw.me and my phone stuck on recovery mode

nikias, 2020-05-30:
So now what, are you stuck in recovery mode? Or you can't leave normal mode? Does the device enter recovery mode but idevicerestore doesn't detect it? You have to provide more information! I can't keep guessing what might be happening on your side, you don't even paste any idevicerestore output so one could at least narrow down the source of the problem...

Socially-disabled, 2020-05-30:
> Or you can't leave normal mode? 
That’s it
>You have to provide more information! I can't keep guessing what might be happening on your side
My bad. I didn’t know that there can be different error linked with recovery mode
For now, I don’t have any info to provide as I haven’t gone any further

Socially-disabled, 2020-05-30:
Here's an output

```
idevicerestore iPhone_4.0_64bit_13.5_17F75_Restore.ipsw
Found device in Normal mode
Identified device as n69ap, iPhone8,4
Extracting BuildManifest from IPSW
Product Version: 13.5
Product Build: 17F75 Major: 17
Device supports Image4: true
Variant: Customer Upgrade Install (IPSW)
This restore will update your device without erasing user data.
Checking IPSW for required components...
All required components found in IPSW
Using cached filesystem from 'iPhone_4.0_64bit_13.5_17F75_Restore/038-34986-082.dmg'
Found ECID 638361273945894
Checking if device requires stashbag...
Device requires stashbag.
******************************************************************************
* Please enter your passcode on the device.  The device will store a token   *
* that will be used after restore to access the user data partition.  This   *
* prevents an 'Attempting data recovery' process occurring after reboot that *
* may take a long time to complete and will _also_ require the passcode.     *
******************************************************************************
Stashbag created.
Getting ApNonce in normal mode... 80 ab eb 12 98 45 8e 36 ce e4 5d 75 35 26 3f d1 2a b5 f6 61 
Trying to fetch new SHSH blob
Getting SepNonce in normal mode... f2 09 86 28 5a f2 07 74 bd 3d a2 4b c0 a4 18 5a 2c 02 e7 0f 
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received SHSH blobs
Committing stashbag...
Stashbag committed!
Entering recovery mode...
ERROR: Failed to enter recovery mode
ERROR: Unable to place device into recovery mode from normal mode
```

nikias, 2020-05-31:
Ok and now, *does* the device actually go into recovery mode?
Also, have you pulled the latest source (because there was a recent bug fix about 10 days ago)?

Socially-disabled, 2020-05-31:
> Ok and now, *does* the device actually go into recovery mode?

Still doesn’t 

> Also, have you pulled the latest source (because there was a recent bug fix about 10 days ago)?

Source of what? Idevicerestore? It’s the newest git version available in AUR

nikias, 2020-05-31:
https://aur.archlinux.org/packages/idevicerestore-git
Latest update was months ago, in December 2019.

I can't give support for anything that is not built directly from latest git head souce tree.

If the device stays in normal mode (i.e. you can still access the home screen) then something else must be wrong. Could be the dependencies are not up-to-date either (like libimobiledevice, libirecovery).

Socially-disabled, 2020-05-31:
> If the device stays in normal mode (i.e. you can still access the home screen) then something else must be wrong

Idk what is that state my phone entered, but it showed a laptop with a usb, probably suggesting to connect to some device to restore. But the terminal itself said that it failed to put device in a recovery mode

nikias, 2020-05-31:
Ok that is recovery mode. If it fails to detect the device in recovery mode your system
might lack permissions to access usb devices. Try to run the command as root.

Socially-disabled, 2020-05-31:
> Try to run the command as root.

It... It did the job

closed 2020-05-31 by Socially-disabled

nikias, 2020-05-31:
Just to clarify things, it is not intended to be run as root under normal circumstances. libirecovery has a configuration for udev which is supposed to assign a group to the usb device so that normal users can run idevicerestore too. However this is something the linux distribution/package maintainer has to take care of, and apparently didn't happen for your distribution.
Anyway, glad this was resolved!

Jackalhz1980, 2023-07-17:
idevicerestore -RPnldy ./iPhone10,3,iPhone10,6_16.5.1_20F75_Restore.ipsw.lock
progress: 0 0.000000
irecv_event_cb: device 001a40591087c02e (udid: N/A) connected in DFU mode
progress: 0 0.100000
Found device in DFU mode
opening device 05ac:1227...
found device with ECID 001a40591087c02e
Setting to configuration 1
Setting to interface 0:0
progress: 0 0.200000
Identified device as d22ap, iPhone10,3
*   Trying 104.26.9.156:443...
* Connected to api.ipsw.me (104.26.9.156) port 443 (#0)
* found 423 certificates in /etc/ssl/certs
* GnuTLS ciphers: NORMAL:-ARCFOUR-128:-CTYPE-ALL:+CTYPE-X509:-VERS-SSL3.0
* ALPN: offers h2,http/1.1
* SSL connection using TLS1.3 / ECDHE_RSA_AES_256_GCM_SHA384
*   server certificate verification SKIPPED
*   server certificate status verification SKIPPED
*   common name: sni.cloudflaressl.com (matched)
*   server certificate expiration date OK
*   server certificate activation date OK
*   certificate public key: EC/ECDSA
*   certificate version: #3
*   subject: C=US,ST=California,L=San Francisco,O=Cloudflare\, Inc.,CN=sni.cloudflaressl.com
*   start date: Sun, 05 Feb 2023 00:00:00 GMT
*   expire date: Sun, 04 Feb 2024 23:59:59 GMT
*   issuer: C=US,O=Cloudflare\, Inc.,CN=Cloudflare Inc ECC CA-3
* ALPN: server accepted h2
* using HTTP/2
* h2h3 [:method: GET]
* h2h3 [:path: /v3/device/iPhone10,3]
* h2h3 [:scheme: https]
* h2h3 [:authority: api.ipsw.me]
* h2h3 [user-agent: InetURL/1.0 idevicerestore/1.0.0]
* h2h3 [accept: */*]
* Using Stream ID: 1 (easy handle 0xd509d0)
> GET /v3/device/iPhone10,3 HTTP/2
Host: api.ipsw.me
user-agent: InetURL/1.0 idevicerestore/1.0.0
accept: */*

< HTTP/2 200 
< date: Mon, 17 Jul 2023 03:10:05 GMT
< content-type: application/json
< vary: Accept-Encoding
< vary: Origin
< access-control-allow-origin: *
< access-control-expose-headers: X-Data-Location, X-Server-Location
< cache-control: public, max-age=14400
< content-md5: 3H+ffEczLm3pqhazcSqv6A==
< content-security-policy: default-src https: 'unsafe-eval' 'unsafe-inline'; object-src 'none'
< expires: Mon, 17 Jul 2023 04:08:28 GMT
< x-content-type-options: nosniff
< x-data-location: database
< x-frame-options: DENY
< x-frame-options: DENY
< x-server-location: f4a0475957db
< x-xss-protection: 1; mode=block
< strict-transport-security: max-age=15552000; preload
< last-modified: Mon, 17 Jul 2023 03:08:28 GMT
< cf-cache-status: HIT
< report-to: {"endpoints":[{"url":"https:\/\/a.nel.cloudflare.com\/report\/v3?s=A5Bzv7HsDD0cDKjG2YunjoDl%2FSCWRX0U%2BgXc7gYXDr5x7f9gcgILy%2FujAKm7z4EKdJVj0U3iL%2B%2B6KGg0Q7xJdUw9BugLvRSe8gKa%2B2C18%2BFo%2BRuD18%2F5HYHjEqsy"}],"group":"cf-nel","max_age":604800}
< nel: {"success_fraction":0,"report_to":"cf-nel","max_age":604800}
< server: cloudflare
< cf-ray: 7e7f4a350df645f0-DFW
< alt-svc: h3=":443"; ma=86400
< 
* Connection #0 to host api.ipsw.me left intact
NOTE: Running non-interactively, automatically selecting latest available version
Selected firmware 16.5.1 (build 20F75)
Downloading firmware (https://updates.cdn-apple.com/2023SpringFCS/fullrestores/042-02106/B9658E57-8FC6-4528-9853-42C21A785755/iPhone10,3,iPhone10,6_16.5.1_20F75_Restore.ipsw)
*   Trying 17.253.127.216:443...
* Connected to updates.cdn-apple.com (17.253.127.216) port 443 (#0)
* found 423 certificates in /etc/ssl/certs
* GnuTLS ciphers: NORMAL:-ARCFOUR-128:-CTYPE-ALL:+CTYPE-X509:-VERS-SSL3.0
* ALPN: offers h2,http/1.1
* SSL connection using TLS1.3 / ECDHE_RSA_AES_128_GCM_SHA256
*   server certificate verification SKIPPED
*   server certificate status verification SKIPPED
*   common name: updates.cdn-apple.com (matched)
*   server certificate expiration date OK
*   server certificate activation date OK
*   certificate public key: RSA
*   certificate version: #3
*   subject: CN=updates.cdn-apple.com,O=Apple Inc.,ST=California,C=US
*   start date: Tue, 15 Nov 2022 18:59:47 GMT
*   expire date: Fri, 15 Dec 2023 18:59:46 GMT
*   issuer: CN=Apple IST CA 2 - G1,OU=Certification Authority,O=Apple Inc.,C=US
* ALPN: server accepted http/1.1
* using HTTP/1.1
> GET /2023SpringFCS/fullrestores/042-02106/B9658E57-8FC6-4528-9853-42C21A785755/iPhone10,3,iPhone10,6_16.5.1_20F75_Restore.ipsw HTTP/1.1
Host: updates.cdn-apple.com
User-Agent: InetURL/1.0
Accept: */*

< HTTP/1.1 200 OK
< Content-Type: binary/octet-stream
< Content-Length: 6165698328
< Date: Sat, 15 Jul 2023 14:47:58 GMT
< x-amz-replication-status: COMPLETED
< Last-Modified: Fri, 16 Jun 2023 07:12:56 GMT
< x-amz-server-side-encryption: AES256
< x-amz-meta-digest-sh1: 2e64409ecc8cdde7fd26dc068eac4d89f584f053
< x-amz-meta-digest-sha256: 2c84dc5637f60a3effbf43ce5748334da3ef0a9c4db3da5595705b7ddc73adf7
< x-amz-version-id: VB7xrZ9ffzvgNUdOMoy1y13J9G6W86hV
< Accept-Ranges: bytes
< Server: AmazonS3
< X-Cache: hit-fresh, hit-fresh, none
< Via: http/1.1 usdal4-edge-lx-011.ts.apple.com (acdn/87.14301), http/1.1 usdal4-edge-bx-026.ts.apple.com (acdn/87.14301), http/1.1 usdal4-edge-bx-026.ts.apple.com (acdn/87.14301)
< X-Amz-Cf-Pop: IAD55-P3
< X-Amz-Cf-Id: vWfIZ9_w3PeDpoTA_Ev-L4Lx4m3Jy25roOp-IxpfFIwo3OP7uWt4-w==
< Cache-Control: max-age=604800, public
< CDNUUID: 421de512-4e89-4c37-96c9-da1227782d71-7370809142
< Etag: "6aac363d1f673c1507137dfb3099c5aa-736"
< Age: 130928
< Connection: keep-alive
< 
downloading: 1%
downloading: 2%
downloading: 3%
downloading: 4%
downloading: 5%
downloading: 6%
downloading: 7%
downloading: 8%
downloading: 9%
downloading: 10%
downloading: 11%
downloading: 12%
downloading: 13%
downloading: 14%
downloading: 15%
downloading: 16%
downloading: 17%
downloading: 18%
downloading: 19%
downloading: 20%
downloading: 21%
downloading: 22%
downloading: 23%
downloading: 24%
downloading: 25%
downloading: 26%
downloading: 27%
downloading: 28%
downloading: 29%
downloading: 30%
downloading: 31%
downloading: 32%
downloading: 33%
downloading: 34%
* Failure writing output to destination
* Closing connection 0

Verifying 'iPhone10,3,iPhone10,6_16.5.1_20F75_Restore.ipsw'...
ERROR: File download failed (checksum mismatch).