libimobiledevice/idevicerestore #338: Still unable to run idevicerestore on Iphone 6

TheNegativeZero, 2020-06-03:
Hi guys, after a few days banging my head against a brick wall, "idevicerestore -l" ends on the infamous "Unable to receive message from FDR 0x55b0cfdc1f10 (-2). 0/2 bytes" error.  I modified fdr.c to allow for more retries to open the port and set a longer SLEEP wait but to no avail.

On the other hand, irecovery reports accurately when I am in either RESTORE or DFU mode.  I am able to set the iPhone 6 (A1586) to either state and even send commands thru irecovery.  When in idevicerestore, the program(s) run without any hicups until I hit the aforementioned error.

Questions:
(1). Is there a way to push the iPhone firmware thru an irecovery command (-s).
(2). Is the latest compile of idevicerestore ok?  I am also compiling libplist, libusbmuxd, libimobiledevice, usbmuxd, libirecovery, and of course idevicerestore.  All compiles finish ok just with a few warnings about unused variables.
(3), Is there any procedure to run this that I am missing?

Many thanks in advance.

nikias, 2020-06-03:
No, you can't push fw through irecovery.
`"Unable to receive message from FDR 0x55b0cfdc1f10 (-2). 0/2 bytes"` isn't the reason why it fails. I recently restored an iPhone X with the latest codebase and got a lot of those messages but it still restored fine.
What platform are you on? Any more log output? With -d ?

TheNegativeZero, 2020-06-03:
Thanks for your kind and prompt response Nikias.  The problem is that because I have my iPhone6 stuck in a boot loop, I wanted to make it bootable again.  I read that idevicerestore would clear this condition by reinstalling the .ipsw file and all required files.

Im working on Ubuntu 20.04 but have access to Win10 and MacOS if needed.  I donty want to downgrade or jailbrake the phone, just want to make it bootable again.

Many thanks again in advance.  Your help is much appreciated.

idevicerestore -l -d

irecv_event_cb: device 00162c543831a826 (udid: N/A) connected in Recovery mode
Found device in Recovery mode
opening device 05ac:1281...
found device with ECID 00162c543831a826
Setting to configuration 1
Setting to interface 0:0
Identified device as n61ap, iPhone7,2
*   Trying 172.67.83.193:443...
* TCP_NODELAY set
* Connected to api.ipsw.me (172.67.83.193) port 443 (#0)
* ALPN, offering h2
* ALPN, offering http/1.1
* successfully set certificate verify locations:
*   CAfile: /etc/ssl/certs/ca-certificates.crt
  CApath: /etc/ssl/certs
* SSL connection using TLSv1.3 / TLS_AES_256_GCM_SHA384
* ALPN, server accepted to use h2
* Server certificate:
*  subject: C=US; ST=CA; L=San Francisco; O=Cloudflare, Inc.; CN=sni.cloudflaressl.com
*  start date: Mar 19 00:00:00 2020 GMT
*  expire date: Oct  9 12:00:00 2020 GMT
*  subjectAltName: host "api.ipsw.me" matched cert's "*.ipsw.me"
*  issuer: C=US; ST=CA; L=San Francisco; O=CloudFlare, Inc.; CN=CloudFlare Inc ECC CA-2
*  SSL certificate verify ok.
* Using HTTP2, server supports multi-use
* Connection state changed (HTTP/2 confirmed)
* Copying HTTP/2 data in stream buffer to connection buffer after upgrade: len=0
* Using Stream ID: 1 (easy handle 0x561a4ac75890)
> GET /v3/device/iPhone7,2 HTTP/2
Host: api.ipsw.me
user-agent: InetURL/1.0 idevicerestore/1.0
accept: */*

* old SSL session ID is stale, removing
* Connection state changed (MAX_CONCURRENT_STREAMS == 256)!
< HTTP/2 200 
< date: Wed, 03 Jun 2020 03:04:13 GMT
< content-type: application/json
< set-cookie: __cfduid=dbf3f0e5b1011a056e175e342029780bc1591153452; expires=Fri, 03-Jul-20 03:04:12 GMT; path=/; domain=.ipsw.me; HttpOnly; SameSite=Lax; Secure
< vary: Accept-Encoding
< access-control-allow-origin: *
< access-control-expose-headers: X-Data-Location, X-Server-Location
< cache-control: public
< content-md5: 45Lg4mLuDqZ3ZCyLgR/lxA==
< content-security-policy: default-src https: 'unsafe-eval' 'unsafe-inline'; object-src 'none'
< expires: Wed, 03 Jun 2020 03:33:00 GMT
< vary: Origin
< x-content-type-options: nosniff
< x-data-location: cache
< x-frame-options: DENY
< x-server-location: 11e1a48cc0a2
< x-xss-protection: 1; mode=block
< strict-transport-security: max-age=15552000; preload
< x-frame-options: DENY
< cf-cache-status: EXPIRED
< cf-request-id: 0319bbfeab0000dec7f0b93200000001
< expect-ct: max-age=604800, report-uri="https://report-uri.cloudflare.com/cdn-cgi/beacon/expect-ct"
< server: cloudflare
< cf-ray: 59d62f777cb7dec7-GUA
< alt-svc: h3-27=":443"; ma=86400
< 
* Connection #0 to host api.ipsw.me left intact
[... 768 lines cut ...]
Personalizing IMG4 component BatteryPlugin...
Tag found
Extracting DeviceTree.n61ap.im4p...
Personalizing IMG4 component DeviceTree...
Tag found
Extracting recoverymode@1334~iphone-lightning.im4p...
Personalizing IMG4 component RecoveryMode...
Tag found
Extracting iBoot.n61.RELEASE.im4p...
Personalizing IMG4 component iBoot...
Tag found
Extracting sep-firmware.n61.RELEASE.im4p...
Personalizing IMG4 component RestoreSEP...
Tag found
Extracting sep-firmware.n61.RELEASE.im4p...
Personalizing IMG4 component SEP...
Tag found
common.c:supressed printing 9914630 bytes plist...
Sending NORData now...
Done sending NORData
ERROR: Could not read data (-256). Aborting.
idevice_event_cb: device 00162c543831a826 (udid: dde0ac13d3e811b8273a6774f557143d93c81a77) disconnected
irecv_event_cb: device 00162c543831a826 (udid: dde0ac13d3e811b8273a6774f557143d93c81a77) connected in Recovery mode
FDR 0x561a4ad9bf10 terminating...
ERROR: Unable to restore device

TheNegativeZero, 2020-06-03:
[errorlog.txt](https://github.com/libimobiledevice/idevicerestore/files/4720840/errorlog.txt)
much clearer.

TheNegativeZero, 2020-06-03:
[Summary Log.txt](https://github.com/libimobiledevice/idevicerestore/files/4721048/Summary.Log.txt)

TheNegativeZero, 2020-06-04:
Hi Nikias, did you receive the idevicerestore logs that you requested?
Any hint about how to make this work is greatly appreciated.  Thanks in advance.

TheNegativeZero, 2020-06-08:
Hi @nikias , did you receive the idevicerestore logs that you requested?
Any hint about how to make this work is greatly appreciated. I took a look at all libraries upstream but cannot pinpoint the error as everything compiles ok.
I will continue crunching code but it would be very helpful to hear from you ideas as to where to look for this bug.
Thanks in advance.

nikias, 2020-06-08:
@TheNegativeZero I get every single message on any of these issues. It looks like the connection just fails (after sending the NORData). Have you made sure that you are using the latest code for all the depending libs? libirecovery, libplist, libusbmuxd, and libimobiledevice?

TheNegativeZero, 2020-06-08:
Hi @nikias, I know you are very busy and I dont want to take too much time from you.  
Summary: iPhone 6, Ubuntu 20.04, 4 different USB cables.  Increased timeout, printed checkpoints (below, with --).  FDR fails.  Everything points at something upstream (libimobiledevice, libusbmuxd, etc.) but cannot put my finger on it.

Sending NORData now...
Done sending NORData
Could not read data (-256). Aborting.
--   FDR: 0x5650650e74b0 
--   Device Error: -2 
--   IDEVICE_E_TIMEOUT: -7 
--   IDEVICE_E_SUCCESS: 0 
--   cmd: 32589 
--   sizeof(cmd): 2 
--   bytes: 0
ERROR: Unable to restore device

In every complite I GIT CLONE libirecovery, idevicerestore, usbmuxd, libimobiledevice, libusbmuxd, libplist.  Also installed/updated libcurl4-openssl-dev libplist-dev libzip-dev openssl libssl-dev libusb-1.0-0-dev libreadline-dev build-essential git make automake libtool pkg-config ifuse (some are old and do not install).

Seems pretty straight forward and I have done this before flawlessly.  Dont know why it does not work now.

Thanks in advance.  Appreciated.

yogendra123456, 2020-06-10:
Hi All,

i try to restore my device but it create error again and again on windows 10... i am using libmobiledevice code.

Extracting recoverymode@1334~iphone-lightning.im4p...
Personalizing IMG4 component RecoveryMode...
Extracting iBoot.n71.RELEASE.im4p...
Personalizing IMG4 component iBoot...
Extracting sep-firmware.n71.RELEASE.im4p...
Personalizing IMG4 component RestoreSEP...
Extracting sep-firmware.n71.RELEASE.im4p...
Personalizing IMG4 component SEP...
Sending NORData now...
Done sending NORData
Unmounting filesystems (29)
Unmounting filesystems (29)
Unmounting filesystems (29)
Unmounting filesystems (29)
Checking filesystems (15)
ERROR: Unable to receive message from FDR 00000124E29C2130 (-7). 0/2 bytes

Details in attached file..
[ErrorRelease.txt](https://github.com/libimobiledevice/idevicerestore/files/4757500/ErrorRelease.txt)

kindly provide the update what is the solution of this error, i am using latest libmobiledevice code with visual studio 2019...

thanks in advanced....

yogendra123456, 2020-06-11:
Hi @TheNegativeZero ,

thanks for concern, if you work on windows and find any solution then mail me...

TheNegativeZero, 2020-06-14:
hi @nikias, after several recompiles I have learned a few things but the problem persists:

1. I have gitcloned and recompiled all 5 libraries in addition to idevicerestore.  All libs were generated on /usr/local/lib while idevicerestore is on /usr/local/bin.  LDD idevicerestore checks out.

2. installed/updated every single library dependency.  ==> running Ubuntu 20.04

3. Added checkpoints to trace the error message in the code for fdr.c and restore.c in idevicerestore.  Also increased connection retries and sleep (wait) times.

4. Run usbmuxd -f -v -v to track down the moment idevicerestore fails.  Found out that usbmuxd shows an error -71 before shutting down the connection.  I read somewhere that this is caused by a timeout waiting for an ack from the phone:

[45.784566] [00022f43] libusb: debug [libusb_free_transfer] transfer 0x564f488fe020
[45.784604] [00022f43] libusb: debug [reap_for_handle] urb type=3 status=-71 transferred=0
[45.784633] [00022f43] libusb: debug [handle_bulk_completion] handling completion status -71 of bulk urb 1/1
[45.784662] [00022f43] libusb: debug [handle_bulk_completion] low level error -71
[45.784688] [00022f43] libusb: debug [usbi_handle_transfer_completion] transfer 0x564f48921860 has callback 0x564f469788b0
[14:28:24.099][4] Device 1-18 RX aborted due to error or disconnect

5. Irecovery works well and idevicerestore runs well until it hits the error.  If just out of fun I run idevicerestore -l -x, the program runs to completion, obviously without copying any firmware.

6. This info leads me to believe that somehow the bulk transfer between Linux (idevicerestore and libs) and the iPhone screws up, maybe at the libusbmuxd level?  I am looking for ways to test this connection without using idevicerestore.  What puzzles me is that I am using the latest libusbmuxd and usbmuxd libs as published on this git.

I have tested with original Apple cable, different usb ports, different laptop (Ubuntu 18.04).
I have run out of ideas.  --Thanks.

TheNegativeZero, 2020-06-26:
Closing this unresolved issue as no solution is on sight.

closed 2020-06-26 by TheNegativeZero

nikias, 2023-01-23:
I wouldn't give up on this. @MrSaatci what error situations are you confronted with?

MrSaatci, 2023-01-23:
It is weird, I randomly fail at either entering restore mode (which goes into recovery instead) or I get the FDR error.

Sending RestoreDeviceTree (125714 bytes)...
DEBUG: tss_response_get_path_by_entry: No entry 'RestoreKernelCache' in TSS response
NOTE: No path for component RestoreKernelCache in TSS, will fetch from build_identity
Extracting kernelcache.release.iphone7...
Personalizing IMG4 component RestoreKernelCache...
Tag found
Sending RestoreKernelCache (14069209 bytes)...
DEBUG: Waiting for device to disconnect...
irecv_event_cb: device 001d595218f94f26 (udid: N/A) disconnected
Waiting for device to enter restore mode...
irecv_event_cb: device 001d595218f94f26 (udid: N/A) connected in Recovery mode
ERROR: Device failed to enter restore mode.

FDR

FDR Sent 89 bytes
FDR Received 105 bytes
common.c:printing 334 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>Command</key>
	<string>BeginCtrl</string>
	<key>CtrlProtoVersion</key>
	<integer>2</integer>
	<key>ConnPort</key>
	<integer>49159</integer>
</dict>
</plist>
Ctrl handshake done (ConnPort = 49159)
TZ0RequiredCapacity: 0xC00000
FDR 0x55a8dc46ed30 waiting for message...
About to send RootTicket...
Sending RootTicket now...
Done sending RootTicket
Waiting for NAND (28)
ERROR: Could not read data (-256). Aborting.
ERROR: Unable to receive message from FDR 0x55a8dc46ed30 (-2). 0/2 bytes
FDR 0x55a8dc46ed30 terminating...
ERROR: Unable to restore device
idevice_event_cb: device 001d595218f94f26 (udid: ffffffffffffffffffffffffffffffff00000001) disconnected
eden@eden-ubuntu:~$ sudo idevicerestore -