libimobiledevice/idevicerestore #339: Unable To Restore iPhoneX

PwnSource, 2020-06-05:
irecv_event_cb: device 001c61402853a02e (udid: N/A) connected in Recovery mode
Found device in Recovery mode
opening device 05ac:1281...
found device with ECID null
Setting to configuration 1
Setting to interface 0:0
Identified device as d22ap, iPhone10,3
Extracting BuildManifest from IPSW
Product Version: 13.5.5
Product Build: 17G5035d Major: 17
opening device 05ac:1281...
found device with ECID 001c61402853a02e
Setting to configuration 1
Setting to interface 0:0
INFO: device serial number is NULL
Device supports Image4: true
Variant: Developer Erase Install (IPSW)
This restore will erase your device data.
Checking IPSW for required components...
All required components found in IPSW
Extracting filesystem from IPSW: 038-57002-008.dmg
[==================================================] 100.0%
Getting device's ECID for TSS request
Found ECID 7988227530268718
Getting ApNonce in recovery mode... 83 ca de 36 b7 31 b0 42 f8 1f fc 5c 57 e5 f0 ca 31 cb 67 93 3f 71 75 7a 6e 7b 67 9e b0 a4 c8 5d 
Trying to fetch new SHSH blob
Getting SepNonce in recovery mode... 97 d3 ed 63 76 05 d0 44 71 c7 3c 93 fc 0b 15 7a cb 86 77 d2 
NOTE: Unable to find BbCalibrationManifestKeyHash node
NOTE: Unable to find BbFactoryActivationManifestKeyHash node
NOTE: Unable to find BbSkeyId node
DEBUG: Applying restore request rules for entry AOP
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry AppleLogo
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry AudioCodecFirmware
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry BatteryCharging0
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry BatteryCharging1
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry BatteryFull
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry BatteryLow0
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry BatteryLow1
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry BatteryPlugin
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry DeviceTree
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry ISP
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry KernelCache
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry LLB
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry Liquid
[... 2543 lines cut ...]
restore-step-ids = {0x11030677:41;0x1103065B:49}
restore-step-names = {0x11030677:perform_main_os_prepare;0x1103065B:asr_and_invert_image}
restore-step-uptime = 61
restore-step-user-progress = 15
[03:23:13.0869-GMT]{3>4} CHECKPOINT BEGIN: RESTORED:[0x0647] cleanup_check_result
restore-step-ids = {0x11030677:41;0x1103065B:49;0x11030647:52}
restore-step-names = {0x11030677:perform_main_os_prepare;0x1103065B:asr_and_invert_image;0x11030647:cleanup_check_result}
restore-step-uptime = 61
restore-step-user-progress = 15
[03:23:13.0870-GMT]{3>4} CHECKPOINT END: RESTORED:[0x0647] cleanup_check_result
restore-step-ids = {0x11030677:41;0x1103065B:49}
restore-step-names = {0x11030677:perform_main_os_prepare;0x1103065B:asr_and_invert_image}
restore-step-uptime = 61
restore-step-user-progress = 15
[03:23:13.0870-GMT]{3>4} CHECKPOINT BEGIN: RESTORED:[0x0648] cleanup_send_final_status
restore-step-ids = {0x11030677:41;0x1103065B:49;0x11030648:53}
restore-step-names = {0x11030677:perform_main_os_prepare;0x1103065B:asr_and_invert_image;0x11030648:cleanup_send_final_status}
restore-step-uptime = 61
restore-step-user-progress = 15

FDR 0x560d97962080 timeout waiting for command
FDR 0x560d97962080 terminating...
ERROR: Unable to restore device

**Any ideas?**

nikias, 2020-06-05:
The connection fails. Are you using a 3rd party cable or a usb hub?

PwnSource, 2020-06-06:
No, I’m using either an Apple original lightning adapter or a dcsd cable. Both of these result in the same issue. This started happening upon release of iOS 13.5.1 and it takes several attempts 1:3 before it will go through with the restore. 13.5.5, on the other hand, fails every time. This device also will only take the 13.5.1 update in recovery mode and not any other mode like DFU.

I’m not sure if Apple changed something? I noticed iProxy no longer works for Linux either?

demhademha, 2020-06-06:
Yep, iproxy doesn’t work for me either 
On Ubuntu

nikias, 2020-06-06:
I introduced a bug in iproxy where it only listens on ipv6 socket. Until I fix it either revert to an earlier commit or add`-s 127.0.0.1` to the command line.

PwnSource, 2020-06-07:
The restore in the other hand, is it possible that it’s a bug?

nikias, 2020-06-07:
iproxy is now fixed in latest git master https://github.com/libimobiledevice/libusbmuxd

PwnSource, 2020-06-08:
Thank you nikias, This commit fixed iProxy and the restore problem! I really appreciate!

Closing issue!

closed 2020-06-08 by PwnSource