libimobiledevice/idevicerestore #340: idevicerestore: undefined symbol: preboard_client_new

Texlo-Dev, 2020-06-06:
Hey all, compiled and installed idevicerestore recently. Tried to run `idevicerestore -l` and I get the following error: 

```
idevicerestore: undefined symbol: preboard_client_new
```

What does this mean? For reference, I'm on Ubuntu 18.04 LTS.

closed 2020-06-06 by Texlo-Dev

grayguest, 2020-06-08:
> Hey all, compiled and installed idevicerestore recently. Tried to run `idevicerestore -l` and I get the following error:
> 
> ```
> idevicerestore: undefined symbol: preboard_client_new
> ```
> 
> What does this mean? For reference, I'm on Ubuntu 18.04 LTS.

hello, i met the same problem, can you tell me some solution policy?

nikias, 2020-06-08:
libimobiledevice is too old and you need to compile an updated version yourself.