libimobiledevice/idevicerestore #342: T2 restore failed with ERROR: Unable to find any build identities

zerishpho, 2020-06-15:
i face this issue with the following models only iBridge2,14,iBridge2,15,iBridge2,16,iBridge2,21,iBridge2,22 
@rickmark @FunkyM @nikias 
verifying 'ibridge2,1 .....
checksum matches.
extracting buildmanifest from ipsw
product version: 4.5
product build: 17P5300 Major:17
opening device ..
setting to configuration 1
setting to interface 0:0
device supports image4: true
ERROR: unable to find any build identities

zerishpho, 2020-06-15:
[image]

rickmark, 2020-06-15:
From my understanding a number of those model may not be public at this time:

If they are feel free to provide us with the matching Board-ID etc for them in the same way as:
https://www.theiphonewiki.com/wiki/T8012

zerishpho, 2020-06-15:
what info do you need that i may provide from the device? or is it you mean that apple still didn't release there product ID?

rickmark, 2020-06-15:
So we've tracked it down to two things:

1) We don't have a good mapping of iBridge IPSWs to their Apple MacBook/MacMini etc models.  So far we've had to rely on someone with a product booting into DFU to collect this information like here: https://www.theiphonewiki.com/wiki/T8012

2) `libimobiledevice/libirestore` requires this mapping, and has to be updated for every release.  We're discussing adding a "wildcard T2" to allow recovery to proceed even when the release hasn't been added to the map.

zerishpho, 2020-06-16:
ok ive dozens of this devices i can get you the information you need. also ive send you an email on your outlook. please check and let me know thanks!