libimobiledevice/idevicerestore #347: Unable to restore: failing to send NORImageData data

mcg29, 2020-06-24:
device in Recovery mode
Identified device as n104ap, iPhone12,1
Extracting BuildManifest from IPSW
Product Version: 14.0
Product Build: 18A5301v Major: 18
INFO: device serial number is ************
Device supports Image4: true
Variant: Developer Erase Install (IPSW)
This restore will erase your device data.
################################ [ WARNING ] #################################
# You are about to perform an *ERASE* restore. ALL DATA on the target device #
# will be IRREVERSIBLY DESTROYED. If you want to update your device without  #
# erasing the user data, hit CTRL+C now and restart without -e or --erase    #
# command line switch.                                                       #
# If you want to continue with the ERASE, please type YES and press ENTER.   #
##############################################################################
> YES
Checking IPSW for required components...
All required components found in IPSW
Using cached filesystem from '/Volumes/Done_for_now/iPhone 11, iPhone XR/038-44337-056.dmg'
Found ECID ****************
Getting ApNonce in recovery mode... f6 d6 31 ac 1b 02 92 08 00 e1 5a 75 0b 78 24 0d 4c d5 9c be 62 65 bf 1b 04 8b be df ea dd 08 73 
Trying to fetch new SHSH blob
Getting SepNonce in recovery mode... aa 3b a0 54 df 5b be 7a 9d 3c 2b d5 7e 7f 8c a7 26 fb 7b ff 
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received SHSH blobs
Extracting iBEC.n104.RELEASE.im4p...
Personalizing IMG4 component iBEC...
Sending iBEC (1565965 bytes)...
Recovery Mode Environment:
iBoot build-version=iBoot-6603.0.0.110.6
iBoot build-style=RELEASE
Sending RestoreLogo...
Extracting applelogo@1792~iphone.im4p...
Personalizing IMG4 component RestoreLogo...
Sending RestoreLogo (20509 bytes)...
Extracting h12_ane_fw_metis.im4p...
Personalizing IMG4 component ANE...
Sending ANE (12684834 bytes)...
Extracting aopfw-iphone12baop.im4p...
Personalizing IMG4 component AOP...
Sending AOP (1929994 bytes)...
Extracting AppleAVE2FW_H12.im4p...
Personalizing IMG4 component AVE...
Sending AVE (1148826 bytes)...
Extracting armfw_g12p.im4p...
Personalizing IMG4 component GFX...
Sending GFX (768930 bytes)...
Extracting adc-zelus-n104.im4p...
Personalizing IMG4 component ISP...
Sending ISP (11492234 bytes)...
Extracting t8030pmp.im4p...
Personalizing IMG4 component PMP...
Sending PMP (299786 bytes)...
Extracting 038-44135-093.dmg.trustcache...
Personalizing IMG4 component RestoreTrustCache...
Sending RestoreTrustCache (13809 bytes)...
Extracting SmartIOFirmware_ASCv2.im4p...
Personalizing IMG4 component SIO...
Sending SIO (1161694 bytes)...
Extracting WirelessPower.iphone12b.im4p...
Personalizing IMG4 component WCHFirmwareUpdater...
Sending WCHFirmwareUpdater (57129 bytes)...
ramdisk-size=0x20000000
Extracting 038-44135-093.dmg...
Personalizing IMG4 component RestoreRamDisk...
Sending RestoreRamDisk (103469325 bytes)...
Extracting DeviceTree.n104ap.im4p...
Personalizing IMG4 component RestoreDeviceTree...
Sending RestoreDeviceTree (227259 bytes)...
Extracting kernelcache.release.iphone12b...
Personalizing IMG4 component RestoreKernelCache...
Sending RestoreKernelCache (15523033 bytes)...
Waiting for device to enter restore mode...
About to restore device... 
Connecting now...
Connected to com.apple.mobile.restored, version 15
Device 00008030-0014316C3A63802E has successfully entered restore mode
Hardware Information:
BoardID: 4
ChipID: 32816
UniqueChipID: 5683840440041518
ProductionMode: true
Previous restore exit status: 0x100
Starting FDR listener thread
About to send RootTicket...
Sending RootTicket now...
Done sending RootTicket
Waiting for NAND (28)
Updating NAND Firmware (58)
About to send NORData...
Found firmware path Firmware/all_flash
Getting firmware manifest from build identity
Extracting LLB.n104.RELEASE.im4p...
Personalizing IMG4 component LLB...
Extracting applelogo@1792~iphone.im4p...
Personalizing IMG4 component AppleLogo...
Extracting batterycharging0@1792~iphone.im4p...
Personalizing IMG4 component BatteryCharging0...
Extracting batterycharging1@1792~iphone.im4p...
Personalizing IMG4 component BatteryCharging1...
Extracting batteryfull@2x~iphone.im4p...
Personalizing IMG4 component BatteryFull...
Extracting batterylow0@2x~iphone.im4p...
Personalizing IMG4 component BatteryLow0...
Extracting batterylow1@2x~iphone.im4p...
Personalizing IMG4 component BatteryLow1...
Extracting glyphplugin@1792~iphone-lightning.im4p...
Personalizing IMG4 component BatteryPlugin...
Extracting DeviceTree.n104ap.im4p...
Personalizing IMG4 component DeviceTree...
Extracting lowpowermode@1792~iphone-lightning.im4p...
Personalizing IMG4 component LowPowerWallet0...
Extracting recoverymode@1792~iphone-lightning.im4p...
Personalizing IMG4 component RecoveryMode...
Extracting iBoot.n104.RELEASE.im4p...
Personalizing IMG4 component iBoot...
Extracting sep-firmware.n104.RELEASE.im4p...
Personalizing IMG4 component RestoreSEP...
Extracting sep-firmware.n104.RELEASE.im4p...
Personalizing IMG4 component SEP...
Sending NORData now...
ERROR: Unable to send NORImageData data
ERROR: Unable to send NOR data
ERROR: Unable to successfully restore device
ERROR: Unable to restore device

nikias, 2020-06-24:
Can you run with -d ? Attach the last few ~100 lines. Also please attach the BuildManifest.plist from the ipsw.

mcg29, 2020-06-24:
> Can you run with -d ? Attach the last few ~100 lines. Also please attach the BuildManifest.plist from the ipsw.

ProductionMode: true
Starting FDR listener thread
Connecting to FDR client at port 1082
About to do ctrl handshake
FDR sending 89 bytes:
/Users/noname/libimobiledevice/idevicerestore/src/common.c:printing 287 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>Command</key>
	<string>BeginCtrl</string>
	<key>CtrlProtoVersion</key>
	<integer>2</integer>
</dict>
</plist>
FDR Sent 89 bytes
FDR Received 105 bytes
/Users/noname/libimobiledevice/idevicerestore/src/common.c:printing 334 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>Command</key>
	<string>BeginCtrl</string>
	<key>CtrlProtoVersion</key>
	<integer>2</integer>
	<key>ConnPort</key>
	<integer>49174</integer>
</dict>
</plist>
Ctrl handshake done (ConnPort = 49174)
FDR 0x7f809b47cfe0 waiting for message...
About to send RootTicket...
Sending RootTicket now...
Done sending RootTicket
Waiting for NAND (28)
Updating NAND Firmware (58)
About to send NORData...
DEBUG: tss_response_get_path_by_entry: No entry 'LLB' in TSS response
NOTE: Could not get LLB path from TSS data, will fetch from build identity
Found firmware path Firmware/all_flash
Getting firmware manifest from build identity
Extracting LLB.n104.RELEASE.im4p...
Personalizing IMG4 component LLB...
Tag found
Extracting applelogo@1792~iphone.im4p...
Personalizing IMG4 component AppleLogo...
Tag found
Extracting batterycharging0@1792~iphone.im4p...
Personalizing IMG4 component BatteryCharging0...
Tag found
Extracting batterycharging1@1792~iphone.im4p...
Personalizing IMG4 component BatteryCharging1...
Tag found
Extracting batteryfull@2x~iphone.im4p...
Personalizing IMG4 component BatteryFull...
Tag found
Extracting batterylow0@2x~iphone.im4p...
Personalizing IMG4 component BatteryLow0...
Tag found
Extracting batterylow1@2x~iphone.im4p...
Personalizing IMG4 component BatteryLow1...
Tag found
Extracting glyphplugin@1792~iphone-lightning.im4p...
Personalizing IMG4 component BatteryPlugin...
Tag found
Extracting DeviceTree.n104ap.im4p...
Personalizing IMG4 component DeviceTree...
Tag found
Extracting lowpowermode@1792~iphone-lightning.im4p...
Personalizing IMG4 component LowPowerWallet0...
Tag found
Extracting recoverymode@1792~iphone-lightning.im4p...
Personalizing IMG4 component RecoveryMode...
Tag found
Extracting iBoot.n104.RELEASE.im4p...
Personalizing IMG4 component iBoot...
Tag found
Extracting sep-firmware.n104.RELEASE.im4p...
Personalizing IMG4 component RestoreSEP...
Tag found
Extracting sep-firmware.n104.RELEASE.im4p...
Personalizing IMG4 component SEP...
Tag found
/Users/noname/libimobiledevice/idevicerestore/src/common.c:supressed printing 26969424 bytes plist...
Sending NORData now...
ERROR: Unable to send NORImageData data
ERROR: Unable to send NOR data
ERROR: Unable to successfully restore device

nikias, 2020-06-24:
Hmm it looks okay, can you still extract the build manifest from ipsw and attach it here?

mcg29, 2020-06-24:
> Hmm it looks okay, can you still extract the build manifest from ipsw and attach it here?

sorry about that, here it is: [https://drive.google.com/file/d/19qiW1ws9rI2BEMWUEEeGbWa5NqseMo6y/view?usp=sharing](url)

nikias, 2020-06-25:
So it worked for this guy: https://twitter.com/im4ch3t3/status/1275936810670133248?s=21
iPhone 11 he said which is what you have too. Maybe it's a completely different problem, can you configure libimobiledevice with `--enable-debug` so we get libimobiledevice debug output when running idevicerestore with -d ?

mcg29, 2020-06-26:
> So it worked for this guy: https://twitter.com/im4ch3t3/status/1275936810670133248?s=21
> iPhone 11 he said which is what you have too. Maybe it's a completely different problem, can you configure [...]

compiled and configured the newest libimobiledevice and attempted a restore with -d, here is more of the log:

DEBUG: tss_response_get_path_by_entry: No entry 'RestoreKernelCache' in TSS response
NOTE: No path for component RestoreKernelCache in TSS, will fetch from build_identity
Extracting kernelcache.release.iphone12b...
Personalizing IMG4 component RestoreKernelCache...
Tag found
Sending RestoreKernelCache (15523033 bytes)...
DEBUG: Waiting for device to disconnect...
irecv_event_cb: device 0014316c3a63802e (udid: 00008030-0014316C3A63802E) disconnected
Waiting for device to enter restore mode...
idevice_event_cb: device 0014316c3a63802e (udid: 00008030-0014316C3A63802E) connected in restore mode
About to restore device... 
restore_is_current_device: Connected to com.apple.mobile.restored, version 15
Connecting now...
Connected to com.apple.mobile.restored, version 15
Device **************************** has successfully entered restore mode
Hardware Information:
BoardID: 4
ChipID: 32816
UniqueChipID: **********************
ProductionMode: true
Previous restore exit status: 0x100
Starting FDR listener thread
Connecting to FDR client at port 1082
About to do ctrl handshake
FDR sending 89 bytes:
idevicerestore/src/common.c:printing 287 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>Command</key>
	<string>BeginCtrl</string>
	<key>CtrlProtoVersion</key>
	<integer>2</integer>
</dict>
</plist>
FDR Sent 89 bytes
FDR Received 105 bytes
idevicerestore/src/common.c:printing 334 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>Command</key>
	<string>BeginCtrl</string>
	<key>CtrlProtoVersion</key>
	<integer>2</integer>
	<key>ConnPort</key>
	<integer>49174</integer>
</dict>
</plist>
Ctrl handshake done (ConnPort = 49174)
FDR 0x7fbfb471bc10 waiting for message...
About to send RootTicket...
Sending RootTicket now...
Done sending RootTicket
Waiting for NAND (28)
Updating NAND Firmware (58)
About to send NORData...
DEBUG: tss_response_get_path_by_entry: No entry 'LLB' in TSS response
NOTE: Could not get LLB path from TSS data, will fetch from build identity
Found firmware path Firmware/all_flash
Getting firmware manifest from build identity
Extracting LLB.n104.RELEASE.im4p...
Personalizing IMG4 component LLB...
Tag found
Extracting applelogo@1792~iphone.im4p...
Personalizing IMG4 component AppleLogo...
Tag found
Extracting batterycharging0@1792~iphone.im4p...
Personalizing IMG4 component BatteryCharging0...
Tag found
Extracting batterycharging1@1792~iphone.im4p...
Personalizing IMG4 component BatteryCharging1...
Tag found
Extracting batteryfull@2x~iphone.im4p...
Personalizing IMG4 component BatteryFull...
Tag found
Extracting batterylow0@2x~iphone.im4p...
Personalizing IMG4 component BatteryLow0...
Tag found
Extracting batterylow1@2x~iphone.im4p...
Personalizing IMG4 component BatteryLow1...
Tag found
Extracting glyphplugin@1792~iphone-lightning.im4p...
Personalizing IMG4 component BatteryPlugin...
Tag found
Extracting DeviceTree.n104ap.im4p...
Personalizing IMG4 component DeviceTree...
Tag found
Extracting lowpowermode@1792~iphone-lightning.im4p...
Personalizing IMG4 component LowPowerWallet0...
Tag found
Extracting recoverymode@1792~iphone-lightning.im4p...
Personalizing IMG4 component RecoveryMode...
Tag found
Extracting iBoot.n104.RELEASE.im4p...
Personalizing IMG4 component iBoot...
Tag found
Extracting sep-firmware.n104.RELEASE.im4p...
Personalizing IMG4 component RestoreSEP...
Tag found
Extracting sep-firmware.n104.RELEASE.im4p...
Personalizing IMG4 component SEP...
Tag found
idevicerestore/src/common.c:supressed printing 26969424 bytes plist...
Sending NORData now...
ERROR: Unable to send NORImageData data
ERROR: Unable to send NOR data
ERROR: Unable to successfully restore device
FDR 0x7fbfb471bc10 timeout waiting for command
FDR 0x7fbfb471bc10 waiting for message...
No data to read (timeout)
FDR 0x7fbfb471bc10 terminating...
ERROR: Unable to restore device

mcg29, 2020-06-26:
Here is the full log:

device_event_cb: device 0014316c3a63802e (udid: ****************************) connected in normal mode
Found device in Normal mode
Identified device as n104ap, iPhone12,1
Extracting BuildManifest from IPSW
Product Version: 14.0
Product Build: 18A5301v Major: 18
Device supports Image4: true
Variant: Developer Upgrade Install (IPSW)
This restore will update your device without erasing user data.
Checking IPSW for required components...
All required components found in IPSW
Using cached filesystem from 'iPhone 11, iPhone XR/038-44337-056.dmg'
Getting device's ECID for TSS request
Found ECID ****************************
Checking if device requires stashbag...
NOTE: Unable to find BbCalibrationManifestKeyHash node
NOTE: Unable to find BbFactoryActivationManifestKeyHash node
NOTE: Unable to find BbSkeyId node
DEBUG: Applying restore request rules for entry ANE
DEBUG: Adding EPRO=false to TSS entry
DEBUG: Adding ESEC=false to TSS entry
DEBUG: Applying restore request rules for entry AOP
DEBUG: Adding EPRO=false to TSS entry
DEBUG: Adding ESEC=false to TSS entry
DEBUG: Applying restore request rules for entry AVE
DEBUG: Adding EPRO=false to TSS entry
DEBUG: Adding ESEC=false to TSS entry
DEBUG: tss_request_add_ap_tags: Skipping 'Ap,HapticAssets' as it is not trustedDEBUG: tss_request_add_ap_tags: Skipping 'Ap,SystemVolumeCanonicalMetadata' as it is neither firmware nor secondary nor FUD firmware payload
DEBUG: Applying restore request rules for entry AppleLogo
DEBUG: Adding EPRO=false to TSS entry
DEBUG: Adding ESEC=false to TSS entry
DEBUG: tss_request_add_ap_tags: Skipping 'AudioCodecFirmware' as it is not trustedDEBUG: tss_request_add_ap_tags: Skipping 'BMU,DigestMap' as it is not trustedDEBUG: tss_request_add_ap_tags: Skipping 'BMU,FirmwareMap' as it is not trustedDEBUG: Applying restore request rules for entry BatteryCharging0
DEBUG: Adding EPRO=false to TSS entry
DEBUG: Adding ESEC=false to TSS entry
DEBUG: Applying restore request rules for entry BatteryCharging1
DEBUG: Adding EPRO=false to TSS entry
DEBUG: Adding ESEC=false to TSS entry
DEBUG: Applying restore request rules for entry BatteryFull
DEBUG: Adding EPRO=false to TSS entry
DEBUG: Adding ESEC=false to TSS entry
DEBUG: Applying restore request rules for entry BatteryLow0
DEBUG: Adding EPRO=false to TSS entry
DEBUG: Adding ESEC=false to TSS entry
DEBUG: Applying restore request rules for entry BatteryLow1
DEBUG: Adding EPRO=false to TSS entry
DEBUG: Adding ESEC=false to TSS entry
DEBUG: Applying restore request rules for entry BatteryPlugin
DEBUG: Adding EPRO=false to TSS entry
DEBUG: Adding ESEC=false to TSS entry
DEBUG: Applying restore request rules for entry DeviceTree
DEBUG: Adding EPRO=false to TSS entry
DEBUG: Adding ESEC=false to TSS entry
DEBUG: Applying restore request rules for entry GFX
DEBUG: Adding EPRO=false to TSS entry
DEBUG: Adding ESEC=false to TSS entry
DEBUG: Applying restore request rules for entry ISP
DEBUG: Adding EPRO=false to TSS entry
DEBUG: Adding ESEC=false to TSS entry
DEBUG: tss_request_add_ap_tags: Skipping 'KernelCache' as it is neither firmware nor secondary nor FUD firmware payload
DEBUG: Applying restore request rules for entry LLB
DEBUG: Adding EPRO=false to TSS entry
DEBUG: Adding ESEC=false to TSS entry
DEBUG: tss_request_add_ap_tags: Skipping 'LeapHaptics' as it is not trustedDEBUG: Applying restore request rules for entry LowPowerWallet0
DEBUG: Adding EPRO=false to TSS entry
DEBUG: Adding ESEC=false to TSS entry
DEBUG: tss_request_add_ap_tags: Skipping 'Multitouch' as it is not trustedDEBUG: tss_request_add_ap_tags: Skipping 'OS' as it is neither firmware nor secondary nor FUD firmware payload
DEBUG: Applying restore request rules for entry PMP
DEBUG: Adding EPRO=false to TSS entry
[... 2033 lines cut ...]
Tag found
Extracting lowpowermode@1792~iphone-lightning.im4p...
Personalizing IMG4 component LowPowerWallet0...
Tag found
Extracting recoverymode@1792~iphone-lightning.im4p...
Personalizing IMG4 component RecoveryMode...
Tag found
Extracting iBoot.n104.RELEASE.im4p...
Personalizing IMG4 component iBoot...
Tag found
Extracting sep-firmware.n104.RELEASE.im4p...
Personalizing IMG4 component RestoreSEP...
Tag found
Extracting sep-firmware.n104.RELEASE.im4p...
Personalizing IMG4 component SEP...
Tag found
idevicerestore/src/common.c:supressed printing 26969424 bytes plist...
Sending NORData now...
ERROR: Unable to send NORImageData data
ERROR: Unable to send NOR data
ERROR: Unable to successfully restore device
FDR 0x7fad14f44430 timeout waiting for command
FDR 0x7fad14f44430 waiting for message...
No data to read (timeout)
FDR 0x7fad14f44430 terminating...

nikias, 2020-06-26:
It still doesn't show the libimobiledevice debug output so you didn't pass `--enable-debug` to autogen.sh for libimobiledevice. You can also change these lines in idevicerestore:
https://github.com/libimobiledevice/idevicerestore/blob/master/src/restore.c#L1218-L1219
to
```
	restored_error_t rerr = restored_send(restore, dict);
	if (rerr != RESTORE_E_SUCCESS) {
		error("ERROR: Unable to send NORData, error %d\n", rerr);
```
So we get the error code.

mcg29, 2020-06-26:
Update: recompiled idevicerestore 1.0.1 and libimobiledevice 1.3.1, and this time the restore succeeded

nikias, 2020-06-26:
Okay thanks!

closed 2020-06-26 by nikias