libimobiledevice/idevicerestore #350: ompilation error on the latest version of idevicerestore on windows

iFred09, 2020-06-28:
Hello,

I'm trying to compile the latest commit of idevicerestore and I have this error, and I never had this error before (i successfully compiled this two months ago).
Same error when I compile with the version in the "Release" tab.

[image]

Thank's!

nikias, 2020-06-28:
Well the configure script sets `-O2` which is clearly not negative. You can run `make V=1` to get verbose output and check what it passes as `-O`.

iFred09, 2020-07-04:
I got this when I run `make V=1` :

[image]

nikias, 2020-07-04:
As you can see in the output there's a `-O2-DWIN32` which lacks a space.
I will commit a change adding a space here: https://github.com/libimobiledevice/idevicerestore/blob/master/configure.ac#L56

nikias, 2020-07-04:
See ac01193.

closed 2020-07-04 by nikias