libimobiledevice/idevicerestore #351: ERROR: Unable to receive message from FDR 000001A1EABB7DC0 (-2). 0/2 bytes

windowalter, 2020-07-02:
Hi,
I am try to restore my device using  idevicerestore having pramater '-l' but i got an error "ERROR: Unable to receive message from FDR 000001A1EABB7DC0 (-2). 0/2 bytes" in end. I am using latest code libimobiledevice and after searching on web ,I got no specific answer which solve my problem. 
One answer I got on web is replace CPPFLAGS instead of libimobiledevice_CFLAGS from issue #284. But I can`t understand where or  in specific file I have to replace this.
My output Given Below:-

Done sending NORData
13:35:34 C:\libimobiledevice-vs-master\libimobiledevice\src\property_list_service.c:205 internal_plist_receive_timeout(): initial read=4
13:35:34 C:\libimobiledevice-vs-master\libimobiledevice\src\property_list_service.c:211 internal_plist_receive_timeout(): 327 bytes following
13:35:34 C:\libimobiledevice-vs-master\libimobiledevice\src\property_list_service.c:224 internal_plist_receive_timeout(): received 327 bytes
13:35:34 C:\libimobiledevice-vs-master\libimobiledevice\src\property_list_service.c:253 internal_plist_receive_timeout(): printing 327 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
        <key>MsgType</key>
        <string>ProgressMsg</string>
        <key>Operation</key>
        <integer>15</integer>
        <key>Progress</key>
        <integer>-1</integer>
</dict>
</plist>
Checking filesystems (15)
13:35:37 C:\libimobiledevice-vs-master\libimobiledevice\src\property_list_service.c:205 internal_plist_receive_timeout(): initial read=4
13:35:37 C:\libimobiledevice-vs-master\libimobiledevice\src\property_list_service.c:211 internal_plist_receive_timeout(): 327 bytes following
13:35:37 C:\libimobiledevice-vs-master\libimobiledevice\src\property_list_service.c:224 internal_plist_receive_timeout(): received 327 bytes
13:35:37 C:\libimobiledevice-vs-master\libimobiledevice\src\property_list_service.c:253 internal_plist_receive_timeout(): printing 327 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
        <key>MsgType</key>
        <string>ProgressMsg</string>
        <key>Operation</key>
        <integer>15</integer>
        <key>Progress</key>
        <integer>-1</integer>
</dict>
</plist>
Checking filesystems (15)
FDR 000001A1EABB80C0 got sync message
Connecting to FDR client at port 49158
13:35:37 C:\libimobiledevice-vs-master\libimobiledevice\src\idevice.c:605 idevice_connection_send(): internal_connection_send 10, sent 10
FDR Received 131 bytes
Got device identifier 6b8b4567327b23c6643c9869
FDR connected in reply to sync message, starting command thread
FDR 000001A1EABB80C0 waiting for message...
FDR 000001A1EABB7DC0 waiting for message...
FDR 000001A1EABB7DC0 got plist message
FDR Received 59 bytes
FDR sending 52 bytes:
C:\libimobiledevice-vs-master\idevicerestore\src\common.c:printing 214 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
        <key>Pong</key>
        <true/>
</dict>
</plist>
13:35:37 C:\libimobiledevice-vs-master\libimobiledevice\src\idevice.c:605 idevice_connection_send(): internal_connection_send 4, sent 4
13:35:37 C:\libimobiledevice-vs-master\libimobiledevice\src\idevice.c:605 idevice_connection_send(): internal_connection_send 52, sent 52
FDR Sent 52 bytes
FDR 000001A1EABB7DC0 waiting for message...
13:35:37 C:\libimobiledevice-vs-master\libimobiledevice\src\idevice.c:647 internal_connection_receive_timeout(): ERROR: usbmuxd_recv_timeout returned -108 (Unknown error)
ERROR: Unable to receive message from FDR 000001A1EABB7DC0 (-2). 0/2 bytes
FDR 000001A1EABB7DC0 terminating...
13:35:37 C:\libimobiledevice-vs-master\libimobiledevice\src\property_list_service.c:205 internal_plist_receive_timeout(): initial read=4
13:35:37 C:\libimobiledevice-vs-master\libimobiledevice\src\property_list_service.c:211 internal_plist_receive_timeout(): 327 bytes following
13:35:37 C:\libimobiledevice-vs-master\libimobiledevice\src\property_list_service.c:224 internal_plist_receive_timeout(): received 327 bytes
13:35:37 C:\libimobiledevice-vs-master\libimobiledevice\src\property_list_service.c:253 internal_plist_receive_timeout(): printing 327 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
        <key>MsgType</key>
        <string>ProgressMsg</string>
        <key>Operation</key>
        <integer>29</integer>
        <key>Progress</key>
        <integer>-1</integer>
</dict>
</plist>
Unmounting filesystems (29)

[Restore_Debug.txt](https://github.com/libimobiledevice/idevicerestore/files/4862865/Restore_Debug.txt)
 
Your help is highly appreciable 
Thank you

vipinbeni, 2020-12-21:
same issue i also facing any help is appreciable .

himavanth0007, 2022-10-14:
hey there im also finding this error of **Unable to receive message from FDR 0CFC94A0 (-4). 0/2 bytes**
can any one of you please email me the solution to **himavanth7@icloud.com**

thank you for your time.

15:46:50.030:  LibVersion: 20220922
15:46:50.030:  Start flash
15:46:50.030:  Checking PC DiskSpace
15:46:50.030:  Detecting Connection
15:46:50.031:  Found a device with mode: Recovery 
15:46:50.031:  Detecting ProductType
15:46:50.031:  matching device type through original information
15:46:50.032:  Checking iDevice firmware D:\3uTools\Firmware\iPhone12,3,iPhone12,5_16.0.3_20A392_Restore.ipsw
15:46:50.032:  Extract BuildManifest.plist
15:46:50.032:  Extracting Restore.plist from IPSW
15:46:50.042:  Firmware version: 16.0.3
15:46:50.042:  Product Build: 20A392 Major: 20
15:46:50.044:  Is Image4 supported: TRUE
15:46:50.044:  Extracting Firmware
15:46:50.045:  Reading ECID
15:46:50.045:  Found ECID 8462142318149678
15:46:50.045:  Reading NONCE
15:46:50.045:  Requesting SHSH
15:46:53.416:  Enter Recovery Mode
15:46:56.088:  Sending AppleLogo
15:46:57.544:  Sending Ramdisk
15:47:07.464:  Sending DeviceTree
15:47:07.485:  Sending RestoreSep
15:47:07.668:  Sending KernelCache
15:47:19.722:  Restoring
15:47:19.722:  About to restore device... 
15:47:22.149:  ERROR: build identity info does not contain a MacOSVariant
15:47:22.180:  Sending RootTicket
15:47:22.183:  Waiting for NAND
15:47:22.194:  Updating NAND Firmware
15:47:28.562:  Sending NORData
15:47:36.461:  ERROR: Unable to receive message from FDR 0CFC94A0 (-4). 0/2 bytes