libimobiledevice/idevicerestore #354: ERROR: Unable get path for 'Rap,RestoreRTKitOS' component

daxiongaijingxiang, 2020-07-08:
report ERROR: Unable get path for 'Rap,RestoreRTKitOS' component  four times,
and then report Status: Verification Error，
Result in failure。
Can you tell me why? think you!

daxiongaijingxiang, 2020-07-08:
comp_name = "Rap,RestoreRTKitOS"; 
if (build_identity_has_component(build_identity, comp_name))

I determined whether the file existed or not。
and Done sending FirmwareUpdater data。
but  the device asked Rose several times,and then report Status: Verification Error

daxiongaijingxiang, 2020-07-08:
iPhone 11 pro Max   flash  iPhone11,2,iPhone11,4,iPhone11,6,iPhone12,3,iPhone12,5_13.5.1_17F80_Restore.ipsw

daxiongaijingxiang, 2020-07-08:
creating RoseUpdater obj
updater_log: Rose: ACFUCommon::parseDebugArgs: Empty DebugArgs

updater_log: Rose: ACFURestore::init: ACFU Version: AppleConvergedFirmwareUpdater-59~1

updater_log: Rose: RoseTransport::roseControllerLogSink: unable to query rose boot_done value

updater_log: Rose: RoseUpdater::RoseUpdaterCreate: Updater object created successfully

updater_log: Rose: ACFURestore::logACFUConfig: ---- Restore State ----

updater_log: Rose: ACFURestore::logACFUConfig: Preflight: No

updater_log: Rose: ACFURestore::logACFUConfig: Post Sealing: Yes

updater_log: Rose: ACFURestore::logACFUConfig: Internal Build: No

updater_log: Rose: ACFURestore::logACFUConfig: ---- Restore State ----

updater_log: Rose: ACFURestore::executeCommand: Currently performing: Update

updater_log: Rose: ACFUCommon::logParameters: ChipID: 0x2006

updater_log: Rose: ACFUCommon::logParameters: BoardID: 0x0008

updater_log: Rose: ACFUCommon::logParameters: ECID: 0x0064010240245D18

updater_log: Rose: ACFUCommon::logParameters: Boot Nonce Hash: 0x0013E64E52FA7C489A7C33ED43C03F44EB09D1E03CD71131D949DAA6C2A5B9DA

updater_log: Rose: ACFUCommon::logParameters: Production Mode: 0x01

updater_log: Rose: ACFUCommon::logParameters: Security Mode: 0x01

updater_log: Rose: ACFUCommon::logParameters: Security Domain: 0x01

updater_log: Rose: ACFUCommon::logParameters: Minimum Epoch: 0x01

updater_log: Rose: ACFUCommon::logParameters: Chip Revision: 0x0010

updater_log: Rose: ACFURestore::executeCommand: Current command Query Command!

copying Rose updater response for loop 0
update_generic_firmware: response = {
    "@ServerVersion" = "2.1.0";
    FirmwareData = {length = 582375, bytes = 0x01000000 ffffffff 00000000 00000000 ... 32302034 2032310a };
    "Rap,Ticket" = {length = 2378, bytes = 0x30820946 1604494d 344d0201 00318202 ... 3bc8fd24 f781d701 };
}
updater_log: Rose: ACFURestore::logACFUConfig: ---- Restore State ----

updater_log: Rose: ACFURestore::logACFUConfig: Preflight: No

updater_log: Rose: ACFURestore::logACFUConfig: Post Sealing: Yes

updater_log: Rose: ACFURestore::logACFUConfig: Internal Build: No

updater_log: Rose: ACFURestore::logACFUConfig: ---- Restore State ----

updater_log: Rose: ACFURestore::executeCommand: Currently performing: Update

updater_log: Rose: ACFURTKitROM::updateFirmware: Booting firmware...

updater_log: Rose: RoseTransport::roseControllerLogSink: : AssertMacros: nextEntry,  file: /Library/Caches/com.apple.xbs/Sources/AppleSPUKext/AppleSPUKext-656.100.11/RoseControllerLib/RoseController.cpp, line: 1465 value:0

updater_log: Rose: RoseTransport::roseControllerLogSink: : AssertMacros: nextEntry,  file: /Library/Caches/com.apple.xbs/Sources/AppleSPUKext/AppleSPUKext-656.100.11/RoseControllerLib/RoseController.cpp, line: 1465 value:0

updater_log: Rose: RoseTransport::pushFirmware: Boot mode set to: local

updater_log: Rose: RoseTransport::pushFirmware: DownloadCustomFirmware failed with error (ret: 0xe00002d6)

updater_log: Rose: ACFURTKitROM::bootFirmware: failed to push firmware (3001)
[... 131 lines cut ...]
[13:32:12.0808-GMT]{3>4} CHECKPOINT BEGIN: RESTORED:[0x1614] cleanup_recovery_os_volume
restore-step-ids = {0x1103067B:57;0x11031310:81;0x11030668:97;0x11031503:100;0x11031614:102}
restore-step-names = {0x1103067B:perform_restore_installing;0x11031310:update_rose;0x11030668:update_firmware_post_sealing}
restore-step-uptime = 473
restore-step-user-progress = 90
[13:32:12.0808-GMT]{3>4} CHECKPOINT END: RESTORED:[0x1614] cleanup_recovery_os_volume
restore-step-ids = {0x1103067B:57;0x11031310:81;0x11030668:97;0x11031503:100}
restore-step-names = {0x1103067B:perform_restore_installing;0x11031310:update_rose;0x11030668:update_firmware_post_sealing}
restore-step-uptime = 473
restore-step-user-progress = 90
[13:32:12.0808-GMT]{3>4} CHECKPOINT BEGIN: RESTORED:[0x0647] cleanup_check_result
restore-step-ids = {0x1103067B:57;0x11031310:81;0x11030668:97;0x11031503:100;0x11030647:103}
restore-step-names = {0x1103067B:perform_restore_installing;0x11031310:update_rose;0x11030668:update_firmware_post_sealing}
restore-step-uptime = 473
restore-step-user-progress = 90
[13:32:12.0808-GMT]{3>4} CHECKPOINT END: RESTORED:[0x0647] cleanup_check_result
restore-step-ids = {0x1103067B:57;0x11031310:81;0x11030668:97;0x11031503:100}
restore-step-names = {0x1103067B:perform_restore_installing;0x11031310:update_rose;0x11030668:update_firmware_post_sealing}
restore-step-uptime = 473
restore-step-user-progress = 90
[13:32:12.0808-GMT]{3>4} CHECKPOINT BEGIN: RESTORED:[0x0648] cleanup_send_final_status
restore-step-ids = {0x1103067B:57;0x11031310:81;0x11030668:97;0x11031503:100;0x11030648:104}
restore-step-names = {0x1103067B:perform_restore_installing;0x11031310:update_rose;0x11030668:update_firmware_post_sealing}
restore-step-uptime = 473
restore-step-user-progress = 90

daxiongaijingxiang, 2020-07-08:
Received Rose ticket
Extracting ftab.bin...
NOTE: Build identity does not have a 'Rap,RestoreRTKitOS' component.
Sending FirmwareResponse data now...
Done sending FirmwareUpdater data
FDR 03944FE0 timeout waiting for command
FDR 03944FE0 waiting for message...

No data to read

No data to read
Updating Rose (65)
Got status message
Status: Verification Error

nikias, 2020-07-09:
HI, are you trying this on Windows?
I recently restored iPhone 11 Pro and iPhone 11 Pro Max without problems, on macOS.
The `Rap,RestoreRTKitOS` component does not exist on recent firmwares, and you only get a `NOTE:` which is not an error. Something else must be broken.

daxiongaijingxiang, 2020-07-09:
Yes,is in windows

nikias, 2020-07-09:
Windows support for idevicerestore is experimental. Unfortunately providing help for Windows is extremely difficult, since it depends on how (what compiler and potential manual changes) the binaries have been compiled. If you didn't compile everything yourself you we can pretty much not help you at all. It could be anything from libplist to libimobiledevice being faulty, over the dependencies like libzip behaving differently, to actual issues in idevicerestore, but we simply can't tell just by the above error message.
For Windows, the whole stack and idevicerestore are supposed to be compiled using MinGW. Everything else is not supported.

daxiongaijingxiang, 2020-07-09:
iPhone 5s is OK,but iPhone 11 pro is not OK

daxiongaijingxiang, 2020-07-09:
if ( v31 && !strncmp(v31, "12.", 3u) )
    v32 = 1;
  v33 = *(_DWORD *)(v29 + 64);
  if ( v33 && *(_QWORD *)(v33 + 24) >= 0xFui64 && v32 )
  {
    v34 = "Erase";
    if ( !(*(_BYTE *)v29 & 4) )
      v34 = "Update";
    v35 = plist_new_string(v34);
    v36 = (void (__cdecl *)(int, const char *, int))plist_dict_set_item;
    plist_dict_set_item(v30, "AuthInstallRestoreBehavior", v35);
    v37 = plist_new_uint(0, 0);
    plist_dict_set_item(v30, "AutoBootDelay", v37);
    v38 = plist_new_bool(1);
    plist_dict_set_item(v30, "CreateFilesystemPartitions", v38);
    v39 = plist_new_string("C:\\Users\\ADMNI~1\\AppData\\Local\\Temp\\Per52BF.tmp\\Firmware");
    plist_dict_set_item(v30, "FirmwareDirectory", v39);
    v40 = plist_new_bool(1);
    plist_dict_set_item(v30, "FlashNOR", v40);
    v41 = plist_new_bool(1);
    plist_dict_set_item(v30, "FormatForAPFS", v41);
    v42 = plist_new_bool(0);
    plist_dict_set_item(v30, "FormatForLwVM", v42);
    v43 = plist_new_bool(0);
    plist_dict_set_item(v30, "InstallDiags", v43);
    v44 = plist_new_string("production");
    plist_dict_set_item(v30, "NORImageType", v44);
    v45 = plist_new_string("C:\\Users\\ADMNI~1\\AppData\\Local\\Temp\\Per52BF.tmp");
    plist_dict_set_item(v30, "PersonalizedRestoreBundlePath", v45);
    if ( *(_DWORD *)(a1 + 96) )
    {
      v46 = plist_new_string(*(_DWORD *)(a1 + 96));
      plist_dict_set_item(v30, "RestoreBootArgs", v46);
    }
    v47 = plist_new_string("C:\\Users\\ADMNI~1\\AppData\\Local\\Temp\\Per52BF.tmp");
    v48 = v190;
    plist_dict_set_item(v190, "RestoreBundlePath", v47);
    v49 = plist_new_bool(0);
    plist_dict_set_item(v48, "RootToInstall", v49);
    v50 = plist_new_bool(0);
    plist_dict_set_item(v48, "SkipPreflightPersonalization", v50);
    v51 = plist_new_string("C:\\ProgramData\\Apple Computer\\iTunes\\iPhone Temporary Files");
    plist_dict_set_item(v48, "SourceRestoreBundlePath", v51);
    v52 = plist_new_dict();
    v53 = plist_new_bool(0);
    plist_dict_set_item(v52, "BasebandBootData", v53);
    v54 = plist_new_bool(0);
    plist_dict_set_item(v52, "BasebandData", v54);
    v55 = plist_new_bool(0);
    plist_dict_set_item(v52, "BasebandStackData", v55);
    v56 = plist_new_bool(0);
    plist_dict_set_item(v52, "BasebandUpdaterOutputData", v56);
    v57 = plist_new_bool(0);
    plist_dict_set_item(v52, "DataType", v57);
    v58 = plist_new_bool(0);
    plist_dict_set_item(v52, "DiagData", v58);
    v59 = plist_new_bool(0);
    plist_dict_set_item(v52, "FDRMemoryCommit", v59);
    v60 = plist_new_bool(0);
    plist_dict_set_item(v52, "FDRTrustData", v60);
    v61 = plist_new_bool(0);
    plist_dict_set_item(v52, "FUDData", v61);
    v62 = plist_new_bool(0);
    plist_dict_set_item(v52, "FirmwareUpdaterData", v62);
    v63 = plist_new_bool(0);
    plist_dict_set_item(v52, "GrapeFWData", v63);
    v64 = plist_new_bool(0);
    plist_dict_set_item(v52, "HPMFWData", v64);
    v65 = plist_new_bool(0);
    plist_dict_set_item(v52, "KernelCache", v65);
[... 187 lines cut ...]
    }
    v134 = plist_new_bool(0);
    plist_dict_set_item(v30, "UpdateBaseband", v134);
    v135 = plist_access_path(a2, 3, "Manifest", "SEP");
    if ( v135 )
    {
      v136 = plist_dict_get_item(v135, "RequiredCapacity");
      v137 = v136;
      if ( v136 )
      {
        if ( plist_get_node_type(v136) == 3 )
        {
          i = 0;
          plist_get_string_val(v137, &i);
          sub_10002CE0("TZ0RequiredCapacity: %s\n", (char)i);
          v138 = plist_copy(v137);
          plist_dict_set_item(v30, "TZ0RequiredCapacity", v138);
          free(i);
        }
      }
    }
    v139 = plist_new_bool(1);
    plist_dict_set_item(v30, "PersonalizedDuringPreflight", v139);
    v94 = v190;
  }

daxiongaijingxiang, 2020-07-09:
I decompilated someone else's Windows code and found that he made a judgment when plist_DICT_set_item，His code works fine to flash iPhone 11 pro max

nikias, 2020-07-09:
Can you point out the exact difference? It might be helpful to others.

daxiongaijingxiang, 2020-07-10:
I'm sorry, I haven't solved it all yet

daxiongaijingxiang, 2020-07-10:
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>AMRError</key>
	<integer>-1</integer>
	<key>Error</key>
	<dict>
		<key>Content</key>
		<dict>
			<key>Code</key>
			<integer>-1</integer>
			<key>Domain</key>
			<string>AMRestoreErrorDomain</string>
			<key>UserInfo</key>
			<dict>
				<key>Content</key>
				<dict>
					<key>NSDescription</key>
					<dict>
						<key>Content</key>
						<string>failed post-sealing updater</string>
						<key>EncodedObjectType</key>
						<string>CFStringRef</string>
					</dict>
					<key>NSUnderlyingError</key>
					<dict>
						<key>Content</key>
						<dict>
							<key>Code</key>
							<integer>-1</integer>
							<key>Domain</key>
							<string>AMRestoreErrorDomain</string>
							<key>UserInfo</key>
							<dict>
								<key>Content</key>
								<dict>
									<key>NSDescription</key>
									<dict>
										<key>Content</key>
										<string>ramrod post-sealing firmware update failed</string>
										<key>EncodedObjectType</key>
										<string>CFStringRef</string>
									</dict>
									<key>NSUnderlyingError</key>
									<dict>
										<key>Content</key>
										<dict>
											<key>Code</key>
											<integer>1017</integer>
											<key>Domain</key>
											<string>RamrodErrorDomain</string>
											<key>UserInfo</key>
											<dict>
												<key>Content</key>
												<dict>
													<key>NSDescription</key>
													<dict>
														<key>Content</key>
														<string>update_generic_firmware: failed to execute performNextStage on Rose updater</string>
														<key>EncodedObjectType</key>
														<string>CFStringRef</string>
													</dict>
													<key>NSUnderlyingError</key>
													<dict>
														<key>Content</key>
														<dict>
															<key>Code</key>
															<integer>3001</integer>
															<key>Domain</key>
															<string>ACFURestore::RoseRestore</string>
[... 81 lines cut ...]
restore-step-user-progress = 90
[13:44:05.0833-GMT]{3&gt;5} CHECKPOINT BEGIN: RESTORED:[0x0647] cleanup_check_result
restore-step-ids = {0x1103067B:57;0x11031310:81;0x11031309:82;0x11030668:97;0x11031503:100;0x11030647:103}
restore-step-names = {0x1103067B:perform_restore_installing;0x11031310:update_rose;0x11031309:update_se;0x11030668:update_firmware_post_sealing}
restore-step-uptime = 1028
restore-step-user-progress = 90
[13:44:05.0833-GMT]{3&gt;5} CHECKPOINT END: RESTORED:[0x0647] cleanup_check_result
restore-step-ids = {0x1103067B:57;0x11031310:81;0x11031309:82;0x11030668:97;0x11031503:100}
restore-step-names = {0x1103067B:perform_restore_installing;0x11031310:update_rose;0x11031309:update_se;0x11030668:update_firmware_post_sealing}
restore-step-uptime = 1028
restore-step-user-progress = 90
[13:44:05.0833-GMT]{3&gt;5} CHECKPOINT BEGIN: RESTORED:[0x0648] cleanup_send_final_status
restore-step-ids = {0x1103067B:57;0x11031310:81;0x11031309:82;0x11030668:97;0x11031503:100;0x11030648:104}
restore-step-names = {0x1103067B:perform_restore_installing;0x11031310:update_rose;0x11031309:update_se;0x11030668:update_firmware_post_sealing}
restore-step-uptime = 1028
restore-step-user-progress = 90
</string>
	<key>MsgType</key>
	<string>StatusMsg</string>
	<key>Status</key>
	<integer>-1</integer>
	<key>Successful</key>
	<false/>
</dict>
</plist>

daxiongaijingxiang, 2020-07-10:
update_generic_firmware: response = {
    "@ServerVersion" = "2.1.0";
    FirmwareData = {length = 582375, bytes = 0x01000000 ffffffff 00000000 00000000 ... 32302034 2032310a };
    "Rap,Ticket" = {length = 2378, bytes = 0x30820946 1604494d 344d0201 00318202 ... 3bc8fd24 f781d701 };
}
updater_log: Rose: ACFURestore::logACFUConfig: ---- Restore State ----

updater_log: Rose: ACFURestore::logACFUConfig: Preflight: No

updater_log: Rose: ACFURestore::logACFUConfig: Post Sealing: Yes

updater_log: Rose: ACFURestore::logACFUConfig: Internal Build: No

updater_log: Rose: ACFURestore::logACFUConfig: ---- Restore State ----

updater_log: Rose: ACFURestore::executeCommand: Currently performing: Update

updater_log: Rose: ACFURTKitROM::updateFirmware: Booting firmware...

updater_log: Rose: RoseTransport::roseControllerLogSink: : AssertMacros: nextEntry,  file: /Library/Caches/com.apple.xbs/Sources/AppleSPUKext/AppleSPUKext-656.100.11/RoseControllerLib/RoseController.cpp, line: 1465 value:0

updater_log: Rose: RoseTransport::roseControllerLogSink: : AssertMacros: nextEntry,  file: /Library/Caches/com.apple.xbs/Sources/AppleSPUKext/AppleSPUKext-656.100.11/RoseControllerLib/RoseController.cpp, line: 1465 value:0

updater_log: Rose: RoseTransport::pushFirmware: Boot mode set to: local

updater_log: Rose: RoseTransport::pushFirmware: DownloadCustomFirmware failed with error (ret: 0xe00002d6)

updater_log: Rose: ACFURTKitROM::bootFirmware: failed to push firmware (3001)

updater_log: Rose: ACFURTKitROM::updateFirmware: failed to boot firmware (3001)

updater_log: Rose: RoseRestore::performCmd: Failed to perform update firmware (ret: 3001)

updater_log: Rose: ACFURestore::executeCommand: We ran into some issues on try 1 of 1!

updater_log: Rose: RoseTransport::getDebugInfo: current firmware state: 0

updater_log: Rose: ACFUSynchronize::wait: event wait timeout

The file is in my local, why do I need to download it？

daxiongaijingxiang, 2020-07-23:
i known why! when i send Rap,RestoreRTKitOS' component to equipment， I need to  send Rap,ECID to apple server and get tickets，but my CPU is Little endian，I need to convert the big byte to get the correct request to send to the Apple server to get the correct Rap,Ticket；
                        </data>
			<key>Rap,ECID</key>
			<data>
			GF0kQAIBZAA=
			</data>
GF0kQAIBZAA=  is base64，I Error converting long long type

daxiongaijingxiang, 2020-07-23:
now，i can do it ，thank you！

closed 2020-07-27 by daxiongaijingxiang