libimobiledevice/idevicerestore #355: ERROR: Unable to receive message from FDR 0x562b2be01760 (-2). 0/2 bytes

lholliger, 2020-07-26:
I am attempting to restore T2 firmware on my 2018 MacBook Pro. The same issue occurs on my desktop and a separate linux system.

some recent log data:
```
Ctrl handshake done (ConnPort = 49157)
FDR 0x562b2be01760 waiting for message...
TZ0RequiredCapacity: 0x1000000
About to send RootTicket...
Sending RootTicket now...
Done sending RootTicket
Waiting for NAND (28)
Updating NAND Firmware (58)
About to send NORData...
DEBUG: tss_response_get_path_by_entry: No entry 'LLB' in TSS response
NOTE: Could not get LLB path from TSS data, will fetch from build identity
Found firmware path Firmware/all_flash
Getting firmware manifest from build identity
Extracting LLB.j132.RELEASE.im4p...
Personalizing IMG4 component LLB...
Tag found
Extracting DeviceTree.j132ap.im4p...
Personalizing IMG4 component DeviceTree...
Tag found
Extracting iBoot.j132.RELEASE.im4p...
Personalizing IMG4 component iBoot...
Tag found
Extracting sep-firmware.j132.RELEASE.im4p...
Personalizing IMG4 component RestoreSEP...
Tag found
Extracting sep-firmware.j132.RELEASE.im4p...
Personalizing IMG4 component SEP...
Tag found
common.c:supressed printing 12717642 bytes plist...
Sending NORData now...
Done sending NORData
Unmounting filesystems (29)
Unmounting filesystems (29)
Unmounting filesystems (29)
Checking filesystems (15)
About to send FDR Trust data...
Sending FDR Trust data now...
Done sending FDR Trust Data
Unmounting filesystems (29)
Unmounting filesystems (29)
Unmounting filesystems (29)
Requesting FUD data (36)
Found IsFUDFirmware component AOP
Found IsFUDFirmware component InputDevice
Found IsFUDFirmware component MacEFI
Found IsFUDFirmware component RestoreTrustCache
Found IsFUDFirmware component StaticTrustCache
Sending IsFUDFirmware image list
Extracting aopfw-t8012aop.im4p...
Personalizing IMG4 component AOP...
Tag found
Sending IsFUDFirmware for AOP...
Extracting J132_InputDevice.im4p...
Personalizing IMG4 component InputDevice...
Tag found
Sending IsFUDFirmware for InputDevice...
Extracting J132.RELEASE.im4p...
Personalizing IMG4 component MacEFI...
Tag found
Sending IsFUDFirmware for MacEFI...
Extracting 038-51468-074.dmg.trustcache...
Personalizing IMG4 component RestoreTrustCache...
Tag found
Sending IsFUDFirmware for RestoreTrustCache...
Extracting 038-51387-072.dmg.trustcache...
Personalizing IMG4 component StaticTrustCache...
Tag found
Sending IsFUDFirmware for StaticTrustCache...
Creating partition map (11)
ERROR: Could not read data (-256). Aborting.
ERROR: Unable to receive message from FDR 0x562b2be01760 (-2). 0/2 bytes
FDR 0x562b2be01760 terminating...
ERROR: Unable to restore device
idevice_event_cb: device 001c65ec3ef00026 (udid: 00008012-001C65EC3EF00026) disconnected
```

daxiongaijingxiang, 2020-07-31:
Are your Apple Mobile Device Service fail?

user813, 2020-12-09:
This also happens when restoring the iPhone13,2.